Koenkk/zigbee2mqtt #20119: [Feature request]: Allow finer, manual control of the LED Indicator on Aqara H1 Switches (Xiaomi WS-EUK01/02/03/04)

AkshayRao27, 2023-12-09:
### Is your feature request related to a problem? Please describe

My SO is annoyed at the inconsistency of the LED Indicators on the various Aqara H1 Switches we have throughout our flat and wants the LED indicators to behave consistently on all switches.

To achieve this, I need finer control of the LED indicators so that I can program the LED indicator to turn on or off regardless of `control_mode`.

### Describe the solution you'd like

An expose that allows manual control of the indicator LED on every switch (for multi-gang swithes) would perhaps be the best solution here. 

If, by default, when `operation_mode` is set to `decoupled`, every click of the switch toggles the LED indicator, the behaviour of the LED between the `decoupled` and `control_relay` modes would become more consistent.

The LED indicator is physically capable of turning on, turning off, blinking slowly, blinking fast, and has two colours - red and blue. If the LED indicator can be set to a specific value via this or additional exposes, it would open up the possiblity of integrating LED indicator behaviour into various automations.

### Describe alternatives you've considered

I have tried toggling `flip_indicator_light` option, but this does not make the behaviour more consistent between different switches or even within two gangs of the same switch if the operation modes are different.

### Additional context

The `operation_mode` on some of our switches that control regular lamps is set to `control_relay`, while others that control smart-bulbs are set to `decoupled`. In some rooms, we have double-rocker switches where the left is set to `control_relay` and the right is set to `decoupled`.
- When `operation_mode` is set to `control_relay`, the LED indicator turns on when the switch is turned on, and turns off when the switch is turned off
- When `operation_mode` is set to `decoupled`, the LED indicator never turns off at all.
This behaviour can be slightly modified by toggling `flip_indicator_light`, but this option does not make behaviour between the two operation modes more consistent. If this option is toggled, then:
- When `operation_mode` is set to `control_relay`, the LED indicator turns off when the switch is turned on, and turns on when the switch is turned off
- When `operation_mode` is set to `decoupled`, the LED indicator never turns on at all.

julianfs, 2023-12-09:
I know what you mean here, for me it is also a minor annoyance.

But, how do you keep track of what the LED status should be? If you have a light that can also be controlled by voice (e.g. Amazon Echo) or motion, when the voice command turns the light on, what should happen to the H1 LED?

arpaterson, 2023-12-09:
I also have this 'issue'. I'm not sure if what you are asking for is available in the H1 switch firmware or not. I really don't know, I have never had an aqara hub.

Are you willing to set up automations to control the LED, if you had unhindered control of it, independant of the relay?
It would be quite a hassle, but I would.
For example, I use one of the of the switches, decoupled, to control a water heater, and would like it to flash while the water heater is coming up to temperature. Another example is the bathroom fan automation - I would like it to flash to indicate the bathroom fan is not just ON but running an automation that will stop eventually.

AkshayRao27, 2023-12-09:
> But, how do you keep track of what the LED status should be? If you have a light that can also be controlled by voice (e.g. Amazon Echo) or motion, when the voice command turns the light on, what should happen to the H1 LED?

I don't imagine it being possible to keep track of what the LED should do without interacting with the switch. If I toggle a bulb directly, I don't see any out-of-the-box way for the switch to know that the LED should be toggled. I would instead need to send a `single action` to the switch, which would need to toggle the bulb, thereby toggling the LED at the same time. Alternatively, every automation would need to have an additional step included in it to toggle the LED. I agree that this is absolutely not an ideal solution, but I'm not sure if leaving the LED on permanently is ideal either. The current behaviour could of course be the fallback behaviour if this ever gets implemented.

> Are you willing to set up automations to control the LED, if you had unhindered control of it, independant of the relay? It would be quite a hassle, but I would.

I had already imagined a ton of use cases when I first bought these switches last year. I have been using them as "dumb switches" this whole time because I had not gotten around to setting my home server up till a few weeks ago, and I have started using Home Assistant for the first time ever literally last weekend.

Thw most simple automaton I want is for the state of the LED to correlate with the state of the lamp or whatever the switch is controlling. As I mentioned above, this is perhaps not trivial at all, but I hadn't imagined it being anyway.

Another automation that I wanted to implement is similar to your idea - there are ventilation fans that these my switches are hooked up to, and these fans have a delay before they come on. When I click the switch, I want the LED to blink during this lag period, and then be a steady on when the fan finally starts running. There's also a delay to the fan turning off, so when clicked to turn off, I want the LED to blink and then go off once the fan has turned off. The blinking tells me that the fan has already been switched on or off, and that I haven't in fact forgotten to toggle it.

There are also a few other automation I had thought up but I won't go on listing everything here.

Amusingly, when I first installed the switches and had no experience whatsoever with HA, I didn't know that it wasn't possible to control the LED indicator by default. To me, it seemed like such an obvious thing for Aqara to build into a device such as this 😄

arpaterson, 2023-12-09:
The thing is, we are asking for centralized control of every piece of i/o in the device, and none of the devices normal behaviour. Which is fine if you want granular control of literally everything and want to implement the entire behavior of devices centrally.

But that's not really how many zigbee products work/are implemented.

The H1 is a smart wall switch, the real intention for this switch is to control a dumb load/bulb. I got a few because they can be decoupled, and are then mains powered zigbee buttons, but that is already moving away from the intended application. For example, I don't think you can bind decoupled H1 switches to zigbee bulbs so that they will work even when the coordinator goes down.

If I was going to do it again, I would get wall dimmers, and do away with the expensive smart bulbs. Or assess a different product.

Getting off topic I guess, but worth thinking also about expectations.

julianfs, 2023-12-10:
Off topic, maybe, but a good discussion :)

Keeping track of the indicator lights I think would be possible in Home Assistant; wait for a light (or device) event, and flip the LED setting based checking if they're on or off.

The additional requirement would be individual control over each LED on the switch. In my usage, I'll often have a double rocker with the left side controlling lamps in the room and the right side for ceiling lights. Ceiling lights are typically dumb bulbs, with lamps being decoupled. So the lamp (decoupled) side is always on (and LED on), with the ceiling side typically off (and LED off). Flipping the LED settings only reverses the lights, so I can never have them how I want them.

I'd have to disagree with @arpaterson on the comment that these switches are meant for dumb bulbs; the ultimate (but expensive) implementation for me is smart switches & smart bulbs, allowing different brightness settings and soft on/off etc. I seem to remember the last time I took home assistant down, the switches reverted the decoupled sides to coupled, so as long as the smart bulbs are set to come on with power they work (a nice feature I thought).

Back to the original request by @AkshayRao27 - yes it would be great to have that finer control on these devices, if the firmware supports it.

AkshayRao27, 2023-12-12:
> if the firmware supports it

How can I find out if it does? Or is this something the devs will need to look into?

julianfs, 2023-12-13:
That's a tough one, outside of my knowledge.

Maybe a question for Aqara themselves? I don't know how keen they are on sharing info. The firmware on the switches is quite old (early 2021 on my device), and a google search didn't find any mention of updates.

AkshayRao27, 2023-12-13:
I just emailed them about it. I'll report back whenever I hear from them. Hopefully one of the devs can dig into this in the mean time 😄

julianfs, 2023-12-13:
Thanks, good luck!

Something I forgot to mention earlier. I've also got an H1 wireless double-rocker in my house and these have more functions than the wired devices, which maybe suggests Z2M isn't exposing all functions (or they simply are newer and more advanced)?

In addition to the single click, double click, single-both click, the wireless ones have: hold-left, hold-right, hold both, double-both click, triple-left click and triple-right click. 11 actions on a 2 rocker switch!

AkshayRao27, 2023-12-19:
Well Aqara has been less than helpful
[image]

ShadowFistOkami, 2024-01-22:
@AkshayRao27 most led indicators on switches are hardwired to the **relay** circuit. It's been that way since the first wifi smart switches showed up (used to be a disclaimer in Tasmota), and I doubt it'll be any different for the H1's.
Your only option is to make sure any switch gangs set to decoupled mode are wired to a permanent live (same as switches which don't support decoupled mode). That way you can still control your smart bulbs with automations, but only if you also include an action which turns on the relay. 
The downside to this is that you'll have a relay click, which might be mostly fine for a single toggle event, but the clicking will probably drive your wife crazy if you want to implement an led state change while your water heater (slowly) gets up to temperature.

AkshayRao27, 2024-06-26:
I don't know if this changed recently, but the [Aqara Website](https://www.aqara.com/eu/product/smart-wall-switch-h1-eu-with-neutral/) now explicitly states that controlling the LED behaviour is possible. Specially, the LED can be disabled completely.

Is this likely a Hardware change?

nigmastar, 2024-10-20:
Solution at the bottom!

**Side consideration**

For me this is a major issue. It's so annoying that these "smart" switches cost tens of dollars and they can't be bothered to connected the LED on a separate PIN on the board. For example look at the Sonoff M5. It would be the perfect wifi switch if it wasn't for the LEDs that can't be indepently controlled.

Most of my switched are Sonoff TX ultimate. Once flashed with ESPhome you have control over everything. To those asking " how you choose what the state of the LED should be... etc.", I set in the switch firmware that each time the button is clicked the LED indicator should toggle. Then, in homeassistant, the automation makes sure that the state of the switch LED is aligned with the state of the light... this means that if you ask Alexa or the Wiz app or whatever to switch off your lights, then the LED of the switch will also switch off.

**Solution** 

Now, to your problem with the H1. I have 2 of them (to extend my Zigbee network across the staircase) and I set them both in decoupled mode. As everyone told you, you can't control directly the LED (unless you alter the state of the relay). However, in Zigbee2MTQQ (Z2M), there is an option to "flip indicator light". This allows you to effectively decide whether the LED should be on or off when the relay is on (which how it always is when in decoupled mode) or off. 

Therefore all you need is an extra automation that toggles the state of the "flip indicator light" whenever you press any of the buttons controlling the light connected to yout H1.

[image]

AkshayRao27, 2024-10-20:
This works ONLY if you're using a single rocker switch, or if both buttons on a double rocker switch are set to decoupled and toggled at the same time.

Consider these two cases (both of which I have in my flat).

1. Left Switch is decoupled because it controls a light, and right switch is relay because it controls a ventilation fan. If I use this solution, then the one of the LEDs is ALWAYS in regardless of the state of my bulb / fan.
2. Both left and right switches are set to decoupled and control two different bulbs. Again, in this case, changing the Flip indicator light option will result in one or both LEDs being on or off at all times.

The issue is that the flip indicator light function cannot be toggled for each of the switches, but only for the whole unit at once.

nigmastar, 2024-10-20:
Ah true. I have two 1-gang switches, so wasn't a problem for me.

But since each LED is connected to its own relay, maybe it should be
possible to have separate Flip Indicators in future updates.

AkshayRao27, 2024-10-20:
If `flip_indicator` is implemented per switch, then this is certainly a viable workaround.

closed 2025-01-09 by github-actions[bot]