Koenkk/zigbee2mqtt #20121: [Xiaomi GZCGQ11LM] Device didn't respond to OTA reques

christianbalz, 2023-12-09:
### What happened?

Aqara T1 light intensity sensor device didn't respond to OTA reques

[image]

### What did you expect to happen?

Device endpoint linked as expected in order to upgrade the firmware successfully

### How to reproduce it (minimal and precise)

Check OTA update

### Zigbee2MQTT version

1.34.0 commit: aae7312

### Adapter firmware version

2019www - Firmware build date 20200312

### Adapter

Aqara T1 light intensity sensor GZCGQ11LM

### Debug log

info  2023-12-09 12:33:35: MQTT publish: topic 'zigbee2mqtt/bridge/response/device/ota_update/check', payload '{"data":{"id":"Sensor luminosidade sala"},"error":"Failed to check if update available for 'Sensor luminosidade sala' (Device didn't respond to OTA request)","status":"error","transaction":"veu0g-1"}'
error 2023-12-09 12:33:35: Failed to check if update available for 'Sensor luminosidade sala' (Device didn't respond to OTA request)

lux73, 2023-12-10:
you have to Push the pairing Button shortly on the Device after triggering OTA to set the Battery Device into OTA state

this Way works with my [Xiaomi RTCGQ14LM](https://www.zigbee2mqtt.io/devices/RTCGQ14LM.html)

christianbalz, 2023-12-10:
> you have to Push the pairing Button shortly on the Device after triggering OTA to set the Battery Device into OTA state
> 
> this Way works with my [Xiaomi RTCGQ14LM](https://www.zigbee2mqtt.io/devices/RTCGQ14LM.html)

It's works fine. Thank you @lux73

christianbalz, 2023-12-10:
the solution provided worked well

christianbalz, 2023-12-10:
The solution provided worked well

closed 2023-12-10 by christianbalz