Koenkk/zigbee2mqtt #20123: Tuya ZY-M100-24G (Zigbee Model TS0601; Zigbee Manufacturer _TZE204_ijxvkhd0 )

pmrosa-classes, 2023-12-09:
### What happened?

This device is supported and paired & shown as the correct device. It works if nothing is changes in the values exposed. After changing some values the presence stops working. When changing the values in the slides, mainly Motion sensitivity and Presence Sensitivity) they switch back to 10 after some seconds.

### What did you expect to happen?

Values stick to the ones choosen and device works.

### How to reproduce it (minimal and precise)

Changing some values exposed, mainly the two described.

### Zigbee2MQTT version

 1.34.0

### Adapter firmware version

20211115

### Adapter

zStack12

JakoobCZ, 2023-12-09:
Changing the temperature works? It is only the Motion sensitivity and Presence Sensitivity that doesn't work?

pmrosa-classes, 2023-12-09:
This is a presence detector, no temperature available.

pmrosa-classes, 2023-12-09:
Same issue as #18237. Solved with soolution presented by a user.
Thanks.

closed 2023-12-09 by pmrosa-classes