Koenkk/zigbee2mqtt #20124: Xiaomi ZNJLBL01LM blinds report incorrect status

phairplay, 2023-12-09:
### What happened?

I have an automation that opens my blinds in the morning, yet I notice later on in the day HA sees the blinds has closed, yet they are still physically open.
this is causing an issue when I need to the blinds closed at night.

```
info  2023-12-09 08:03:36: MQTT publish: topic 'zigbee2mqtt/Living Room Blinds', payload '{"battery":81,"charging_status":false,"linkquality":83,"motor_state":"rising","position":0,"power_outage_count":18,"running":true,"state":"OFF","update":{"installed_version":-1,"latest_version":-1,"state":null},"update_available":null}'
info  2023-12-09 08:03:40: MQTT publish: topic 'zigbee2mqtt/Living Room Blinds', payload '{"battery":81,"charging_status":false,"linkquality":94,"motor_state":"pause","position":0,"power_outage_count":18,"running":false,"state":"OFF","update":{"installed_version":-1,"latest_version":-1,"state":null},"update_available":null}'
info  2023-12-09 08:03:40: MQTT publish: topic 'zigbee2mqtt/Living Room Blinds', payload '{"battery":81,"charging_status":false,"linkquality":98,"motor_state":"pause","position":100,"power_outage_count":18,"running":false,"state":"OFF","update":{"installed_version":-1,"latest_version":-1,"state":null},"update_available":null}'
info  2023-12-09 08:03:40: MQTT publish: topic 'zigbee2mqtt/Living Room Blinds', payload '{"battery":81,"charging_status":false,"linkquality":98,"motor_state":"pause","position":100,"power_outage_count":18,"running":false,"state":"OFF","update":{"installed_version":-1,"latest_version":-1,"state":null},"update_available":null}'info  2023-12-09 08:03:40: MQTT publish: topic 'zigbee2mqtt/Games Room blinds', payload '{"battery":82,"charging_status":false,"device_temperature":17,"linkquality":105,"motor_state":"pause","position":100,"power_outage_count":1,"running":false,"state":"OFF","update":{"installed_version":-1,"latest_version":-1,"state":null},"update_available":null}'
```

as you can see later this morning, the blinds were updated stating closed "Position 0", yet the motor status never changed

```
info  2023-12-09 11:42:19: MQTT publish: topic 'zigbee2mqtt/Living Room Blinds', payload '{"battery":81,"charging_status":false,"linkquality":105,"motor_state":"pause","position":100,"power_outage_count":18,"running":false,"state":"OFF","update":{"installed_version":-1,"latest_version":-1,"state":null},"update_available":null}'
info  2023-12-09 11:42:21: MQTT publish: topic 'zigbee2mqtt/Living Room Blinds', payload '{"battery":81,"charging_status":false,"linkquality":109,"motor_state":"pause","position":0,"power_outage_count":18,"running":false,"state":"OFF","update":{"installed_version":-1,"latest_version":-1,"state":null},"update_available":null}'
info  2023-12-09 11:42:27: MQTT publish: topic 'zigbee2mqtt/Living Room Blinds', payload '{"battery":81,"charging_status":false,"linkquality":109,"motor_state":"pause","position":0,"power_outage_count":19,"running":false,"state":"OFF","update":{"installed_version":-1,"latest_version":-1,"state":null},"update_available":null}'

```

### What did you expect to happen?

the blinds not to report in incorrect status

### Zigbee2MQTT version

1.34.0-1

### Adapter firmware version

20220219

### Adapter

Silicon_Labs_Sonoff_Zigbee_3.0_USB_Dongle_Plus

phairplay, 2023-12-10:
Updating my zigbee stick to the latest version hasn't resolved the issue

CMDR-Sloma, 2023-12-22:
I have a similar problem. I have Venetian blinds which are controlled by this controller, so both positions 0 and 100 means they are closed. However, it randomly goes back to 100 / open state without even moving a motor.

closed 2024-07-20 by github-actions[bot]