Koenkk/zigbee2mqtt #20130: Zigbee2MQTT Crashes when using Addon in Home Assistant - aqaraOpple - Aqara Pet Feeder C1 goes offline

jeells102, 2023-12-09:
### What happened?

My Z2M addon keeps crashing and I think it's linked to my Aqara C1 Pet Feeders. I have 2 of these, and one or the other keep crashing. If I have watchdog enabled, then the addon will restart but then I can't use one of the feeders, but I have turned it off and captured these errors. 

```
Zigbee2MQTT:info  2023-12-09 21:00:27: MQTT publish: topic 'zigbee2mqtt/Emma's Room Light Switch', payload '{"brightness":238,"linkquality":102,"power_on_behavior":null,"state":"ON"}'
2023-12-09T21:00:27.627Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,100,1,0,100]
2023-12-09T21:00:27.627Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,100,1,0,100]
2023-12-09T21:00:27.627Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 4 - 1 - [0] - 100
2023-12-09T21:00:27.627Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- AF - dataRequest - {"status":0}
2023-12-09T21:00:27.627Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2023-12-09T21:00:27.632Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,3,68,128,0,1,187,125]
2023-12-09T21:00:27.632Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,68,128,0,1,187,125]
2023-12-09T21:00:27.632Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 2 - 4 - 128 - [0,1,187] - 125
2023-12-09T21:00:27.632Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - dataConfirm - {"status":0,"endpoint":1,"transid":187}
2023-12-09T21:00:27.632Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2023-12-09T21:00:27.689Z zigbee-herdsman:adapter:zStack:adapter Response timeout (0x54ef441000711cd0:34757,1)
2023-12-09T21:00:27.689Z zigbee-herdsman:controller:endpoint Write 0x54ef441000711cd0/1 aqaraOpple({"65521":{"value":{"type":"Buffer","data":[0,2,5,4,21,0,85,1,1]},"type":65}}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":4447,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 34757 - 1 - 247 - 64704 - 4 after 10000ms)
Error: Write 0x54ef441000711cd0/1 aqaraOpple({"65521":{"value":{"type":"Buffer","data":[0,2,5,4,21,0,85,1,1]},"type":65}}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":4447,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 34757 - 1 - 247 - 64704 - 4 after 10000ms)
    at Timeout._onTimeout (/app/node_modules/zigbee-herdsman/src/utils/waitress.ts:64:35)
    at listOnTimeout (node:internal/timers:559:17)
    at processTimers (node:internal/timers:502:7)
```

```
Zigbee2MQTT:info  2023-12-09 17:07:10: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"Jake's Room Light Switch","ieee_address":"0x84b4dbfffe089e28"},"type":"device_announce"}'
2023-12-09T17:07:10.427Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,100,2,17,118]
2023-12-09T17:07:10.427Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,100,2,17,118]
2023-12-09T17:07:10.427Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 4 - 2 - [17] - 118
2023-12-09T17:07:10.427Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- AF - dataRequestExt - {"status":17}
2023-12-09T17:07:10.427Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2023-12-09T17:07:10.427Z zigbee-herdsman:adapter:zStack:znp:SREQ --> AF - dataRequest - {"dstaddr":11397,"destendpoint":1,"srcendpoint":1,"clusterid":2820,"transid":231,"options":0,"radius":30,"len":9,"data":{"type":"Buffer","data":[16,13,0,5,5,8,5,11,5]}}
2023-12-09T17:07:10.427Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,19,36,1,133,44,1,1,4,11,231,0,30,9,16,13,0,5,5,8,5,11,5,126]
Error: SREQ '--> AF - dataRequestExt - {"dstaddrmode":2,"dstaddr":"0x000000000000fffd","destendpoint":242,"dstpanid":0,"srcendpoint":242,"clusterid":33,"transid":230,"options":0,"radius":30,"len":6,"data":{"type":"Buffer","data":[25,12,2,11,254,0]}}' failed with status '(0x11: BUFFER_FULL)' (expected '(0x00: SUCCESS)')
    at Object.func (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/znp/znp.ts:322:27)
    at Queue.executeNext (/app/node_modules/zigbee-herdsman/src/utils/queue.ts:32:32)
```

I thought this was due to a Tuya BSD29 smart plug that I have taken out of rotation but it has crashed since. I have an automation using MQTT publish commands to send the feed command rather than using the direct integration changing the buttons but this hasn't resolved my issues. My whole network goes offline because the addon crashes. I am running Proxmox with the Home Assistant VM and I have given it 8GB of RAM and it's still crashing. 

### What did you expect to happen?

Not to have the addon crash on me. 

### How to reproduce it (minimal and precise)

Start addon
Have an automation send feed command
Wait a few hours
Addon crashes

### Zigbee2MQTT version

1.34.0

### Adapter firmware version

20230507

### Adapter

Silicon_Labs_Sonoff_Zigbee_3.0_USB_Dongle_Plus

### Debug log

[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/13625697/log.txt)

maciekdnd, 2024-01-01:
Same here, but for Aqara T1 and ZY-M100-S_1.  SONOFF_Zigbee_3.0_USB_Dongle_Plus_V2_20221202083941-if00

Koenkk, 2024-01-02:
@jeells102 can you try the 20221226 firmware?

ajfriesen, 2024-01-07:
I am using a sonoff dongle 3.0 P and have the same behavior with the crash, when using the feed action.

My dongle path in zigbee2mqtt:
```
/dev/serial/by-id/usb-Silicon_Labs_CP2102N_USB_to_UART_Bridge_Controller_4aa7ceef3d93eb119cf8194f3d98b6d1-if00-port0
```

I just flashed the latest coordinator firmware just to rule out an old firmware:
https://github.com/Koenkk/Z-Stack-firmware/raw/Z-Stack_3.x.0_coordinator_20230507/coordinator/Z-Stack_3.x.0/bin/CC1352P2_CC2652P_launchpad_coordinator_20230507.zip

Used the ZigStar TI CC2652P/P7 FW Flasher AddOnn.

When I add the device to the network I get this:
```
Error 2024-01-07 15:00:32Exception while calling fromZigbee converter: Attempt to access memory outside buffer bounds}
Info 2024-01-07 15:00:32Successfully configured '0x54ef441000458a71'
```

Anything I can provide you with that could help to narrow down the problem @Koenkk

Koenkk, 2024-01-07:
@ajfriesen that error is a bit different, could you provide the herdsman debug log of this?

See [this](https://www.zigbee2mqtt.io/guide/usage/debug.html) on how to enable the herdsman debug logging. Note that this is only logged to STDOUT and not to log files.

ajfriesen, 2024-01-07:
@Koenkk 

I enabled the herdsman debug log but could not reproduce the issue anymore.
Maybe it was related to the ancient firmware I was running on my sonoff 3.0 P stick.
I will monitor the logs if everything works properly, but my testing has not let me find any issues so far.

Here the herdsman log for completeness. The device got added as `0x54ef441000458a71`.

```
2024-01-07T16:13:29.068Z zigbee-herdsman:controller:device:log Interview - got simple descriptor for endpoint '242' device '0x54ef441000458a71'
2024-01-07T16:13:29.069Z zigbee-herdsman:controller:device:log Interview - completed for device '0x54ef441000458a71'
2024-01-07T16:13:29.070Z zigbee-herdsman:controller:database:log Writing database to '/config/zigbee2mqtt/database.db'
2024-01-07T16:13:29.076Z zigbee-herdsman:controller:log Succesfully interviewed '0x54ef441000458a71'
Zigbee2MQTT:info  2024-01-07 17:13:29: Successfully interviewed '0x54ef441000458a71', device has successfully been paired
Zigbee2MQTT:info  2024-01-07 17:13:29: Device '0x54ef441000458a71' is supported, identified as: Xiaomi Aqara pet feeder C1 (ZNCWWSQ01LM)
Zigbee2MQTT:info  2024-01-07 17:13:29: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"definition":{"description":"Aqara pet feeder C1","exposes":[{"access":3,"description":"Start feeding","label":"Feed","name":"feed","property":"feed","type":"enum","values":["","START"]},{"access":1,"description":"Feeding source","label":"Feeding source","name":"feeding_source","property":"feeding_source","type":"enum","values":["schedule","manual","remote"]},{"access":1,"description":"Feeding size","label":"Feeding size","name":"feeding_size","property":"feeding_size","type":"numeric","unit":"portion"},{"access":1,"description":"Portions per day","label":"Portions per day","name":"portions_per_day","property":"portions_per_day","type":"numeric"},{"access":1,"description":"Weight per day","label":"Weight per day","name":"weight_per_day","property":"weight_per_day","type":"numeric","unit":"g"},{"access":1,"description":"Indicates whether there is an error with the feeder","label":"Error","name":"error","property":"error","type":"binary","value_off":false,"value_on":true},{"access":3,"description":"Feeding schedule","item_type":{"access":3,"features":[{"access":3,"label":"Days","name":"days","property":"days","type":"enum","values":["everyday","workdays","weekend","mon","tue","wed","thu","fri","sat","sun","mon-wed-fri-sun","tue-thu-sat"]},{"access":3,"label":"Hour","name":"hour","property":"hour","type":"numeric"},{"access":3,"label":"Minute","name":"minute","property":"minute","type":"numeric"},{"access":3,"label":"Size","name":"size","property":"size","type":"numeric"}],"label":"DayTime","name":"dayTime","type":"composite"},"label":"Schedule","name":"schedule","property":"schedule","type":"list"},{"features":[{"access":3,"description":"Led indicator","label":"State","name":"state","property":"led_indicator","type":"binary","value_off":"OFF","value_on":"ON","value_toggle":"TOGGLE"}],"type":"switch"},{"features":[{"access":3,"description":"Enables/disables physical input on the device","label":"State","name":"state","property":"child_lock","type":"binary","value_off":"UNLOCK","value_on":"LOCK"}],"label":"Child lock","type":"lock"},{"access":3,"description":"Feeding mode","label":"Mode","name":"mode","property":"mode","type":"enum","values":["schedule","manual"]},{"access":3,"description":"One serving size","label":"Serving size","name":"serving_size","property":"serving_size","type":"numeric","unit":"portion","value_max":10,"value_min":1},{"access":3,"description":"Portion weight","label":"Portion weight","name":"portion_weight","property":"portion_weight","type":"numeric","unit":"g","value_max":20,"value_min":1},{"access":1,"description":"Link quality (signal strength)","label":"Linkquality","name":"linkquality","property":"linkquality","type":"numeric","unit":"lqi","value_max":255,"value_min":0}],"model":"ZNCWWSQ01LM","options":[],"supports_ota":true,"vendor":"Xiaomi"},"friendly_name":"0x54ef441000458a71","ieee_address":"0x54ef441000458a71","status":"successful","supported":true},"type":"device_interview"}'
Zigbee2MQTT:info  2024-01-07 17:13:29: Configuring '0x54ef441000458a71'
2024-01-07T16:13:29.133Z zigbee-herdsman:controller:endpoint Read 0x54ef441000458a71/1 aqaraOpple([65521], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":4447,"transactionSequenceNumber":null,"writeUndiv":false})
2024-01-07T16:13:29.133Z zigbee-herdsman:adapter:zStack:adapter sendZclFrameToEndpointInternal 0x54ef441000458a71:64825/1 (0,0,1)
2024-01-07T16:13:29.134Z zigbee-herdsman:adapter:zStack:znp:SREQ --> AF - dataRequest - {"dstaddr":64825,"destendpoint":1,"srcendpoint":1,"clusterid":64704,"transid":18,"options":0,"radius":30,"len":7,"data":{"type":"Buffer","data":[20,95,17,17,0,241,255]}}
2024-01-07T16:13:29.134Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,17,36,1,57,253,1,1,192,252,18,0,30,7,20,95,17,17,0,241,255,130]
Zigbee2MQTT:info  2024-01-07 17:13:29: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_successful","meta":{"description":"Aqara pet feeder C1","friendly_name":"0x54ef441000458a71","model":"ZNCWWSQ01LM","supported":true,"vendor":"Xiaomi"},"type":"pairing"}'
[... cut ...] ommandIdentifier":10},"Payload":[{"attrId":0,"dataType":41,"attrData":1011},{"attrId":20,"dataType":40,"attrData":-1},{"attrId":16,"dataType":41,"attrData":10111}],"Command":{"ID":10,"name":"report","parameters":[{"name":"attrId","type":33},{"name":"dataType","type":32},{"name":"attrData","type":1000}]}},"address":22213,"endpoint":1,"linkquality":69,"groupID":0,"wasBroadcast":false,"destinationEndpoint":1}'
2024-01-07T16:13:40.283Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []

```

ajfriesen, 2024-01-07:
So, I just updated the firmware via OTA of the pet feeder and got later the memory message again with the herdsman debug log enabled.

Does this help @Koenkk ?
If you need anything else let me know.
I did not redact anything since I do not know what to redact 😅 

```
2024-01-07T17:41:31.561Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - incomingMsg - {"groupid":0,"clusterid":64704,"srcaddr":64825,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":102,"securityuse":0,"timestamp":7824250,"transseqnumber":0,"len":18,"data":{"type":"Buffer","data":[28,95,17,6,10,241,255,65,9,0,5,12,14,95,0,85,1,8]}}
2024-01-07T17:41:31.562Z zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":0,"manufacturerSpecific":true,"direction":1,"disableDefaultResponse":true,"reservedBits":0},"transactionSequenceNumber":6,"manufacturerCode":4447,"commandIdentifier":10},"Payload":[{"attrId":65521,"dataType":65,"attrData":{"type":"Buffer","data":[0,5,12,14,95,0,85,1,8]}}],"Command":{"ID":10,"name":"report","parameters":[{"name":"attrId","type":33},{"name":"dataType","type":32},{"name":"attrData","type":1000}]}},"address":64825,"endpoint":1,"linkquality":102,"groupID":0,"wasBroadcast":false,"destinationEndpoint":1}'
2024-01-07T17:41:31.562Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
Zigbee2MQTT:info  2024-01-07 18:41:31: MQTT publish: topic 'zigbee2mqtt/Futterspender', payload '{"child_lock":"UNLOCK","error":null,"feed":null,"feeding_size":null,"feeding_source":null,"led_indicator":"OFF","linkquality":102,"mode":"manual","portion_weight":8,"portions_per_day":0,"schedule":null,"serving_size":1,"update":{"installed_version":9505,"latest_version":9761,"progress":100,"remaining":16,"state":"updating"},"update_available":false,"weight_per_day":null}'
2024-01-07T17:41:31.583Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,5,69,196,57,253,1,228,5,160]
2024-01-07T17:41:31.583Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,5,69,196,57,253,1,228,5,160]
2024-01-07T17:41:31.584Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 5 - 2 - 5 - 196 - [57,253,1,228,5] - 160
2024-01-07T17:41:31.584Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - srcRtgInd - {"dstaddr":64825,"relaycount":1,"relaylist":[1508]}
2024-01-07T17:41:31.584Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2024-01-07T17:41:31.643Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,38,68,129,0,0,192,252,57,253,1,1,0,102,0,125,119,119,0,0,18,28,95,17,7,10,241,255,65,9,0,5,13,4,24,0,85,1,1,228,5,28,183]
2024-01-07T17:41:31.643Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,38,68,129,0,0,192,252,57,253,1,1,0,102,0,125,119,119,0,0,18,28,95,17,7,10,241,255,65,9,0,5,13,4,24,0,85,1,1,228,5,28,183]
2024-01-07T17:41:31.643Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 38 - 2 - 4 - 129 - [0,0,192,252,57,253,1,1,0,102,0,125,119,119,0,0,18,28,95,17,7,10,241,255,65,9,0,5,13,4,24,0,85,1,1,228,5,28] - 183
2024-01-07T17:41:31.643Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - incomingMsg - {"groupid":0,"clusterid":64704,"srcaddr":64825,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":102,"securityuse":0,"timestamp":7829373,"transseqnumber":0,"len":18,"data":{"type":"Buffer","data":[28,95,17,7,10,241,255,65,9,0,5,13,4,24,0,85,1,1]}}
2024-01-07T17:41:31.644Z zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":0,"manufacturerSpecific":true,"direction":1,"disableDefaultResponse":true,"reservedBits":0},"transactionSequenceNumber":7,"manufacturerCode":4447,"commandIdentifier":10},"Payload":[{"attrId":65521,"dataType":65,"attrData":{"type":"Buffer","data":[0,5,13,4,24,0,85,1,1]}}],"Command":{"ID":10,"name":"report","parameters":[{"name":"attrId","type":33},{"name":"dataType","type":32},{"name":"attrData","type":1000}]}},"address":64825,"endpoint":1,"linkquality":102,"groupID":0,"wasBroadcast":false,"destinationEndpoint":1}'
2024-01-07T17:41:31.644Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
Zigbee2MQTT:info  2024-01-07 18:41:31: MQTT publish: topic 'zigbee2mqtt/Futterspender', payload '{"child_lock":"UNLOCK","error":null,"feed":null,"feeding_size":null,"feeding_source":null,"led_indicator":"OFF","linkquality":102,"mode":"schedule","portion_weight":8,"portions_per_day":0,"schedule":null,"serving_size":1,"update":{"installed_version":9505,"latest_version":9761,"progress":100,"remaining":16,"state":"updating"},"update_available":false,"weight_per_day":null}'
2024-01-07T17:41:31.668Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,5,69,196,57,253,1,228,5,160]
2024-01-07T17:41:31.668Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,5,69,196,57,253,1,228,5,160]
2024-01-07T17:41:31.668Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 5 - 2 - 5 - 196 - [57,253,1,228,5] - 160
2024-01-07T17:41:31.668Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - srcRtgInd - {"dstaddr":64825,"relaycount":1,"relaylist":[1508]}
2024-01-07T17:41:31.668Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2024-01-07T17:41:31.724Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,41,68,129,0,0,192,252,57,253,1,1,0,109,0,3,139,119,0,0,21,28,95,17,8,10,241,255,65,12,0,5,14,13,105,0,85,4,0,0,0,0,228,5,28,67]
[... cut ...]
Zigbee2MQTT:info  2024-01-07 18:42:57: Device 'Futterspender' was updated from 'null' to 'null'
Zigbee2MQTT:info  2024-01-07 18:42:57: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"id":"Futterspender","transaction":"auykb-3"},"meta":{"device":"Futterspender","from":null,"status":"update_succeeded","to":null},"type":"ota_update"}'
Zigbee2MQTT:info  2024-01-07 18:42:57: MQTT publish: topic 'zigbee2mqtt/bridge/response/device/ota_update/update', payload '{"data":{"from":null,"id":"Futterspender","to":null},"status":"ok","transaction":"auykb-3"}'
```

Koenkk, 2024-01-08:
@ajfriesen could you also set the log level to debug? That should show the stack trace.

See [this](https://www.zigbee2mqtt.io/guide/usage/debug.html) on how to enable debug logging.

ajfriesen, 2024-01-08:
@Koenkk 
I enabled the debug log for the zigbee addon. Then I removed and added the pet dispenser again.

The log was too big  (59MB) therefore I had to zip it.
[log-zigbee.tgz](https://github.com/Koenkk/zigbee2mqtt/files/13866570/log-zigbee.tgz)

ajfriesen, 2024-01-08:
I was able to set a feeding schedule via zigbee2mqtt got these logs:

```
2024-01-08T22:33:08.683Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,41,68,129,0,0,192,252,253,88,1,1,0,120,0,156,239,82,0,0,21,28,95,17,29,10,241,255,65,12,0,5,120,4,21,2,188,4,48,48,48,49,253,88,29,80]
2024-01-08T22:33:08.683Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,41,68,129,0,0,192,252,253,88,1,1,0,120,0,156,239,82,0,0,21,28,95,17,29,10,241,255,65,12,0,5,120,4,21,2,188,4,48,48,48,49,253,88,29,80]
2024-01-08T22:33:08.683Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 41 - 2 - 4 - 129 - [0,0,192,252,253,88,1,1,0,120,0,156,239,82,0,0,21,28,95,17,29,10,241,255,65,12,0,5,120,4,21,2,188,4,48,48,48,49,253,88,29] - 80
2024-01-08T22:33:08.683Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - incomingMsg - {"groupid":0,"clusterid":64704,"srcaddr":22781,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":120,"securityuse":0,"timestamp":5435292,"transseqnumber":0,"len":21,"data":{"type":"Buffer","data":[28,95,17,29,10,241,255,65,12,0,5,120,4,21,2,188,4,48,48,48,49]}}
2024-01-08T22:33:08.684Z zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":0,"manufacturerSpecific":true,"direction":1,"disableDefaultResponse":true,"reservedBits":0},"transactionSequenceNumber":29,"manufacturerCode":4447,"commandIdentifier":10},"Payload":[{"attrId":65521,"dataType":65,"attrData":{"type":"Buffer","data":[0,5,120,4,21,2,188,4,48,48,48,49]}}],"Command":{"ID":10,"name":"report","parameters":[{"name":"attrId","type":33},{"name":"dataType","type":32},{"name":"attrData","type":1000}]}},"address":22781,"endpoint":1,"linkquality":120,"groupID":0,"wasBroadcast":false,"destinationEndpoint":1}'
2024-01-08T22:33:08.684Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
Zigbee2MQTT:debug 2024-01-08 23:33:08: Received Zigbee message from '0x54ef441000458a71', type 'attributeReport', cluster 'aqaraOpple', data '{"65521":{"data":[0,5,120,4,21,2,188,4,48,48,48,49],"type":"Buffer"}}' from endpoint 1 with groupID 0
Zigbee2MQTT:info  2024-01-08 23:33:08: MQTT publish: topic 'zigbee2mqtt/0x54ef441000458a71', payload '{"child_lock":null,"error":false,"feed":"","feeding_size":1,"feeding_source":"schedule","led_indicator":null,"linkquality":120,"mode":"schedule","portion_weight":1,"portions_per_day":6,"schedule":[{"days":"everyday","hour":23,"minute":33,"size":1}],"serving_size":1,"update":{"installed_version":-1,"latest_version":-1,"state":null},"update_available":null,"weight_per_day":13}'
2024-01-08T22:33:08.747Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,39,68,129,0,0,192,252,253,88,1,1,0,116,0,95,255,82,0,0,19,28,95,17,30,10,241,255,65,10,0,5,121,13,104,0,85,2,0,7,253,88,29,28]
2024-01-08T22:33:08.747Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,39,68,129,0,0,192,252,253,88,1,1,0,116,0,95,255,82,0,0,19,28,95,17,30,10,241,255,65,10,0,5,121,13,104,0,85,2,0,7,253,88,29,28]
2024-01-08T22:33:08.747Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 39 - 2 - 4 - 129 - [0,0,192,252,253,88,1,1,0,116,0,95,255,82,0,0,19,28,95,17,30,10,241,255,65,10,0,5,121,13,104,0,85,2,0,7,253,88,29] - 28
2024-01-08T22:33:08.747Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - incomingMsg - {"groupid":0,"clusterid":64704,"srcaddr":22781,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":116,"securityuse":0,"timestamp":5439327,"transseqnumber":0,"len":19,"data":{"type":"Buffer","data":[28,95,17,30,10,241,255,65,10,0,5,121,13,104,0,85,2,0,7]}}
2024-01-08T22:33:08.748Z zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":0,"manufacturerSpecific":true,"direction":1,"disableDefaultResponse":true,"reservedBits":0},"transactionSequenceNumber":30,"manufacturerCode":4447,"commandIdentifier":10},"Payload":[{"attrId":65521,"dataType":65,"attrData":{"type":"Buffer","data":[0,5,121,13,104,0,85,2,0,7]}}],"Command":{"ID":10,"name":"report","parameters":[{"name":"attrId","type":33},{"name":"dataType","type":32},{"name":"attrData","type":1000}]}},"address":22781,"endpoint":1,"linkquality":116,"groupID":0,"wasBroadcast":false,"destinationEndpoint":1}'
2024-01-08T22:33:08.750Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
Zigbee2MQTT:debug 2024-01-08 23:33:08: Received Zigbee message from '0x54ef441000458a71', type 'attributeReport', cluster 'aqaraOpple', data '{"65521":{"data":[0,5,121,13,104,0,85,2,0,7],"type":"Buffer"}}' from endpoint 1 with groupID 0
Zigbee2MQTT:info  2024-01-08 23:33:08: MQTT publish: topic 'zigbee2mqtt/0x54ef441000458a71', payload '{"child_lock":null,"error":false,"feed":"","feeding_size":1,"feeding_source":"schedule","led_indicator":null,"linkquality":116,"mode":"schedule","portion_weight":1,"portions_per_day":7,"schedule":[{"days":"everyday","hour":23,"minute":33,"size":1}],"serving_size":1,"update":{"installed_version":-1,"latest_version":-1,"state":null},"update_available":null,"weight_per_day":13}'
2024-01-08T22:33:08.780Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,41,68,129,0,0,192,252,253,88,1,1,0,120,0,72,7,83,0,0,21,28,95,17,31,10,241,255,65,12,0,5,122,13,105,0,85,4,0,0,0,14,253,88,29,252]
2024-01-08T22:33:08.780Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,41,68,129,0,0,192,252,253,88,1,1,0,120,0,72,7,83,0,0,21,28,95,17,31,10,241,255,65,12,0,5,122,13,105,0,85,4,0,0,0,14,253,88,29,252]
[... cut ...] r":null,"linkquality":116,"mode":"schedule","portion_weight":1,"portions_per_day":7,"schedule":[{"days":"everyday","hour":23,"minute":33,"size":1}],"serving_size":1,"update":{"installed_version":-1,"latest_version":-1,"state":null},"update_available":null,"weight_per_day":14}'
```

Also, no crash happening anymore.
My assumption is that the crash only happens with old firmware on the zigbee stick. At least I could reproduce it anymore after the zigbee firmware update on the sonoff 3.0 P variant.

ajfriesen, 2024-01-08:
I could not crash the zigbee addon anymore after the firmware update from the sonoff 3.0 p variant stick.

However, when deleting the last schedule I can reliably create an exception. This only happens when you have 1 schedule defined and remove that one schedule. It does not happen when you go from 2 to 1 schedules:
```
Zigbee2MQTT:debug 2024-01-08 23:37:58: Received MQTT message on 'zigbee2mqtt/0x54ef441000458a71/set' with data '{"schedule":[]}'
Zigbee2MQTT:debug 2024-01-08 23:37:58: Publishing 'set' 'schedule' to '0x54ef441000458a71'
2024-01-08T22:37:58.462Z zigbee-herdsman:controller:endpoint Write 0x54ef441000458a71/1 aqaraOpple({"65521":{"value":{"type":"Buffer","data":[0,2,43,8,0,8,200,1,0]},"type":65}}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":4447,"transactionSequenceNumber":null,"writeUndiv":false})
2024-01-08T22:37:58.462Z zigbee-herdsman:adapter:zStack:adapter sendZclFrameToEndpointInternal 0x54ef441000458a71:22781/1 (0,0,1)
2024-01-08T22:37:58.463Z zigbee-herdsman:adapter:zStack:znp:SREQ --> AF - dataRequest - {"dstaddr":22781,"destendpoint":1,"srcendpoint":1,"clusterid":64704,"transid":222,"options":0,"radius":30,"len":18,"data":{"type":"Buffer","data":[20,95,17,183,2,241,255,65,9,0,2,43,8,0,8,200,1,0]}}
2024-01-08T22:37:58.463Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,28,36,1,253,88,1,1,192,252,222,0,30,18,20,95,17,183,2,241,255,65,9,0,2,43,8,0,8,200,1,0,59]
Zigbee2MQTT:info  2024-01-08 23:37:58: MQTT publish: topic 'zigbee2mqtt/0x54ef441000458a71', payload '{"child_lock":null,"error":false,"feed":"","feeding_size":1,"feeding_source":"schedule","led_indicator":"OFF","linkquality":123,"mode":"manual","portion_weight":1,"portions_per_day":7,"schedule":[],"serving_size":1,"update":{"installed_version":-1,"latest_version":-1,"state":null},"update_available":null,"weight_per_day":14}'
2024-01-08T22:37:58.481Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,100,1,0,100]
2024-01-08T22:37:58.481Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,100,1,0,100]
2024-01-08T22:37:58.481Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 4 - 1 - [0] - 100
2024-01-08T22:37:58.481Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- AF - dataRequest - {"status":0}
2024-01-08T22:37:58.481Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2024-01-08T22:37:58.486Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,3,68,128,0,1,222,24]
2024-01-08T22:37:58.486Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,68,128,0,1,222,24]
2024-01-08T22:37:58.486Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 2 - 4 - 128 - [0,1,222] - 24
2024-01-08T22:37:58.486Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - dataConfirm - {"status":0,"endpoint":1,"transid":222}
2024-01-08T22:37:58.487Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2024-01-08T22:37:58.567Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,26,68,129,0,0,192,252,253,88,1,1,0,123,0,86,100,119,0,0,6,28,95,17,183,4,0,253,88,29,39]
2024-01-08T22:37:58.567Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,26,68,129,0,0,192,252,253,88,1,1,0,123,0,86,100,119,0,0,6,28,95,17,183,4,0,253,88,29,39]
2024-01-08T22:37:58.567Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 26 - 2 - 4 - 129 - [0,0,192,252,253,88,1,1,0,123,0,86,100,119,0,0,6,28,95,17,183,4,0,253,88,29] - 39
2024-01-08T22:37:58.568Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - incomingMsg - {"groupid":0,"clusterid":64704,"srcaddr":22781,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":123,"securityuse":0,"timestamp":7824470,"transseqnumber":0,"len":6,"data":{"type":"Buffer","data":[28,95,17,183,4,0]}}
2024-01-08T22:37:58.568Z zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":0,"manufacturerSpecific":true,"direction":1,"disableDefaultResponse":true,"reservedBits":0},"transactionSequenceNumber":183,"manufacturerCode":4447,"commandIdentifier":4},"Payload":[{"status":0}],"Command":{"ID":4,"name":"writeRsp","parameters":[{"name":"status","type":32},{"name":"attrId","type":33,"conditions":[{"type":"statusNotEquals","value":0}]}]}},"address":22781,"endpoint":1,"linkquality":123,"groupID":0,"wasBroadcast":false,"destinationEndpoint":1}'
2024-01-08T22:37:58.569Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2024-01-08T22:37:58.624Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,39,68,129,0,0,192,252,253,88,1,1,0,123,0,198,113,119,0,0,19,28,95,17,40,10,241,255,65,10,0,5,43,8,0,8,200,2,47,47,253,88,29,186]
2024-01-08T22:37:58.624Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,39,68,129,0,0,192,252,253,88,1,1,0,123,0,198,113,119,0,0,19,28,95,17,40,10,241,255,65,10,0,5,43,8,0,8,200,2,47,47,253,88,29,186]
2024-01-08T22:37:58.624Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 39 - 2 - 4 - 129 - [0,0,192,252,253,88,1,1,0,123,0,198,113,119,0,0,19,28,95,17,40,10,241,255,65,10,0,5,43,8,0,8,200,2,47,47,253,88,29] - 186
2024-01-08T22:37:58.624Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - incomingMsg - {"groupid":0,"clusterid":64704,"srcaddr":22781,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":123,"securityuse":0,"timestamp":7827910,"transseqnumber":0,"len":19,"data":{"type":"Buffer","data":[28,95,17,40,10,241,255,65,10,0,5,43,8,0,8,2 [... cut ...]
    at /app/node_modules/zigbee-herdsman-converters/src/devices/xiaomi.ts:240:47
    at Array.forEach (<anonymous>)
    at /app/node_modules/zigbee-herdsman-converters/src/devices/xiaomi.ts:237:33
    at Array.forEach (<anonymous>)
    at Object.convert (/app/node_modules/zigbee-herdsman-converters/src/devices/xiaomi.ts:206:38)
    at Receive.onDeviceMessage (/app/lib/extension/receive.ts:147:51)
    at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:167:23)
    at EventEmitter.emit (node:events:529:35)
    at EventBus.emitDeviceMessage (/app/lib/eventBus.ts:101:22)
```

jeells102, 2024-01-09:
> @jeells102 can you try the 20221226 firmware?

Is this for the co-ordinator you're talking about?

Koenkk, 2024-01-10:
@ajfriesen can you provide the debug log when going from 2 -> 1 schedules?

ajfriesen, 2024-01-10:
> @ajfriesen can you provide the debug log when going from 2 -> 1 schedules?

@Koenkk 
sure thing, here are the logs:

```
Zigbee2MQTT:debug 2024-01-10 22:49:31: Saving state to file /config/zigbee2mqtt/state.json
Zigbee2MQTT:debug 2024-01-10 22:49:35: Received MQTT message on 'zigbee2mqtt/0x54ef441000458a71/set' with data '{"schedule":[{"days":"everyday","hour":1,"minute":1,"size":1}]}'
Zigbee2MQTT:debug 2024-01-10 22:49:35: Publishing 'set' 'schedule' to '0x54ef441000458a71'
2024-01-10T21:49:35.990Z zigbee-herdsman:controller:endpoint Write 0x54ef441000458a71/1 aqaraOpple({"65521":{"value":{"type":"Buffer","data":[0,2,9,8,0,8,200,11,55,102,48,49,48,49,48,49,48,48,0]},"type":65}}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":4447,"transactionSequenceNumber":null,"writeUndiv":false})
2024-01-10T21:49:35.991Z zigbee-herdsman:adapter:zStack:adapter sendZclFrameToEndpointInternal 0x54ef441000458a71:22781/1 (0,0,1)
2024-01-10T21:49:35.994Z zigbee-herdsman:adapter:zStack:znp:SREQ --> AF - dataRequest - {"dstaddr":22781,"destendpoint":1,"srcendpoint":1,"clusterid":64704,"transid":12,"options":0,"radius":30,"len":28,"data":{"type":"Buffer","data":[20,95,17,7,2,241,255,65,19,0,2,9,8,0,8,200,11,55,102,48,49,48,49,48,49,48,48,0]}}
2024-01-10T21:49:35.995Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,38,36,1,253,88,1,1,192,252,12,0,30,28,20,95,17,7,2,241,255,65,19,0,2,9,8,0,8,200,11,55,102,48,49,48,49,48,49,48,48,0,15]
Zigbee2MQTT:info  2024-01-10 22:49:35: MQTT publish: topic 'zigbee2mqtt/0x54ef441000458a71', payload '{"child_lock":"UNLOCK","error":false,"feed":"","feeding_size":1,"feeding_source":"schedule","led_indicator":"OFF","linkquality":127,"mode":"manual","portion_weight":1,"portions_per_day":0,"schedule":[{"days":"everyday","hour":1,"minute":1,"size":1}],"serving_size":1,"update":{"installed_version":-1,"latest_version":-1,"state":null},"update_available":null,"weight_per_day":0}'
2024-01-10T21:49:36.014Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,100,1,0,100]
2024-01-10T21:49:36.015Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,100,1,0,100]
2024-01-10T21:49:36.015Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 4 - 1 - [0] - 100
2024-01-10T21:49:36.015Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- AF - dataRequest - {"status":0}
2024-01-10T21:49:36.015Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2024-01-10T21:49:36.020Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,3,68,128,0,1,12,202]
2024-01-10T21:49:36.020Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,68,128,0,1,12,202]
2024-01-10T21:49:36.020Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 2 - 4 - 128 - [0,1,12] - 202
2024-01-10T21:49:36.021Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - dataConfirm - {"status":0,"endpoint":1,"transid":12}
2024-01-10T21:49:36.021Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2024-01-10T21:49:36.145Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,3,69,196,253,88,0,39,254,26,68,129,0,0,192,252,253,88,1,1,0,127,0,154,146,102,0,0,6,28,95,17,7,4,0,253,88,29,184]
2024-01-10T21:49:36.145Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,69,196,253,88,0,39,254,26,68,129,0,0,192,252,253,88,1,1,0,127,0,154,146,102,0,0,6,28,95,17,7,4,0,253,88,29,184]
2024-01-10T21:49:36.145Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 2 - 5 - 196 - [253,88,0] - 39
2024-01-10T21:49:36.145Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - srcRtgInd - {"dstaddr":22781,"relaycount":0,"relaylist":[]}
2024-01-10T21:49:36.145Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,26,68,129,0,0,192,252,253,88,1,1,0,127,0,154,146,102,0,0,6,28,95,17,7,4,0,253,88,29,184]
2024-01-10T21:49:36.145Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 26 - 2 - 4 - 129 - [0,0,192,252,253,88,1,1,0,127,0,154,146,102,0,0,6,28,95,17,7,4,0,253,88,29] - 184
2024-01-10T21:49:36.145Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - incomingMsg - {"groupid":0,"clusterid":64704,"srcaddr":22781,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":127,"securityuse":0,"timestamp":6722202,"transseqnumber":0,"len":6,"data":{"type":"Buffer","data":[28,95,17,7,4,0]}}
2024-01-10T21:49:36.146Z zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":0,"manufacturerSpecific":true,"direction":1,"disableDefaultResponse":true,"reservedBits":0},"transactionSequenceNumber":7,"manufacturerCode":4447,"commandIdentifier":4},"Payload":[{"status":0}],"Command":{"ID":4,"name":"writeRsp","parameters":[{"name":"status","type":32},{"name":"attrId","type":33,"conditions":[{"type":"statusNotEquals","value":0}]}]}},"address":22781,"endpoint":1,"linkquality":127,"groupID":0,"wasBroadcast":false,"destinationEndpoint":1}'
2024-01-10T21:49:36.146Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2024-01-10T21:49:36.200Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,47,68,129,0,0,192,252,253,88,1,1,0,127,0,82,158,102,0,0,27,28,95,17,65,10,241,255,65,18,0,5,9,8,0,8,200,10,55,70,48,49,48,49,48,49,48,48,253,88,29,255]
2024-01-10T21:49:36.200Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,47,68,129,0,0,192,252,253,88,1,1,0,127,0,82,158,102,0,0,27,28,95,17,65,10,241,255,65,18,0,5,9,8,0,8,200,10,55,70,48,49, [... cut ...] ays":"everyday","hour":1,"minute":1,"size":1}],"serving_size":1,"update":{"installed_version":-1,"latest_version":-1,"state":null},"update_available":null,"weight_per_day":0}'
```

On a side note:

Could it be, that we have different firmware running on that device?

I remember having the option to set the schedule in home assistant. Now this is just a sensor with (i believe) a json list.
After a OTA update I can not do that in home assistant anymore, but still can do that via the Z2M interface.

Koenkk, 2024-01-13:
Fixed the issue with the parsing!

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

> Could it be, that we have different firmware running on that device?

I don't have this device.

jeells102, 2024-01-13:
Once this goes live in production @Koenkk I'll test my cat feeders and see if it resolves my crashing issue :) Thanks for the hard work to look at the parsing issue.

closed 2024-08-11 by github-actions[bot]