Koenkk/zigbee2mqtt #20132: [New device support]: Siemens RDZ100ZB

JFFilip, 2023-12-09:
### Link

https://hit.sbt.siemens.com/RWD/app.aspx?RC=HQEU&lang=en&MODULE=Catalog&ACTION=ShowProduct&KEY=S55772-T113

### Database entry

{"id":20,"type":"EndDevice","ieeeAddr":"0x0080e1250167790e","nwkAddr":61017,"manufId":4311,"manufName":"Siemens","powerSource":"Mains (single phase)","modelId":"RDZ100","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":769,"inClusterList":[0,3,513,1],"outClusterList":[16,25],"clusters":{"genPowerCfg":{"attributes":{"144":0,"65534":1,"batteryPercentageRemaining":192}},"hvacThermostat":{"attributes":{"80":0,"81":0,"82":0,"83":4,"84":0,"87":1,"88":0,"65534":1,"localTemp":2300,"pIHeatingDemand":0,"localTemperatureCalibration":5,"occupiedHeatingSetpoint":1800,"unoccupiedHeatingSetpoint":2000,"minHeatSetpointLimit":500,"maxHeatSetpointLimit":3000,"ctrlSeqeOfOper":2,"systemMode":4,"remoteSensing":-32768}},"genBasic":{"attributes":{"powerSource":1}}},"binds":[{"cluster":1,"type":"endpoint","deviceIeeeAddress":"0xe0798dfffe9986d0","endpointID":1},{"cluster":513,"type":"endpoint","deviceIeeeAddress":"0xe0798dfffe9986d0","endpointID":1},{"cluster":0,"type":"endpoint","deviceIeeeAddress":"0xe0798dfffe9986d0","endpointID":1}],"configuredReportings":[{"cluster":1,"attrId":33,"minRepIntval":3600,"maxRepIntval":65000,"repChange":0,"manufacturerCode":null},{"cluster":513,"attrId":0,"minRepIntval":0,"maxRepIntval":3600,"repChange":10,"manufacturerCode":null},{"cluster":513,"attrId":18,"minRepIntval":0,"maxRepIntval":3600,"repChange":10,"manufacturerCode":null},{"cluster":513,"attrId":20,"minRepIntval":0,"maxRepIntval":3600,"repChange":10,"manufacturerCode":null},{"cluster":513,"attrId":8,"minRepIntval":0,"maxRepIntval":3600,"repChange":10,"manufacturerCode":null}],"meta":{}}},"appVersion":1,"stackVersion":1,"hwVersion":1,"dateCode":"010000","swBuildId":"0.00.04","zclVersion":1,"interviewCompleted":true,"meta":{"configured":-1688112927},"lastSeen":1702162086155,"defaultSendRequestWhen":"immediate"}

### Comments

Hello, I would like to submit request for new device: Siemens RDZ100ZB smart thermostat.

I tried it on my own, but partly successful. I am reading basic battery and thermostat properties, but submiting new `occupied_heating_setpoint` was sometimes successful and sometimes zigbee returns timeout error.

Error change request:

```
Error 2023-12-10 00:27:49Publish 'set' 'occupied_heating_setpoint' to 'Siemens RDZ100ZB' failed: 'Error: Write 0x0080e1250167790e/1 hvacThermostat({"occupiedHeatingSetpoint":1800}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 61017 - 1 - 139 - 513 - 4 after 10000ms)'
```

Some communication snapshot:

```
Debug 2023-12-09 23:31:02Received Zigbee message from 'Siemens RDZ100ZB', type 'attributeReport', cluster 'hvacThermostat', data '{"65534":1,"88":0}' from endpoint 1 with groupID 0
Debug 2023-12-09 23:31:31Received Zigbee message from 'Siemens RDZ100ZB', type 'attributeReport', cluster 'genPowerCfg', data '{"144":0,"65534":1}' from endpoint 1 with groupID 0
Debug 2023-12-09 23:32:43Received Zigbee message from 'Siemens RDZ100ZB', type 'attributeReport', cluster 'hvacThermostat', data '{"65534":1,"80":0,"81":0,"82":0,"83":4,"84":0,"87":1,"ctrlSeqeOfOper":2,"maxHeatSetpointLimit":3000,"minHeatSetpointLimit":500,"remoteSensing":-32768}' from endpoint 1 with groupID 0
Info 2023-12-09 23:32:43MQTT publish: topic 'zigbee2mqtt/Siemens RDZ100ZB', payload '{"88":1,"battery":96,"control_sequence_of_operation":"heating_only","linkquality":216,"local_temperature":23,"local_temperature_calibration":0.5,"max_heat_setpoint_limit":30,"min_heat_setpoint_limit":5,"occupancy":false,"occupied_heating_setpoint":18,"pi_heating_demand":0,"remote_sensing":{"local_temperature":"internally","occupancy":"internally","outdoor_temperature":"internally"},"system_mode":"heat","unoccupied_heating_setpoint":20}'
Debug 2023-12-09 23:33:02Received Zigbee message from 'Siemens RDZ100ZB', type 'attributeReport', cluster 'hvacThermostat', data '{"65534":1,"88":1}' from endpoint 1 with groupID 0
```

Device docs: https://sid.siemens.com/r/A6V13360586/21494412811_22237829771__en-US_21535228683

### External converter

```shell
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const constants = require('zigbee-herdsman-converters/lib/constants');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

/**
 * @type {import('zigbee-herdsman-converters').Definition}
 */
const definition = {
  zigbeeModel: ['RDZ100'],
  model: 'RDZ100',
  vendor: 'Siemens',
  description: 'Thermostat Siemens Connected Home RDZ100ZB',
  fromZigbee: [
    fz.battery,
    fz.thermostat,
  ],
  toZigbee: [
    tz.thermostat_local_temperature,
    tz.thermostat_local_temperature_calibration,
    tz.thermostat_remote_sensing,
    tz.thermostat_system_mode,
    tz.thermostat_occupied_heating_setpoint,
    tz.thermostat_unoccupied_heating_setpoint,
    tz.thermostat_pi_heating_demand,
  ],
  exposes: [
    e.battery(),
    e.climate()
      .withSetpoint('occupied_heating_setpoint', 5, 30, 0.5)
      .withSetpoint('unoccupied_heating_setpoint', 5, 30, 0.5)
      .withSystemMode(['off', 'heat'])
      .withLocalTemperature()
      .withLocalTemperatureCalibration(-2.5, 2.5, 0.5)
      .withControlSequenceOfOperation(['heating_only'])
      .withPiHeatingDemand(),
  ],
  configure: async (device, coordinatorEndpoint, logger) => {
    const endpoint = device.getEndpoint(1);
    const clusters = ['genBasic', 'genPowerCfg', 'hvacThermostat'];
    await reporting.bind(endpoint, coordinatorEndpoint, clusters);

    await reporting.batteryPercentageRemaining(endpoint);
    await reporting.thermostatTemperature(endpoint);
    await reporting.thermostatOccupiedHeatingSetpoint(endpoint);
    await reporting.thermostatUnoccupiedHeatingSetpoint(endpoint);
    await reporting.thermostatPIHeatingDemand(endpoint);
  },
};

module.exports = definition;
```

devkot1, 2023-12-17:
These new devices from Siemens look promising. Please keep us updated

JFFilip, 2023-12-18:
For anyone wondering how my convertor works, I am using it with home assistant for about a week with no issues. Errors was in my opinion tied to rapid temperature change I did during testing.

devkot1, 2023-12-18:
> For anyone wondering how my convertor works, I am using it with home assistant for about a week with no issues. Errors was in my opinion tied to rapid temperature change I did during testing.

Good to hear! I am planning to buy and use this thermostat with my sonoff zigbee 3.0 usb -e dongle in HA. Did you encounter any problems when pair the device? Also, is there a way to select TPI mode (slow/medium/fast) or desired 2-position via zigbee2mqtt?

JFFilip, 2023-12-18:
`I am planning to buy and use this thermostat with my sonoff zigbee 3.0 usb -e dongle in HA` - I have same dongle and setup.
`Did you encounter any problems when pair the device?` - Pairing device was without problem.
`Also, is there a way to select TPI mode (slow/medium/fast) or desired 2-position via zigbee2mqtt?` - No, the converter does not cover this yet. My guess is that it uses default `TPI slow` mode (it is noted in [docs](https://sid.siemens.com/r/A6V13360586/21494412811_22237829771__en-US_21535228683) as default value).

From docs it is definitely possible to change these settings over zigbee (it is even mandatory, because you can not change these on device by hand), but siemens have it hidden under custom keys in communication, so it is impossible to decifer it without help.

devkot1, 2023-12-20:
Since there are five modes of operation (Control behavior: 2-Pos 1K, 2-Pos 0.3K, TPI Slow, TPI Medium, TPI fast) they could be enumerated as 0,1,2,3 and 4. The default, TPI Slow, is third in the list (2) which is the value of "ctrlSeqeOfOper" in your trace. Just a wild guess...

devkot1, 2024-01-19:
> For anyone wondering how my convertor works, I am using it with home assistant for about a week with no issues. Errors was in my opinion tied to rapid temperature change I did during testing.

Hi, I am using the same model for a month now via HA and it works fine. The only issue that I have is that every about 2 weeks the display becomes unresponsive, while the thermostat operates fine via zigbee. Removing and re-inserting the batteries fixes the display. Have you encountered a similar issue? The firmware version (sw_build_id) is 0.00.04.

JFFilip, 2024-01-20:
Hi, no, I do not, but I barely touch the device, mostly change values remotely.

VladislavVesely, 2024-05-21:
Any progress on this topic?

Konyagii, 2024-09-11:
Is this device supported yet?

closed 2025-01-09 by github-actions[bot]

variety-group, 2025-01-12:
Hi and thank you for your work here. I'm using the same setup as you, with same dongle and using External converter above, but I have a lot of issues with timeout errors, exactly the same as mentioned in first post. I don't think it's because of too many temperture changes in my case.

Sometime submitting new occupied_heating_setpoint is fine, but mostly timeout error, so my scheduled tasks are not stable, and also changing values from termostat dashboard card is almost impossible.

Did you guys maybe make some updates or improvements with RDZ100 integration here, or found anything related to this issue?

Thank you a lot!

VeselyVladislav, 2025-01-12:
> Hi and thank you for your work here. I'm using the same setup as you, with same dongle and using External converter above, but I have a lot of issues with timeout errors, exactly the same as [...]

I implemented this one, but both queue as well as timeout does not seems to be doing much. It kind of work ;D
`const {identify, battery, ota} = require('zigbee-herdsman-converters/lib/modernExtend');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting  = require('zigbee-herdsman-converters/lib/reporting');
const constants  = require('zigbee-herdsman-converters/lib/constants');
const utils  = require('zigbee-herdsman-converters/lib/utils');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const {precisionRound, postfixWithEndpointName} = require('zigbee-herdsman-converters/lib/utils');
const e = exposes.presets;
const ea = exposes.access;

// We can only read and write after the device will post something. This particual device is sending messages one per minute.
// Also metadata from main object does not seems to be working at the time of implementing this device, so all values are defined here again.
const siemensTimeoutMs = 300000
const readWriteMeta = {sendPolicy: 'queue'};

const siemensOperationModes = {
    0: '2-position, 1 K',
    1: '2-position, 0.3 K',
    2: 'TPI slow',
    3: 'TPI medium',
    4: 'TPI fast',
};

const siemensFromZigbee = {
    cluster: 'hvacThermostat',
    type: ['attributeReport', 'readResponse'],
    convert: (model, msg, publish, options, meta) => {
	var result = {};
        if (msg.data.hasOwnProperty('localTemperatureCalibration')) {
            result[postfixWithEndpointName('local_temperature_calibration', msg, model, meta)] =
                precisionRound(msg.data['localTemperatureCalibration'], 2) / 10;
        }
	if (msg.data.hasOwnProperty('localTemp')) {
            const value = precisionRound(msg.data['localTemp'], 2) / 100;
            if (value >= -273.15) {
                result[postfixWithEndpointName('local_temperature', msg, model, meta)] = value;
            }
        }
        if (msg.data.hasOwnProperty('occupiedHeatingSetpoint')) {
            const value = precisionRound(msg.data['occupiedHeatingSetpoint'], 2) / 100;
            // Stelpro will return -325.65 when set to off, value is not realistic anyway
            if (value >= -273.15) {
                result[postfixWithEndpointName('occupied_heating_setpoint', msg, model, meta)] = value;
            }
        }
        if (msg.data.hasOwnProperty('unoccupiedHeatingSetpoint')) {
            result[postfixWithEndpointName('unoccupied_heating_setpoint', msg, model, meta)] =
                precisionRound(msg.data['unoccupiedHeatingSetpoint'], 2) / 100;
        }
        if (msg.data.hasOwnProperty('systemMode')) {
            result[postfixWithEndpointName('system_mode', msg, model, meta)] = constants.thermostatSystemModes[msg.data['systemMode']];
        }
        if (msg.data.hasOwnProperty('ctrlSeqeOfOper')) {
            result[postfixWithEndpointName('control_sequence_of_operation', msg, model, meta)] =
                siemensOperationModes[msg.data['ctrlSeqeOfOper']];
        }

	return result;
    },
};

const tz_local_temperature = {
    key: ['local_temperature'],
    convertGet: async (entity, key, meta) => {
	await entity.read('hvacThermostat', ['localTemp'], readWriteMeta);
    },
};
[... 103 lines cut ...]
    exposes: (device, options) => {
	const expose = [
	e.climate()
	.withLocalTemperature()
	.withLocalTemperatureCalibration(-2.5, 2.5, 0.5)
	.withSetpoint('occupied_heating_setpoint', 5, 30, 0.5)
	.withSetpoint('unoccupied_heating_setpoint', 5, 30, 0.5)
        .withSystemMode(['off', 'heat'])];
    
	expose.push(e.enum('control_sequence_of_operation', ea.STATE_SET, Object.values(siemensOperationModes)).withDescription('Operation mode'));

	return expose;
    },
    meta: {sendPolicy: 'queue'},
    configure: async (device, coordinatorEndpoint) => {
	device.pendingRequestTimeout = siemensTimeoutMs;
        const endpoint = device.getEndpoint(1);
        await reporting.bind(endpoint, coordinatorEndpoint, ['genPowerCfg', 'hvacThermostat']);
        await reporting.thermostatTemperature(endpoint);
        await reporting.thermostatOccupiedHeatingSetpoint(endpoint);
        await reporting.thermostatUnoccupiedHeatingSetpoint(endpoint);
    },
};

module.exports = definition;`

variety-group, 2025-01-13:
So it seems, that my timeout issue was caused by wrong Zigbee router connection (Sieems like Siemens is using always the one used during Pairing).

I'm now using external converter provided by @VladislavVesely and it is great! Thank you for sharing and for this work!

VladislavVesely, 2025-01-13:
> So it seems, that my timeout issue was caused by wrong Zigbee router connection (Sieems like Siemens is using always the one used during Pairing).
> 
> I'm now using external converter provided by @VladislavVesely and it is great! Thank you for sharing and for this work!

I was uploading that from phone last night, so it the formatting went wrong :D Here it is again for anyone who want to try their luck :D Just rename it to ".js" since comments does not allow to upload js files directly.
[RDZ100ZB.txt](https://github.com/user-attachments/files/18394457/RDZ100ZB.txt)

nefh, 2025-02-05:
Hi! I was about to buy Siemens RDZ100ZB, to upgrade my 30 years old Siemens thermostat, which have its problems, but I see it is not on supported list, it was before though (https://github.com/Koenkk/zigbee2mqtt.io/commit/e9118084e2a17e271cfd5945c80c2c6341a4d7fa). 

What is the current state? Topic flair is "not planned" is there technical problem? Or is it only this unresponsive display mentioned before?

Notis21, 2025-04-30:
Hello, any news for this thermostat? 
Should I buy it for home heating with zigbee conection to homeassistant?

devkot1, 2025-04-30:
> Hello, any news for this thermostat? Should I buy it for home heating with zigbee conection to homeassistant?

It does work with HA + ZHA as a generic thermostat. Since it is a battery powered device, sometimes it is slow to respond. The TPI algorithm works well in my setting.

Notis21, 2025-04-30:
> > Hello, any news for this thermostat? Should I buy it for home heating with zigbee conection to homeassistant?
> 
> It does work with HA + ZHA as a generic thermostat. Since it is a battery powered device, sometimes it is slow to respond. The TPI algorithm works well in my setting.

How does respond with zigbee2mqtt?

nefh, 2025-05-01:
Should we start new topic? Since this one is tagged by bot as "not-planned", this thread is not even visible under issues.

And I still consider upgrade if it would work with zigbee2mqtt.

Notis21, 2025-05-01:
> Should we start new topic? Since this one is tagged by bot as "not-planned", this thread is not even visible under issues and nobody out of people in already it see new responses.
> 
> I still consider upgrade if it would work with zigbee2mqtt, I'll be glad to know what is current state.

Hello!how to start new topic? This thermostat is very popular in Greece and I am very interested if it will work with HA and zigbee2mqtt.

indyone, 2026-08-25:
Hey @adamspir, basic stuff like reading room temp, adjusting the setpoint, and turning it on/off should work out of the box in ZHA since it uses standard Zigbee HVAC clusters.

The JS code in this thread is strictly for Zigbee2MQTT. ZHA is Python-based, so it can't run JS external converters. Without a custom ZHA Python quirk, you'll miss out on the deeper Siemens settings like switching between the TPI control modes (2-position vs TPI slow/medium/fast). If you only care about basic heating control, ZHA will be fine. If you want full control over the TPI parameters, you'll either need to write a Python quirk for ZHA or stick to Z2M with this converter.

mhostak, 2026-10-26:
Hey.

I have an RDZ100ZB and I’ve got support ready for Zigbee Herdsman converters. Fork: https://github.com/mhostak/zigbee-herdsman-converters/tree/feat/siemens-rdz100zb

I wrote the code with the help of AI, but I’ve been testing it for several weeks now and it works – can I send a PR?

Mapping: the control mode is in attribute 0x0053 (0 = 2-position 1 K, 1 = 2-position 0.3 K, 2 = TPI fast, 3 = TPI medium, 4 = TPI slow). Window detection is 0x0050 and window status is 0x0051. 

Mapping taken from the Siemens Connected Home app, verified on the device (ctrlSeqeOfOper is not the control mode).

After writing 0x0053 = 0, the thermostat operated in two-point control mode with a hysteresis of 1 K for 10 days.

Idaho947, 2026-11-16:
Someone can reopen this issue ?