Koenkk/zigbee2mqtt #20135: Hue motion sensors temp and lux update issue

whcrg, 2023-12-10:
### What happened?

It seem hue motion sensors temperature and luminosity is updated very infrequently. Googling shows some people have updates every 5min, so I think it could be somehow done.

SmartThing driver seems to even have some sort of configuration for update intervals (line 288 on) but I havent yet gone deep enough to understand what it maybe sends to the device.
https://github.com/bogdanalexe90/hueMotionSensor/blob/master/devicetypes/bogdanalexe90/hue-motion-sensor.src/hue-motion-sensor.groovy

Could it be that there is some update interval settings somewhere, and for example the hue bridge sets such when pairing (and they maybe even carry over for people who move sensor from hue bridge to other systems?

### What did you expect to happen?

Update of temp I expect to get more often than every 30-60min. Every 2-5min would be what I suppose should happen.

### How to reproduce it (minimal and precise)

Buy new hue motion sensor, pair to z2m (never been paired to any other system) (I have two, the inside and the outside models)

### Zigbee2MQTT version

1.34.0

### Adapter firmware version

not shown in settings/about

### Adapter

SkyConnect

### Debug log

The valiues keep repeating frequently but not actually updated in long time. Even several degree temperature change does not trigger faster update.

Debug 2023-12-10 11:15:55Received Zigbee message from 'LroomPIR03', type 'attributeReport', cluster 'msOccupancySensing', data '{"occupancy":0}' from endpoint 2 with groupID 0
Info 2023-12-10 11:15:55MQTT publish: topic 'zigbee2mqtt/LroomPIR03', payload '{"battery":100,"illuminance":8183,"illuminance_lux":7,"led_indication":false,"linkquality":164,"motion_sensitivity":"low","occupancy":false,"occupancy_timeout":2,"temperature":21.82}'
Info 2023-12-10 11:15:55MQTT publish: topic 'zigbee2mqtt/LroomPIR03', payload '{"battery":100,"illuminance":8183,"illuminance_lux":7,"led_indication":false,"linkquality":164,"motion_sensitivity":"low","no_occupancy_since":{"undefined":60},"occupancy":false,"occupancy_timeout":2,"temperature":21.82}'

lux73, 2023-12-10:
there are some Aqara Movement Sensors available with integrated Lux measuring - they only update new lx Values if movement is registered

quite not usable - i don't know if the Hue Sensors are working similar ^^

i had to buy dedicated Lux Sensors to handle this Problem...

mundschenk-at, 2023-12-12:
The actual update interval is something like 5 minutes, I believe. This is a known hardware/firmware limitation of the Hue motion sensors.

whcrg, 2023-12-13:
I am getting around 20 updates per 24/h. Motion works perfect and lux is bit more regular and signal strength good. 
[image]

mundschenk-at, 2023-12-13:
The update interval issue I was mentioning is specific to the lux measurement.

whcrg, 2023-12-13:
Seems the reporting is configurable, have to just figure out how.
Hubitats drivers offer such. https://community.hubitat.com/t/philips-hue-motion-sensor-more-frequent-lux-reading/126477/9

whcrg, 2023-12-13:
Maybe it just has it as the standard intends... pity my coding skills are not up to this task
[image]

whcrg, 2023-12-13:
Haa! Its apparently implemented and working in z2m and there is just something weird going on in my individual system and the reporting tab under device is empty and that is why I cannot see the settings.

closed 2023-12-13 by whcrg