Koenkk/zigbee2mqtt #20136: [New device support]:   Tuya 8 Switch _TZE200_wktrysab

rt400, 2023-12-10:
### Link

https://www.aliexpress.com/item/1005005848035933.html?spm=a2g0o.refund_spm.0.0.5f8872e4TczdDt

### Database entry

{}

### Comments

Please add this device support , _TZE200_wktrysab

### External converter

```shell
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const utils = require('zigbee-herdsman-converters/lib/utils');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    fingerprint: tuya.fingerprint('TS0601', ['_TZE200_wktrysab']),
    model: 'WLS098-8GANAG',
    vendor: 'TuYa',
    description: 'Tuya 8 Gang Switch',
    fromZigbee: [tuya.fzDataPoints],
    toZigbee: [tuya.tzDataPoints],
    configure: tuya.configureMagicPacket,
    exposes: [
        e.switch().withEndpoint('l1'), 
        e.switch().withEndpoint('l2'), 
        e.switch().withEndpoint('l3'), 
        e.switch().withEndpoint('l4'), 
        e.switch().withEndpoint('l5'),
        e.switch().withEndpoint('l6'),
        e.switch().withEndpoint('l7'),
        e.switch().withEndpoint('l8')
    ],
    endpoint: (device) => {
        return {'l1': 1, 'l2': 1, 'l3': 1, 'l4': 1, 'l5': 1, 'l6': 1, 'l7': 1, 'l8': 1};
    },
    meta: {
        multiEndpoint: true,
        tuyaDatapoints: [
            [1, 'state_l1', tuya.valueConverter.onOff],
            [2, 'state_l2', tuya.valueConverter.onOff],
            [3, 'state_l3', tuya.valueConverter.onOff],
            [4, 'state_l4', tuya.valueConverter.onOff],
            [5, 'state_l5', tuya.valueConverter.onOff],
            [6, 'state_l6', tuya.valueConverter.onOff],
            [101, 'state_l7', tuya.valueConverter.onOff],
            [102, 'state_l8', tuya.valueConverter.onOff],
        ],
    },
};

module.exports = definition;
```
```

### Supported color modes

no

Koenkk, 2023-12-10:
Added!

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

closed 2023-12-10 by Koenkk

tiket84, 2023-12-11:
Hello, 
I have the same problem. I have installed the Zigbee2Mqtt edge 10 ninutes ago but the 8 gang ligh switch (https://www.aliexpress.com/item/1005005649115393.html?spm=a2g0o.order_list.order_list_main.91.65851802X1ILCV)  is joined but result not supported and when switch on or off I receive wornings "receive message from unsupporteddevice with Zigbee model "TS0601" and manufacturer name "_TZE200_wktrysab" and then "please see: https://www.zigbee2mqtt.io/danced/support-new-devices/01_support_new_devices.html" 
can you help me?

rt400, 2023-12-11:
you need to update to dev branch to get this update...

tiket84, 2023-12-11:
I'm using Home assistant add on so, as instruction,  the last dev update (latest dev) is present in edge version of Zigbee2MQTT https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html#home-assistant-addon 
Am I wrong?