Koenkk/zigbee2mqtt #20139: [UBISYS C4]: no valid firmware description link

Grumpelumpf, 2023-12-10:
### Link

https://www.ubisys.de/en/products/additional-options/control-unit-c4/

### Model

C4

### Description

Control unit C4

### Vendor

ubisys

### Picture (link)

https://www.zigbee2mqtt.io/images/devices/C4.jpg

### Database entry

{"id":17,"type":"Router","ieeeAddr":"0x001fee0000008d26","nwkAddr":13320,"manufId":4338,"manufName":"ubisys","powerSource":"Mains (single phase)","modelId":"C4 (5504)","epList":[1,2,3,4,5,6,232,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":260,"inClusterList":[0,3],"outClusterList":[3,5,6,8,768,64514],"clusters":{"genBasic":{"attributes":{"modelId":"C4 (5504)","manufacturerName":"ubisys","powerSource":1,"zclVersion":8,"appVersion":1,"stackVersion":1,"hwVersion":3,"dateCode":"20220218-DE-FB1","swBuildId":"2.3.0"}}},"binds":[{"cluster":5,"type":"endpoint","deviceIeeeAddress":"0x00124b00237ba189","endpointID":1},{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b00237ba189","endpointID":1},{"cluster":8,"type":"endpoint","deviceIeeeAddress":"0x00124b00237ba189","endpointID":1}],"configuredReportings":[],"meta":{}},"2":{"profId":260,"epId":2,"devId":260,"inClusterList":[0,3],"outClusterList":[3,5,6,8,768,64514],"clusters":{},"binds":[{"cluster":5,"type":"endpoint","deviceIeeeAddress":"0x00124b00237ba189","endpointID":1},{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b00237ba189","endpointID":1},{"cluster":8,"type":"endpoint","deviceIeeeAddress":"0x00124b00237ba189","endpointID":1}],"configuredReportings":[],"meta":{}},"3":{"profId":260,"epId":3,"devId":260,"inClusterList":[0,3],"outClusterList":[3,5,6,8,768,64514],"clusters":{},"binds":[{"cluster":5,"type":"endpoint","deviceIeeeAddress":"0x00124b00237ba189","endpointID":1},{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b00237ba189","endpointID":1},{"cluster":8,"type":"endpoint","deviceIeeeAddress":"0x00124b00237ba189","endpointID":1}],"configuredReportings":[],"meta":{}},"4":{"profId":260,"epId":4,"devId":260,"inClusterList":[0,3],"outClusterList":[3,5,6,8,768,64514],"clusters":{},"binds":[{"cluster":5,"type":"endpoint","deviceIeeeAddress":"0x00124b00237ba189","endpointID":1},{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b00237ba189","endpointID":1},{"cluster":8,"type":"endpoint","deviceIeeeAddress":"0x00124b00237ba189","endpointID":1}],"configuredReportings":[],"meta":{}},"5":{"profId":260,"epId":5,"devId":515,"inClusterList":[0,3],"outClusterList":[3,5,258,64514],"clusters":{},"binds":[{"cluster":5,"type":"endpoint","deviceIeeeAddress":"0x00124b00237ba189","endpointID":1},{"cluster":258,"type":"endpoint","deviceIeeeAddress":"0x00124b00237ba189","endpointID":1}],"configuredReportings":[],"meta":{}},"6":{"profId":260,"epId":6,"devId":515,"inClusterList":[0,3],"outClusterList":[3,5,258,64514],"clusters":{},"binds":[{"cluster":5,"type":"endpoint","deviceIeeeAddress":"0x00124b00237ba189","endpointID":1},{"cluster":258,"type":"endpoint","deviceIeeeAddress":"0x00124b00237ba189","endpointID":1}],"configuredReportings":[],"meta":{}},"232":{"profId":260,"epId":232,"devId":1287,"inClusterList":[0,64512],"outClusterList":[3,25],"clusters":{"manuSpecificUbisysDeviceSetup":{"attributes":{"inputConfigurations":[0,0,0,0],"inputActions":[{"type":"Buffer","data":[0,13,1,6,0,2]},{"type":"Buffer","data":[0,3,1,6,0,2]},{"type":"Buffer","data":[1,13,2,6,0,2]},{"type":"Buffer","data":[2,13,3,6,0,2]},{"type":"Buffer","data":[3,13,4,6,0,2]}]}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":1,"stackVersion":1,"hwVersion":3,"dateCode":"20220218-DE-FB1","swBuildId":"2.3.0","zclVersion":8,"interviewCompleted":true,"meta":{"configured":1},"lastSeen":1702210297131,"defaultSendRequestWhen":"immediate"}

### Notes

the link to the firmware change log is not valid and has to be updated: 
https://www.ubisys.de/en/support/firmware/change-logs-c4/
[image]

Grumpelumpf, 2024-06-08:
Not fixed yet

sjorge, 2024-06-08:
I could not found referenced to the URL in z2m, zh, or zhc.

Seems like the frontend might be getting it from somewhere.

sjorge, 2024-06-08:
https://github.com/nurikk/zigbee2mqtt-frontend/blob/dev/src/components/vendor-links/vendor-links.tsx

OK so try opening an issue on that repo, the URL is generated in that file.

closed 2025-01-09 by github-actions[bot]