Koenkk/zigbee2mqtt #20146: Hue Play Gradient PC LCX006 appears as NOT supported

Rett1, 2023-12-10:
### What happened?

Hello dear community,

I have just received a new [Hue Play Gradient Lightstrip PC](https://www.philips-hue.com/de-de/p/lightstrips-hue-play-gradient-lightstrip-fur-pc/8719514434530). When pairing it, I get a message that it is not supported. 
If I'm not mistaken, it should be [this one](https://www.zigbee2mqtt.io/devices/929003498601.html) that is marked as supported. 

Now I'm wondering if there is a new model out there or if I just do something wrong. 
Any ideas welcome. 

Thanks a lot in advance!

### What did you expect to happen?

Successfully paired and interviewed device available in the Z2M device overview.

### How to reproduce it (minimal and precise)

- Reset Hue Lightstrip with Hue Dimmer Switch as described [here](https://www.zigbee2mqtt.io/devices/929003498601.html#hue-dimmer-switch)
- Start `Permit join (All)`

### Zigbee2MQTT version

1.34.0-1

### Adapter firmware version

7.1.1.0 build 273

### Adapter

Home Assistant SkyConnect (EZSP v9)

### Debug log

Pairing
```
Info 2023-12-10 21:34:31 Device '0x001788010b8a6f80' joined
Info 2023-12-10 21:34:31 MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x001788010b8a6f80","ieee_address":"0x001788010b8a6f80"},"type":"device_joined"}'
Info 2023-12-10 21:34:31 Starting interview of '0x001788010b8a6f80'
Info 2023-12-10 21:34:32 MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x001788010b8a6f80","ieee_address":"0x001788010b8a6f80","status":"started"},"type":"device_interview"}'
Info 2023-12-10 21:34:32 Successfully interviewed '0x001788010b8a6f80', device has successfully been paired
Warning 2023-12-10 21:34:32 Device '0x001788010b8a6f80' with Zigbee model 'LCX006' and manufacturer name 'Signify Netherlands B.V.' is NOT supported, please follow https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html
Info 2023-12-10 21:34:33 MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"definition":null,"friendly_name":"0x001788010b8a6f80","ieee_address":"0x001788010b8a6f80","status":"successful","supported":false},"type":"device_interview"}'
```

When I try to "Reconfigure"
```
Debug 2023-12-10 21:35:43 Received MQTT message on 'zigbee2mqtt/bridge/request/device/configure' with data '{"id":"0x001788010b8a6f80","transaction":"ayacc-2"}'
Info 2023-12-10 21:35:43 MQTT publish: topic 'zigbee2mqtt/bridge/response/device/configure', payload '{"data":{"id":"0x001788010b8a6f80"},"error":"Device '0x001788010b8a6f80' cannot be configured","status":"error","transaction":"ayacc-2"}'
```

Koenkk, 2023-12-11:
Added!

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

closed 2023-12-11 by Koenkk

Rett1, 2023-12-11:
Oh wow, that was quick. 
I can confirm that it is already working using the Dev branch. 
Thank you very much!