Koenkk/zigbee2mqtt #20147: [Wrong device]: TS011F _TZ3000_helyqdvs

Testmangh, 2023-12-10:
### Link

https://pt.aliexpress.com/item/1005004106731547.html?spm=a2g0o.order_list.order_list_main.577.2b36caa4EDLngd&gatewayAdapt=glo2bra

### Model

Its is the same device as Zigbee Model TS011F / Zigbee Manufacturer _TZ3000_6s5dc9lx

### Description

Double wall socket no power, current and energy measuring 

### Vendor

SHAWADER

### Picture (link)

https://pt.aliexpress.com/item/1005004106731547.html?spm=a2g0o.order_list.order_list_main.577.2b36caa4EDLngd&gatewayAdapt=glo2bra

### Database entry

{"id":18,"type":"Router","ieeeAddr":"0xdc8e95fffe5de5d3","nwkAddr":31460,"manufId":4098,"manufName":"_TZ3000_helyqdvs","powerSource":"Mains (single phase)","modelId":"TS011F","epList":[1,2,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":266,"inClusterList":[0,3,4,5,6,57344,57345],"outClusterList":[25,10],"clusters":{"genOnOff":{"attributes":{"32768":0,"onOff":1,"onTime":0,"offWaitTime":0,"tuyaBacklightSwitch":1,"tuyaBacklightMode":1,"moesStartUpOnOff":0}},"manuSpecificTuya_3":{"attributes":{"powerOnBehavior":0}},"genBasic":{"attributes":{"65503":"��\b-\u0011","65506":31,"65508":0,"65534":0,"stackVersion":0,"dateCode":"","manufacturerName":"_TZ3000_helyqdvs","zclVersion":3,"appVersion":67,"modelId":"TS011F","powerSource":1}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b0029e8bf92","endpointID":242},{"cluster":2820,"type":"endpoint","deviceIeeeAddress":"0x00124b0029e8bf92","endpointID":242},{"cluster":1794,"type":"endpoint","deviceIeeeAddress":"0x00124b0029e8bf92","endpointID":242},{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b0029e8bf92","endpointID":1},{"cluster":2820,"type":"endpoint","deviceIeeeAddress":"0x00124b0029e8bf92","endpointID":1},{"cluster":1794,"type":"endpoint","deviceIeeeAddress":"0x00124b0029e8bf92","endpointID":1}],"configuredReportings":[],"meta":{}},"2":{"profId":260,"epId":2,"devId":266,"inClusterList":[3,4,5,6,57344,57345],"outClusterList":[],"clusters":{"genOnOff":{"attributes":{"onOff":0,"onTime":0,"offWaitTime":0}},"manuSpecificTuya_3":{"attributes":{"powerOnBehavior":0}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":67,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1702248585808,"defaultSendRequestWhen":"immediate"}

### Notes

It is supported but I have control for only one socket.

[image]

Testmangh, 2023-12-11:
I added the device in the external converter from #15720 and it is working properly.

`const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const utils = require('zigbee-herdsman-converters/lib/utils');
const globalStore = require('zigbee-herdsman-converters/lib/store');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    fingerprint: [{modelID: 'TS011F', manufacturerName: '_TZ3000_mvn6jl7x'},
        {modelID: 'TS011F', manufacturerName: '_TZ3000_raviyuvk'}, {modelID: 'TS011F', manufacturerName: '_TYZB01_hlla45kx'},
        {modelID: 'TS011F', manufacturerName: '_TZ3000_92qd4sqa'}, {modelID: 'TS011F', manufacturerName: '_TZ3000_zwaadvus'},
        {modelID: 'TS011F', manufacturerName: '_TZ3000_k6fvknrr'}, {modelID: 'TS011F', manufacturerName: '_TZ3000_6s5dc9lx'},
        {modelID: 'TS011F', manufacturerName: '_TZ3000_helyqdvs'}],
    model: 'TS011F_2_gang_wall',
    vendor: 'TuYa',
    description: '2 gang wall outlet',
    extend: tuya.extend.switch({backlightModeLowMediumHigh: true, childLock: true, endpoints: ['l1', 'l2']}),
    whiteLabel: [{vendor: 'ClickSmart+', model: 'CMA30036'}],
    endpoint: (device) => {
        return {'l1': 1, 'l2': 2};
    },
    meta: {multiEndpoint: true},
    configure: tuya.configureMagicPacket,
};

module.exports = definition;` 

[image]

Koenkk, 2023-12-11:
Added!

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

closed 2023-12-11 by Koenkk