Koenkk/zigbee2mqtt #20148: Getting error of 'Adapter disconnected, stopping' every two hours.

MostHated, 2023-12-11:
### What happened?

I am using a UZG-01 device via PoE.  Tthe Z2M docker gives the following error after 1h 59m 55s and does this repeatedly after that amount of time over and over:

I did a continuous port check every 500ms to the UZG-01 on the port Z2M is connecting to, as well as the web interface, my home assistant box, etc, and didn't see the UZG-01's zigbee port go down, nor any of the others I was checking. While the connections were still reporting as up, the following error still occurred and when this happens, often a handful of devices will fall offline.

```sh
error 2023-12-05 15:16:52: Adapter disconnected, stopping
info  2023-12-05 15:16:52: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'offline'
info  2023-12-05 15:16:52: Disconnecting from MQTT server
info  2023-12-05 15:16:52: Stopping zigbee-herdsman...
error 2023-12-05 15:16:52: Failed to stop Zigbee2MQTT
```

### Zigbee2MQTT version

1.34.0-1

### Adapter firmware version

20230507

### Adapter

ZigStar UZG-01 PoE (https://uzg.zig-star.com/product/)

docstalek, 2023-12-13:
I have the same issue. Z2M running on an Alpine container on Proxmox. I have also tried running on an Ubuntu container and as Home Assistant Add-on. All show the same pattern, disconnect after 2 hours.

I have tried both with LAN and Wifi, with same result.

As a part of my debugging process, I tried to open a TCP connection towards the UZG-01 using telnet. After 2 hours, the connection is dropped by the UZG-01 with message "Connection closed by foreign host.".

For me the the problem seems to be on the UZG-01 side, with either the ESP32 firmware or the Z-Stack firmware.

sadabla, 2023-12-14:
I have the same issue. Z2M and Mosquito running on Home Assistant using the SLZB-06 via ethernet. 
This is a new setup and the issue is there since the beginning.

My logfile:
Zigbee2MQTT:error 2023-12-14 11:13:24: Adapter disconnected, stopping
Zigbee2MQTT:info  2023-12-14 11:13:24: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload '{"state":"offline"}'
Zigbee2MQTT:info  2023-12-14 11:13:24: Disconnecting from MQTT server
Zigbee2MQTT:info  2023-12-14 11:13:24: Stopping zigbee-herdsman...
Zigbee2MQTT:error 2023-12-14 11:13:24: Failed to stop Zigbee2MQTT

comazone, 2023-12-18:
I have the same problem. ZigStar lan, every 2 hours disconnected

dellorianes, 2023-12-19:
Same here with a tube-zb-gw-cc2652p2-v2 over ethernet.

comazone, 2023-12-19:
I changing DHCP lease time for IP adresses on my PfSense router and problem is gone. But i dont know why DHCP lease time influences on static IP adress my LAN ZigStar

comazone, 2023-12-19:
Make sure DHCP is disabled in coordinator setup

[image]

MostHated, 2023-12-19:
Oh, really? Interesting. I use PFSense as well (though, that part is probably unrelated). Let me give that a try.

comazone, 2023-12-19:
Try this method. May be it work for you too

MostHated, 2023-12-19:
Yeah, I was just setting a static dhcp mapping in pfsense, as I do for most everything. I set dhcp off on the device now, and put the address in and it is restarting now. I will keep an eye on it the next 2 hours and see how it goes. :+1:

docstalek, 2023-12-19:
I have also changed to static IP now. Let's see if that changes anything 🤞

comazone, 2023-12-19:
On my clock 23:55, last disconnest was 14:46. Problem is gone

MostHated, 2023-12-20:
I am a few moments shy of the 24 hour mark and things have been just fine now. :+1:  Indeed, thanks @comazone

sadabla, 2023-12-22:
I also use pfsense and static mapping in DHCP for the SLZB-06. Setting a static IP address solved the issue. Thanks a lot @comazone!

udivankin, 2023-12-26:
As for me with Hamgeek LAN coordinator built with RF-BM-2652P2 and [MJ-Ethernet01 from ColtSmart](https://www.coltsmart.com/product2_1.html), setting static IP didn't help.

mercenaruss, 2023-12-27:
Guys please open a issue here: https://github.com/mercenaruss/uzg-firmware
We will try to solve the issue.

ashishpandey, 2024-01-03:
While static IP works around this issue, should zigbee2mqtt be resiliant to disconnects, specially when an immediate reconnection within a small window of time works? That should help a lot with stability. Sockets drop, this should not be a fatal error

fliespl, 2024-03-29:
Anyone else is having this problem even with static ip set? I have had a run of 1 month without issues but then I ve had like 3 disconnections over a week where only device restart helped (uzg-01 with p7 chip).

MostHated, 2024-03-29:
Yeah, I thought I was losing my bananas. Ever since I made my last post, things have been working great up to this last month. 
It has been quite frustrating. Multiple devices dropping off-line every day or two. I feel that it has been better than it was before setting static IP, but during the time it was good, it was great, not a single drop off or anything from Dec 19th until around the start of March or somewhere there abouts.

I am not sure if you are using HomeAssistant, and this could be purely coincidental, but it feels this started when/soon after I installed 2024.3.x.

mercenaruss, 2024-03-29:
this is related to p7 chip firmware.

fliespl, 2024-03-29:
@marcusaram can you explain a little bit more? Is this something that might be fixed in the future? If so where - z2m or uzg firmware?

MostHated, 2024-04-02:
I, too, am very much interested in any additional details about that, it was a rather vague statement. Is there something we can do on our end to help with this? Ex: a firmware update or something?

marcusaram, 2024-04-02:
> @marcusaram can you explain a little bit more? Is this something that might be fixed in the future? If so where - z2m or uzg firmware?

Hi, you should probably mention someone else. I did long time ago some small contributions to this project but actually not using zigbee2mqtt anymore. Besides that I've really no idea what is causing this issue :sweat_smile:

MostHated, 2024-04-02:
I am sure he didn't notice and thought he mentioned the name above his. I didn't realize it was wrong either until just now, lol.

fliespl, 2024-04-02:
@MostHated @marcusaram Of course I did mean @mercenaruss... Sorry about confusion, very close nicknames :) 🤦

closed 2024-10-30 by github-actions[bot]