Koenkk/zigbee2mqtt #20153: [New device support]: TS0601, _TZE200_tsxpl0d0 - KnockautX FMD2C018/maybe MOES ZM-105B-MS

thotha, 2023-12-11:
### Link

https://www.brelag.com/shop/fmd2c018-knockautx-dimmaktor-unterputz-2-kanal-4530?page=2&attrib=31-87#attr=

### Database entry

{"id":2,"type":"Router","ieeeAddr":"0x003c84fffe85c2c0","nwkAddr":38456,"manufId":4098,"manufName":"_TZE200_tsxpl0d0","powerSource":"Mains (single phase)","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[0,4,5,61184],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"�}\b-\u0012�}\b-\u0012","65506":31,"65508":0,"65534":0,"modelId":"TS0601","manufacturerName":"_TZE200_tsxpl0d0","stackVersion":0,"dateCode":"","zclVersion":3,"appVersion":68,"powerSource":1}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":68,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":-708457359},"lastSeen":1702277335078,"defaultSendRequestWhen":"immediate"}

### Comments

[image]
Product URL (German!): https://www.brelag.com/shop/fmd2c018-knockautx-dimmaktor-unterputz-2-kanal-4530?page=2&attrib=31-87#attr=
Documents URL (German!): https://drive.google.com/drive/folders/1NIW2sIIwHBHbNrm0TxEWsk6sk3xyH2cy

[image]
Simmilar Prodct (Based on device pictures and specifications): https://moeshouse.com/products/zigbee-dimmer-module?variant=39820565807185
Document URL: https://drive.google.com/file/d/1shYi-pj7J0Ofx0bPGGN1s2T9c950jVmZ/view

Datapoint extracted according https://www.zigbee2mqtt.io/advanced/support-new-devices/03_find_tuya_data_points.html
| DP | Decription |
|--------|--------|
|  1 | Switch1 |
|  2 | Brightness1 |
|  3 | Minimum brightness1 |
|  4 | Type of light source1 |
|  6 | Countdown1 | 
|  7 | Switch2 |
|  8 | Brightness2 |
|  9 | Minimum brightness2 |
| 10| Type of light source2 |
| 12 | Countdown2 | 

Standard Instruction Set from Device Debugging                       .       
| Code | Type | Values |
|--------|--------|--------|
| switch_led_1 | Boolean | "{true,false}" |
| bright_value_1 | Integer | {"min": 10,"max": 1000,"scale": 0,"step": 1} |
| brightness_min_1 | Integer | {"min": 10,"max": 1000,"scale": 0,"step": 1} |
| led_type_1 | Enum | {"range": ["led","incandescent","halogen"]} |
| countdown_1 | Integer | {"unit": "s","min": 0,"max": 86400,"scale": 0,"step": 1} |                     .       .                  .                	
| switch_led_2 | Boolean | "{true,false}" |
| bright_value_2 | Integer | {"min": 10,"max": 1000,"scale": 0,"step": 1} |
| brightness_min_2 | Integer | {"min": 10,"max": 1000,"scale": 0,"step": 1} |
| led_type_2 | Enum | {"range": ["led","incandescent","halogen"]} |
| countdown_2 | Integer | {"unit": "s","min": 0,"max": 86400,"scale": 0,"step": 1} |   

Problem:
For every datapoint I get the same error message: **No datapoint defined for xyz**
```
Debug 2023-12-11 09:19:50 Received Zigbee message from '0x003c84fffe85c2c0', type 'attributeReport', cluster 'genBasic', data '{"65506":31,"65508":0,"appVersion":68}' from endpoint 1 with groupID 0
Debug 2023-12-11 09:19:51 Received MQTT message on 'zigbee2mqtt/0x003c84fffe85c2c0/set' with data '{"state_l1":"ON"}'
Debug 2023-12-11 09:19:51 Publishing 'set' 'state' to '0x003c84fffe85c2c0'
Error 2023-12-11 09:19:51 Publish 'set' 'state' to '0x003c84fffe85c2c0' failed: 'Error: No datapoint defined for 'state''
Debug 2023-12-11 09:19:51 Error: No datapoint defined for 'state' at Object.convertSet (/app/node_modules/zigbee-herdsman-converters/src/lib/tuya.ts:993:27) at Publish.onMQTTMessage (/app/lib/extension/publish.ts:249:52) at EventEmitter.emit (node:events:525:35) at EventBus.emitMQTTMessage (/app/lib/eventBus.ts:109:22) at MQTT.onMessage (/app/lib/mqtt.ts:141:27) at WebSocket.<anonymous> (/app/lib/extension/frontend.ts:130:27) at WebSocket.emit (node:events:513:28) at Receiver.receiverOnMessage (/app/node_modules/ws/lib/websocket.js:1192:20) at Receiver.emit (node:events:513:28) at Receiver.dataMessage (/app/node_modules/ws/lib/receiver.js:560:14)
Info 2023-12-11 09:19:51 MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'state' to '0x003c84fffe85c2c0' failed: 'Error: No datapoint defined for 'state''","meta":{"friendly_name":"0x003c84fffe85c2c0"},"type":"zigbee_publish_error"}'
``` 

Question:
How should my external converter be programmed so I can use this Tuya device?

### External converter

```shell
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const definition = {
    fingerprint: [{modelID: 'TS0601',manufacturerName: '_TZE200_tsxpl0d0',},],
    model: 'FMD2C018',
    vendor: 'KnockautX Intelligent Home by Brelag AG Switzerland',
    description: '2 gang smart dimmer module',
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [tuya.tz.datapoints],
    onEvent: tuya.onEventSetTime,
    configure: tuya.configureMagicPacket,
    exposes: [
	tuya.exposes.switch().withEndpoint('l1'),
	tuya.exposes.switch().withEndpoint('l1'),
        tuya.exposes.lightBrightness().withEndpoint('l1'),
        tuya.exposes.lightBrightness().withEndpoint('l2'),
        tuya.exposes.countdown().withEndpoint('l1'),
        tuya.exposes.countdown().withEndpoint('l2'),
        tuya.exposes.lightType().withEndpoint('l1'),
        tuya.exposes.lightType().withEndpoint('l2'),
    ],
    meta: {
        multiEndpoint: true,
        tuyaDatapoints: [
            // Channel 1
            [1, 'Switch1', tuya.valueConverter.onOff, {skip: tuya.skip.stateOnAndBrightnessPresent}],
            [2, 'Brightness1', tuya.valueConverter.scale0_254to0_1000],
            [3, 'Minimum brightness1', tuya.valueConverter.scale0_254to0_1000],
            [4, 'Type of light source1', tuya.valueConverterBasic.lookup({'led': tuya.enum(0), 'incandescent': tuya.enum(1), 'halogen': tuya.enum(2)})],
            [6, 'Countdown1', tuya.valueConverter.countdown],
            // Channel 2
            [7, 'Switch2', tuya.valueConverter.onOff, {skip: tuya.skip.stateOnAndBrightnessPresent}],
            [8, 'Brightness2', tuya.valueConverter.scale0_254to0_1000],
            [9, 'Minimum brightness2', tuya.valueConverter.scale0_254to0_1000],
            [10, 'Type of light source2', tuya.valueConverterBasic.lookup({'led': tuya.enum(0), 'incandescent': tuya.enum(1), 'halogen': tuya.enum(2)})],
            [12, 'Countdown2', tuya.valueConverter.countdown],
        ],
    },
    endpoint: (device) => {
        return {
            'l1': 1,
            'l2': 2
        };
    },
};

module.exports = definition;
```

thotha, 2023-12-13:
Dear Koen
As a update information for this device.
If I use the local S1 and S2 switch connection and either use a short press for On/Off or a long press for dimming, on the status page of this device in Z2M, I can see that the On/Off and Dimm commands are recognized and the values are beeing changed. But still the won't be changed on the Details page and also all set commands form Z2M to the device create failures with the simple note "Error: No datapoint defined for 'state'.
If you need any more informations and data, please tell me.

Koenkk, 2023-12-15:
Looks good, could you submit a pull request?

thotha, 2023-12-15:
sorry, but can you explain how to do that? I've never done that before and also what all needs to be written and added to such a pull request. 
thank you for your help.

Koenkk, 2023-12-15:
I've added it!

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

closed 2023-12-15 by Koenkk