Koenkk/zigbee2mqtt #20154: Error on OTA for OSRAM 4052899926110 LED strip

Bastian007, 2023-12-11:
### What happened?

Checking for new firmware for my OSRAM led strip generates the following error message:
`Failed to check if update available for 'Osram Flex RGBW - Wohnzimmerschrank' (No image available for manufacturerCode '4364' imageType '2')`
The front end identifies the Firmware built date as 20140331CNWT**** and the firmware version is V1.05.10

According to the 
https://www.zigbee2mqtt.io/devices/4052899926110.html#osram-4052899926110
that device is supporting OTA.

Ledvance who took over Osram is also showing/offering a newer Firmware:
https://update.ledvance.com/firmware-overview?submit=all

### What did you expect to happen?

No error message, instead
it should show either
No update available for 'Osram Flex RGBW- Wohnzimmerschrank'
or
should identfy a newer Firmware.

### How to reproduce it (minimal and precise)

Run OTA check in Z2M for that device.

### Zigbee2MQTT version

1.34.0-1

### Adapter firmware version

Z-Stack_3.x.0 coordinator 20221226

### Adapter

SONOFF Zigbee 3.0 USB Dongle Plus

### Debug log

```
Debug 2023-12-11 09:25:24Received MQTT message on 'zigbee2mqtt/bridge/request/device/ota_update/check' with data '{"id":"Osram Flex RGBW - Wohnzimmerschrank","transaction":"d3t1y-2"}'
Info 2023-12-11 09:25:24Checking if update available for 'Osram Flex RGBW - Wohnzimmerschrank'
Debug 2023-12-11 09:25:24Received Zigbee message from 'Osram Flex RGBW - Wohnzimmerschrank', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":0,"fileVersion":16909584,"imageType":2,"manufacturerCode":4364}' from endpoint 3 with groupID 0
Info 2023-12-11 09:25:24MQTT publish: topic 'zigbee2mqtt/bridge/response/device/ota_update/check', payload '{"data":{"id":"Osram Flex RGBW - Wohnzimmerschrank"},"error":"Failed to check if update available for 'Osram Flex RGBW - Wohnzimmerschrank' (No image available for manufacturerCode '4364' imageType '2')","status":"error","transaction":"d3t1y-2"}'
Error 2023-12-11 09:25:24Failed to check if update available for 'Osram Flex RGBW - Wohnzimmerschrank' (No image available for manufacturerCode '4364' imageType '2')
```

Bastian007, 2024-01-08:
After update of Z2M to version 1.35.1 I get notification that a newer firmware is available.
But when I try to update the firmware I see this message in the log:
`Error 2024-01-08 08:52:10Update of 'Osram Flex RGBW - Wohnzimmerschrank' failed (Images currently unavailable)`

Also in Home assistant I can see the notification about an available update with strange version information:
[image]

Koenkk, 2024-01-09:
The `-1` has been fixed in the dev branch.

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

Bastian007, 2024-01-11:
Problem fixed. Thanks!

closed 2024-01-11 by Bastian007