Koenkk/zigbee2mqtt #20156: [New device support]: TS0601, _TZE200_iv6ph5tr - KnockautX FMS2C017/maybe MOES ZRM-104B-MS

thotha, 2023-12-11:
### Link

https://www.brelag.com/shop/fms2c017-knockautx-licht-schaltaktor-unterputz-2-kanal-4529?page=2&attrib=31-87#attr=

### Database entry

{"id":3,"type":"Router","ieeeAddr":"0x003c84fffe6fd27a","nwkAddr":24023,"manufId":4098,"manufName":"_TZ3000_iv6ph5tr","powerSource":"Mains (single phase)","modelId":"TS011F","epList":[1,2,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":266,"inClusterList":[0,3,4,5,6,57344,57345],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65534":0,"stackVersion":0,"dateCode":"","manufacturerName":"_TZ3000_iv6ph5tr","zclVersion":3,"appVersion":67,"modelId":"TS011F","powerSource":1}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b002b4889df","endpointID":1},{"cluster":2820,"type":"endpoint","deviceIeeeAddress":"0x00124b002b4889df","endpointID":1},{"cluster":1794,"type":"endpoint","deviceIeeeAddress":"0x00124b002b4889df","endpointID":1}],"configuredReportings":[],"meta":{}},"2":{"profId":260,"epId":2,"devId":266,"inClusterList":[3,4,5,6,57344,57345],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":67,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1702285158789,"defaultSendRequestWhen":"immediate"}

### Comments

[image]

Product URL (German!): https://www.brelag.com/shop/fms2c017-knockautx-licht-schaltaktor-unterputz-2-kanal-4529?page=2&attrib=31-87#attr=
Documents URL (German!): https://drive.google.com/drive/folders/1wAacNL3VOTl6bkXZMSipX1I8P6jw_8GS

[image]

Simmilar Prodct (Based on device pictures and specifications): https://moeshouse.com/products/zigbee-104-module?variant=39820671352913
Document URL: https://drive.google.com/file/d/11lFLA1FxpXL2g96PYHJsJYolxpPDCWYb/view

Datapoint extracted according https://www.zigbee2mqtt.io/advanced/support-new-devices/03_find_tuya_data_points.html

| DP | Decription |
|--------|--------|
|  1 | Switch 1 |
|  2 | Switch 2 |
|  9 | Countdown 1 |
| 10 | Countdown 2 |
| 27 | Restart status | 

Standard Instruction Set from Device Debugging                       .       
| Code | Type | Values |
|--------|--------|--------|
| switch_1 | Boolean | "{true,false}" |
| switch_2 | Boolean | "{true,false}" |
| countdown_1 | Integer | {"unit": "s","min": 0,"max": 43200,"scale": 0,"step": 1} |                     .       .                  .                	
| countdown_2 | Integer | {"unit": "s","min": 0,"max": 43200,"scale": 0,"step": 1} |                     .       .                  .                	
| relay_status | Enum | {"range": ["power_off","power_on","last"]} |

Problem:
Despite I did try to create a converter file, this device will always be recognized as so totaly different product.
Instead of "_TZE200_iv6ph5tr" it will be recognized from Z2M as "_TZ3000_iv6ph5tr" which is a totaly different product, a 1-Channel Socket Switch.

Question:
How should my external converter be programmed so I can use this Tuya device?

### External converter

```shell
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const definition = {
    fingerprint: [{modelID: 'TS0601',manufacturerName: '_TZE200_iv6ph5tr',},],
    model: 'FMS2C017',
    vendor: 'KnockautX Intelligent Home by Brelag AG Switzerland',
    description: '2 gang switch',
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [tuya.tz.datapoints],
	onEvent: tuya.onEventSetTime, 
    configure: tuya.configureMagicPacket,
    exposes: [
        tuya.exposes.switch().withEndpoint('l1'),
        tuya.exposes.switch().withEndpoint('l2'),
        tuya.exposes.countdown().withEndpoint('l1'),
        tuya.exposes.countdown().withEndpoint('l2'),
        // Both of the following exposes do not work either
        // e.enum('power_on_behavior', ea.STATE_SET, ['off', 'on', 'previous']),
        // or
        // e.power_on_behavior(['off', 'on', 'previous']).withAccess(ea.STATE_SET),
    ],
    meta: {
        multiEndpoint: true,
        tuyaDatapoints: [
        	// Channel 1
            [1, 'Switch 1', tuya.valueConverter.onOff],
            [9, 'Countdown 1', tuya.valueConverter.countdown],
            // Channel 2
            [2, 'Switch 2', tuya.valueConverter.onOff,], 
            [10, 'Countdown 2', tuya.valueConverter.countdown],
            // All channels
            [27, 'Restart status', tuya.valueConverter.powerOnBehaviorEnum],
        ],
    },
    endpoint: (device) => {
        return {'l1': 1, 'l2': 1};
    },
};

// export default definitions;
module.exports = definitions;

module.exports = definition;
```

Koenkk, 2023-12-11:
Your ext converter looks good, does it work with this?

thotha, 2023-12-11:
unfortunately no, it doesn't
see my problem description

Koenkk, 2023-12-12:
Try: https://gist.github.com/Koenkk/cd0206b2ed9d76c5430811df37c3cd89

thotha, 2023-12-12:
Hello Keon
Thank you for your suggestion, but there was no change in behaviour.
Any other way to get this working?
Do you need any other outputs, logs and such?

[failure.log](https://github.com/Koenkk/zigbee2mqtt/files/13653740/failure.log)

thotha, 2023-12-13:
Dear Koen
Just a little information.
In ZHA latest version in Hassio, I'm able to add this switch and I can send On/Off commands, despite ZHA also does recognize it as TS011F by _TZ3000_iv6ph5tr and not as TS0601 by _TZE200_iv6ph5tr as it should.
Mybe this is of some help for you. Also let me know if you need anything else.

Koenkk, 2023-12-13:
z2m also detects it as `TS011F`, from your database.db entry: `"modelId":"TS011F",`, I updated https://gist.github.com/Koenkk/cd0206b2ed9d76c5430811df37c3cd89

thotha, 2023-12-15:
Zigbee2MQTT disregards this proposed converter and does not use it at all. Instead Z2M chooses, for no reason, the integrated converter [“Tuya” “TS011F_plug_1”](https://www.zigbee2mqtt.io/devices/TS011F_plug_1.html#tuya-ts011f_plug_1). With this, however, a completely wrong device is assigned to this module.
[image]

Surprisingly, in ZHA this module is recognised as “TS011F by _TZ3000_iv6ph5tr” and the quirk “[zhaoquirks.tuya.ts011f_switch.Tuya_2G_Switch](https://github.com/zigpy/zha-device-handlers/blob/dev/zhaquirks/tuya/ts011f_switch.py#L31)” is used. This allows me to switch the two channels on/off as needed. OK, the functions “Countdown” and “Restart status” are not recognised here either, but they don’t matter to me at the moment.
[image]

My question now is how to integrate this device into Z2M, whether and how to rewrite the ZHA Quirk into a working Z2M converter.
My other module [KnockautX FMD2C018/maybe MOES ZM-105B-MS](https://github.com/Koenkk/zigbee2mqtt/issues/20153) was successfully integrated with the Z2M converter.

Koenkk, 2023-12-15:
Updated https://gist.github.com/Koenkk/cd0206b2ed9d76c5430811df37c3cd89

thotha, 2023-12-15:
Dear Koen
Thank you for your converter config.
1.) I can see now the correct exposes, I can press the buttons On/Off or change the Countdown, but the device doesn't change its state.
2.) I can see in the log file that Z2M does send the commands but nothing happens.
3.) If I do change On/Off values on the device itself, I here the relays like they flip state, I can see the signal arrive in the log file but no change in the exposes view.
4.) I had to change the last entry from `module.exports = definitions;` to `module.exports = definition;`. Else Z2M wouldn't start at all.

So we're getting closer to a working solution.
The device code in question is `0x003c84fffe90671c`.
After successful registration, I first did try several times to with On/Off and then change values for countdown.
then I did change several times the On/Off state on the device it self...
Logfile below.

`023-12-15 21:49:44Received Zigbee message from '0x003c84fffe90671c', type 'attributeReport', cluster 'manuSpecificTuya_3', data '{"powerOnBehavior":0}' from endpoint 2 with groupID 0
Info 2023-12-15 21:49:44MQTT publish: topic 'zigbee2mqtt/0x003c84fffe90671c', payload '{"countdown_l1":null,"countdown_l2":null,"last_seen":"2023-12-15T21:49:44+01:00","linkquality":29,"state_l1":null,"state_l2":null}'
Debug 2023-12-15 21:49:44Received Zigbee message from '0x003c84fffe90671c', type 'readResponse', cluster 'genBasic', data '{"stackVersion":0}' from endpoint 1 with groupID 0
Info 2023-12-15 21:49:44MQTT publish: topic 'zigbee2mqtt/0x003c84fffe90671c', payload '{"countdown_l1":null,"countdown_l2":null,"last_seen":"2023-12-15T21:49:44+01:00","linkquality":29,"state_l1":null,"state_l2":null}'
Debug 2023-12-15 21:49:44Received Zigbee message from '0x003c84fffe90671c', type 'read', cluster 'genTime', data '["localTime"]' from endpoint 1 with groupID 0
Info 2023-12-15 21:49:44MQTT publish: topic 'zigbee2mqtt/0x003c84fffe90671c', payload '{"countdown_l1":null,"countdown_l2":null,"last_seen":"2023-12-15T21:49:44+01:00","linkquality":36,"state_l1":null,"state_l2":null}'
Debug 2023-12-15 21:49:44Received Zigbee message from '0x003c84fffe90671c', type 'readResponse', cluster 'genBasic', data '{"dateCode":""}' from endpoint 1 with groupID 0
Info 2023-12-15 21:49:44MQTT publish: topic 'zigbee2mqtt/0x003c84fffe90671c', payload '{"countdown_l1":null,"countdown_l2":null,"last_seen":"2023-12-15T21:49:44+01:00","linkquality":32,"state_l1":null,"state_l2":null}'
Debug 2023-12-15 21:49:45Received Zigbee message from '0x003c84fffe90671c', type 'readResponse', cluster 'genBasic', data '{}' from endpoint 1 with groupID 0
Info 2023-12-15 21:49:45MQTT publish: topic 'zigbee2mqtt/0x003c84fffe90671c', payload '{"countdown_l1":null,"countdown_l2":null,"last_seen":"2023-12-15T21:49:45+01:00","linkquality":29,"state_l1":null,"state_l2":null}'
Info 2023-12-15 21:49:45Successfully interviewed '0x003c84fffe90671c', device has successfully been paired
Info 2023-12-15 21:49:45Device '0x003c84fffe90671c' is supported, identified as: KnockautX Intelligent Home by Brelag AG Switzerland 2 gang switch (FMS2C017)
Info 2023-12-15 21:49:45MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"definition":{"description":"2 gang switch","exposes":[{"endpoint":"l1","features":[{"access":3,"description":"On/off state of the switch","endpoint":"l1","label":"State","name":"state","property":"state_l1","type":"binary","value_off":"OFF","value_on":"ON","value_toggle":"TOGGLE"}],"type":"switch"},{"endpoint":"l2","features":[{"access":3,"description":"On/off state of the switch","endpoint":"l2","label":"State","name":"state","property":"state_l2","type":"binary","value_off":"OFF","value_on":"ON","value_toggle":"TOGGLE"}],"type":"switch"},{"access":3,"description":"Countdown to turn device off after a certain time","endpoint":"l1","label":"Countdown","name":"countdown","property":"countdown_l1","type":"numeric","unit":"s","value_max":43200,"value_min":0,"value_step":1},{"access":3,"description":"Countdown to turn device off after a certain time","endpoint":"l2","label":"Countdown","name":"countdown","property":"countdown_l2","type":"numeric","unit":"s","value_max":43200,"value_min":0,"value_step":1},{"access":1,"description":"Link quality (signal strength)","label":"Linkquality","name":"linkquality","property":"linkquality","type":"numeric","unit":"lqi","value_max":255,"value_min":0}],"model":"FMS2C017","options":[],"supports_ota":false,"vendor":"KnockautX Intelligent Home by Brelag AG Switzerland"},"friendly_name":"0x003c84fffe90671c","ieee_address":"0x003c84fffe90671c","status":"successful","supported":true},"type":"device_interview"}'
Info 2023-12-15 21:49:45Configuring '0x003c84fffe90671c'
Info 2023-12-15 21:49:45MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_successful","meta":{"description":"2 gang switch","friendly_name":"0x003c84fffe90671c","model":"FMS2C017","supported":true,"vendor":"KnockautX Intelligent Home by Brelag AG Switzerland"},"type":"pairing"}'
Debug 2023-12-15 21:49:45Received Zigbee message from '0x003c84fffe90671c', type 'readResponse', cluster 'genBasic', data '{"65534":0,"appVersion":67,"manufacturerName":"_TZ3000_iv6ph5tr","modelId":"TS011F","powerSource":1,"zclVersion":3}' from endpoint 1 with groupID 0
Info 2023-12-15 21:49:45MQTT publish: topic 'zigbee2mqtt/0x003c84fffe90671c', payload '{"countdown_l1":null,"countdown_l2":null,"last_seen":"2023-12-15T21:49:45+01:00","linkquality":21,"state_l1":null,"state_l2":null}'
[... cut ...]
Info 2023-12-15 21:51:15MQTT publish: topic 'zigbee2mqtt/0x003c84fffe90671c', payload '{"countdown_l1":11849,"countdown_l2":13045,"last_seen":"2023-12-15T21:51:15+01:00","linkquality":76,"state_l1":"OFF","state_l2":"OFF"}'
Debug 2023-12-15 21:51:15Received Zigbee message from '0x003c84fffe90671c', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID 0
Info 2023-12-15 21:51:15MQTT publish: topic 'zigbee2mqtt/0x003c84fffe90671c', payload '{"countdown_l1":11849,"countdown_l2":13045,"last_seen":"2023-12-15T21:51:15+01:00","linkquality":76,"state_l1":"OFF","state_l2":"OFF"}'`

thotha, 2023-12-16:
Good morning Koen
thank you for this updated converter.
Looks good for the switch part.
But for to officially create a pull, the countdown needs to be implemented as well, despite I don't need it for myself.

I did try to add this missing part myself, but either if I try to add `extend: tuya.extend.countdown({endpoints: ['l1', 'l2']}),` command or add `exposes: tuya.exposes.countdown().withEndpoint('l1'), tuya.exposes.countdown().withEndpoint('l12'),` Z2M won't start, so I miss the correct structure to modify the converter myself.

Thank you again for your help.
Have a nice day.
Regards
Thomas

Koenkk, 2023-12-17:
I'm not sure about the countdown, but can you check if switch type works with https://gist.github.com/Koenkk/cd0206b2ed9d76c5430811df37c3cd89

thotha, 2023-12-17:
Good morning Koen
Thank you again for your converter.
This new function "switchType" is not supported from my device. 
It's also not shown as supported feature in the extracted datapoints
DP | Decription
-- | --
1 | Switch 1
2 | Switch 2
9 | Countdown 1
10 | Countdown 2
27 | Restart status

and the standard instruction set from device debugging.

Code | Type | Values
-- | -- | --
switch_1 | Boolean | "{true,false}"
switch_2 | Boolean | "{true,false}"
countdown_1 | Integer | {"unit": "s","min": 0,"max": 43200,"scale": 0,"step": 1}
countdown_2 | Integer | {"unit": "s","min": 0,"max": 43200,"scale": 0,"step": 1}
relay_status | Enum | {"range": ["power_off","power_on","last"]}

The only left functionality is countdown, but is no problem for me as HA has the much better countdown possibilities.
The converter without "switchType" is working for me and I can use it.

Have a nice and quiet Sunday with family and friends.
Kind regards
Thomas

Koenkk, 2023-12-17:
Added the device, let's ignore countdown for now, probably requires sniffing with the original hub because datapoints API doesn't seem to work for this device.

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

closed 2023-12-17 by Koenkk