Koenkk/zigbee2mqtt #20160: MatSee Plus DAC2161C _TZE204_lsanae15

MBRetrofit, 2023-12-11:
### Is your feature request related to a problem? Please describe

Although DAC2161C _TZE204_lsanae15 is correctly recognized in the zigbee2mqtt, the live parameters like Power, Current, Voltage are displayed with a delay of 3 minutes. This transforms the smart device into a dumb switch.

### Describe the solution you'd like

Any solution for changing this delay is welcome,

### Describe alternatives you've considered

OTA, Tasmota, anything....

### Additional context

I use latest Zigbee2mqtt and I also ask seller and receive nothing usefull

MBRetrofit, 2024-01-13:
From today after a new HA Core installed the MatSsee _TZE204_lsanae15
 device update Power at 4 minutes. So maybe is not a Firmware problem.

Antherak, 2024-08-18:
Hello, I bought this equipment because it was indicated as being zigbee2mqtt compatible. [https://www.zigbee2mqtt.io/devices/DAC2161C.html]
But I have the same issue with this device. The data are not available un real time (even  if you use the tuya app/ i try one time). You have to wait 3 or 4 minutes for a refresh so it's just non sense.
I have this warning when i interview the device : 
_[ZDO] Device '40995' is only compliant to revision '21' of the ZigBee specification (current revision: 22)._
could this be related?
Thx