Koenkk/zigbee2mqtt #20164: Aqara smart wall switch H1 EU (single & double rocker) decoupled mode broken

Alfy1080, 2023-12-11:
### What happened?

Ever since upgrading to Z2M 1.34.0, the operation mode switch no longer works. Based on other previously reported bugs, the issue was supposed to be fixed by upgrading to the latest coordinator firmware, but I am already running the latest available firmware on my Sonoff Dongle-P. 
Re-Pairing the device to the same or different router or directly to the coordinator does not help.

Whenever i try to change the operation mode on any of my aqara switches i get the following error:

2023-12-11 18:07:31Publish 'set' 'operation_mode' to 'Hallway_Switch' failed: 'Error: undefined is not a object, got string ("decoupled")'
2023-12-11 18:07:44Publish 'set' 'operation_mode' to 'Bathroom_Switch' failed: 'Error: undefined is not a object, got string ("decoupled")'
2023-12-11 18:07:44Publish 'set' 'operation_mode' to 'Bathroom_Switch' failed: 'Error: undefined is not a object, got string ("decoupled")'
2023-12-11 18:07:56Publish 'set' 'operation_mode' to 'Kids_Room_Switch' failed: 'Error: undefined is not a object, got string ("decoupled")'
2023-12-11 18:07:57Publish 'set' 'operation_mode' to 'Kids_Room_Switch' failed: 'Error: undefined is not a object, got string ("control_relay")'

Workaround: roll back to Z2M 1.33.0, change the operation mode as per your preference and upgrade back to Z2M 1.34.0. The setting is preserved after the upgrade so a rollback will only be necessary if i need to change the operation mode.

### What did you expect to happen?

Operation mode switch should successfully switch between relay control and decoupled mode.

### How to reproduce it (minimal and precise)

Upgrade to 1.34.0
Pair or have paired an Aqara wall switch H1 EU to Z2M
Try to change the operation mode via Z2M UI or Home Assistant UI.

### Zigbee2MQTT version

1.34.0

### Adapter firmware version

20230507

### Adapter

Sonoff ZB Dongle-P (CC2652P)

### Debug log

[Uploading log.txt…]()

Alfy1080, 2023-12-11:
[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/13638339/log.txt)

julianfs, 2023-12-11:
Already logged: https://github.com/Koenkk/zigbee2mqtt/issues/20001

Alfy1080, 2023-12-11:
> Duplicate of #20036 ?

You are right. I stumbled upon that thread but somehow missed the fact that it is fixed in the DEV branch. So i guess i just have to wait for the next stable version release. Thanks for pointing that out! :)