Koenkk/zigbee2mqtt #20166: Aqara T1 with Neutral error 'device_temperature_calibration'

jmcollin78, 2023-12-11:
### What happened?

I have installed Aqara T1 with Neutral with this attributes:

```
Disponibilité En ligne
Type d'appareil Router
Modèle Zigbee lumi.switch.n0agl1
Constructeur Zigbee LUMI
Description Aqara single switch module T1 (with neutral)
État de prise en charge Pris en charge

Adresse IEEE 0xxxxxxx
Adresse réseau 0xDF01
Date du firmware Dec 30 2021
Constructeur [Xiaomi](https://www.zigbee2mqtt.io/supported-devices/#v=Xiaomi)
Modèle [SSM-U01](https://www.zigbee2mqtt.io/devices/SSM-U01.html#xiaomi-ssm-u01)
```

The power info is not reported when the switch is on.

Immediatly after pairing, le power infos are reported. And after many days (don't know exactly when), the power info are no more reported.

In the log I see those logs:
```
Error 2023-12-11 14:49:32Exception while calling fromZigbee converter: 'device_temperature_calibration' is not a number, got string ()}
Debug 2023-12-11 14:49:32Error: 'device_temperature_calibration' is not a number, got string () at toNumber (/app/node_modules/zigbee-herdsman-converters/src/lib/utils.ts:548:15) at calibrateAndPrecisionRoundOptions (/app/node_modules/zigbee-herdsman-converters/src/lib/utils.ts:89:29) at numericAttributes2Payload (/app/node_modules/zigbee-herdsman-converters/src/lib/xiaomi.ts:190:79) at numericAttributes2Payload (/app/node_modules/zigbee-herdsman-converters/src/lib/xiaomi.ts:503:67) at Object.convert (/app/node_modules/zigbee-herdsman-converters/src/lib/xiaomi.ts:1415:51) at Receive.onDeviceMessage (/app/lib/extension/receive.ts:150:51)
```

It seems to be a converter problem.

### What did you expect to happen?

I expect to have the power report and not to have this error log.

### How to reproduce it (minimal and precise)

1. pair the device,
2. check that power report is working as expected,
3. wait (days ?),
4. the power report doesn't work anymore.

### Zigbee2MQTT version

1.34.0-1

### Adapter firmware version

Dec 30 2021

### Adapter

Skyconnect

### Debug log

Error 2023-12-11 14:49:32Exception while calling fromZigbee converter: 'device_temperature_calibration' is not a number, got string ()}
Debug 2023-12-11 14:49:32Error: 'device_temperature_calibration' is not a number, got string () at toNumber (/app/node_modules/zigbee-herdsman-converters/src/lib/utils.ts:548:15) at calibrateAndPrecisionRoundOptions (/app/node_modules/zigbee-herdsman-converters/src/lib/utils.ts:89:29) at numericAttributes2Payload (/app/node_modules/zigbee-herdsman-converters/src/lib/xiaomi.ts:190:79) at numericAttributes2Payload (/app/node_modules/zigbee-herdsman-converters/src/lib/xiaomi.ts:503:67) at Object.convert (/app/node_modules/zigbee-herdsman-converters/src/lib/xiaomi.ts:1415:51) at Receive.onDeviceMessage (/app/lib/extension/receive.ts:150:51)

xEcho1, 2023-12-14:
Maybe check out this thread :https://github.com/Koenkk/zigbee2mqtt/issues/19747. Seems like you're having similar problems with similar hardware, maybe a downgrade to 1.33.0-0 is worth a try.

jmcollin78, 2023-12-14:
That is right @xEcho1. I will follow this tread also. Thank you

dCatfish, 2024-03-13:
I got rid of these errors by adding a zero in the blank offset fields in "Settings (specific)" for all my devices. 

[image]

jmcollin78, 2024-09-11:
I think the issue is still there even If an easy workaround is possible

closed 2025-01-01 by jmcollin78