Koenkk/zigbee2mqtt #20169: Bosch twinguard Pairing fails

Mikkooooo, 2023-12-11:
### What happened?

Hi,
My new Bosch Twinguard just arrived.
I‘ve tried many attempts but everytime the pairing fails.

Can anybody help me please ?

### Zigbee2MQTT version

1.33.1

### Adapter firmware version

6.10.3.0 build 297

### Adapter

Sonoff Dongle E       EZSP v8

Mikkooooo, 2023-12-12:
> You'll need to add the install code from the back of the device into Zigbee2MQTT before it will pair
> 
> `Z2MGUI`->`Settings`->`Tools`->`Add Install Code`

I‘ve done this already. The .log says „successfully added Install code“ but no pairing success.
Weeks ago I added a BSD-2 Bosch Smoke Alarm and I know the Install code procedure.

vaenror, 2024-01-02:
Same here, added a BSD-2 with install code just fine. But the TwinGuard fails at interviewing. Tried everything, no shot.

Mikkooooo, 2024-01-02:
> Same here, added a BSD-2 with install code just fine. But the TwinGuard fails at interviewing. Tried everything, no shot.

Which Coordinator do you use ?
I‘ve tried with Sonoff dongle E and now I‘m switched over to Sonoff dongle P.
I will start to try with the dongle P hoping with success.

vaenror, 2024-01-02:
> > Same here, added a BSD-2 with install code just fine. But the TwinGuard fails at interviewing. Tried everything, no shot.
> 
> Which Coordinator do you use ? I‘ve tried with Sonoff dongle E and now I‘m switched over to Sonoff dongle P. I will start to try with the dongle P hoping with success.

Also Dongle-E flashed with MultiPan. Dongle-P ist now ordered, will try it tomorrow night with the P.

Mikkooooo, 2024-01-03:
> Also Dongle-E flashed with MultiPan. Dongle-P ist now ordered, will try it tomorrow night with the P.

Did you have success ?
My new Twinguard will arrive next days.
Will try it too.

vaenror, 2024-01-03:
> Did you have success ? My new Twinguard will arrive next days. Will try it too.

Sadly my order is delayed, i don’t know when it will arrive. Will update here when i’ve tried it.

vaenror, 2024-01-04:
Successfully paired the Twinguard with the Dongle-P via install code.

Mikkooooo, 2024-01-04:
> Successfully paired the Twinguard with the Dongle-P via install code.

Same as mine. 👍

I‘ve got no measurement values until now lie battery, temperature, humidity …
I think it will take some time ?
How long did you have to wait ?
Smoke alarm, self test, burglar alarm and pre alarm are working well from the beginning of pairing.

Mikkooooo, 2024-01-05:
I've successfully paired the Twinguard.
Smoke alarm, self test, burglar alarm and pre alarm are working well from the beginning of pairing (with a small delay 1-2 sec) but not values like battery, temperature, humidity,...
After switching z2m off and on the measurements are available exept the battery value.
What's going wrong ?

acosys, 2024-01-23:
Hi there,

sams here. I'm using the SkyConnect Dongle and paring (with using the install-code) fails. Tried it several times and with and without push the button to keep the device awake. :(

Christian81IN, 2024-01-23:
Hi,
Similar here, I'm also using the Skyconnect and pairing fails all the time.
Is it an issue of the adapter? 
Is there anyone who sucessfully connected with the Skyconnect to the twinguard?

fabioslomp, 2024-01-24:
Hi there,

I am experiencing the same with an Smlight SLZB-06M PoE Coordinator. Install code starts interview but pairs an unsupported device... 

Any ideas on how to get that interviewing?

Thanks

Christian81IN, 2024-01-26:
I bought a Sonoff P stick, as it was stated above that it works. After I found out that I need to change the adapter zu zstack in the z2m configuration, the twinguard was connected.

Mikkooooo, 2024-01-26:
Hi Christian,
ist the battery Status connected to
 ?  Mine isn't

Christian81IN, 2024-01-26:
Yes, it is connected. All values can be seen. Just the CO2 took some while until ist was not null anymore.

"battery": 100,
    "co2": 540,
    "heartbeat": "ON",
    "humidity": 46.7,
    "illuminance_lux": 63,
    "linkquality": 72,
    "pre_alarm": "ON",
    "self_test": false,
    "siren_state": "clear",
    "smoke": false,
    "temperature": 23.9

fabioslomp, 2024-01-29:
Thanks Christian, I think the zstack could do the trick, problem is that my device has the EFR32MG21 chip (sames as skyconnect) and I only found the zstack firmware for the CC2652P chip. I guess that is the same issue skyconnect folks are having, did some research but did not manage to go much further than that.

Christian81IN, 2024-01-29:
From Chatgpt:
_If the controller and device are based on different Zigbee protocol stacks or support different versions/profiles of the Zigbee standard, they may not be able to properly negotiate these parameters during the pairing process. This can result in failed pairing attempts or incomplete pairing, where the device is recognized but cannot fully integrate into the network or communicate effectively with the controller._

Can it be that this is a Hardware incompatibility problem?

fabioslomp, 2024-01-29:
The EFR32MG21 chip devices still under experimental devices in z2m, I hope it is a question of time since the EFR32MG21 chip seems to be the way to go as it is more powerful than the cc2652p. I have over 100 zigbee devices of all kinds the twinguard is the only one bothering so far

Mar1usW3, 2024-02-03:
some issue with the SkyConnect any idea how to fix this?

lerra, 2024-02-04:
I have the same issue, I had a conbeeII adapter but I now bought a SLZB-06M and I have exactly the same issue :( Anybody that managed to solve this?

technikjunge, 2024-02-23:
Exactly the same problem with the SLZB-06M. Pairing does not work. It would be great if there was a solution for this.

kedane2000, 2024-04-24:
I have the same problem. I add the installation code, can pair my Twinguard, but I get the message that the device is not supported/interview failed.

Failed to interview '0x000d6f0018b5a968', device has not successfully been paired.

[image]

xF4m3, 2024-04-28:
> I've successfully paired the Twinguard. Smoke alarm, self test, burglar alarm and pre alarm are working well from the beginning of pairing (with a small delay 4-5 sec) but not values like battery, temperature, humidity,... After switching z2m off and on the measurements are available exept the battery value. What's going wrong ?

I have the same issue, but restarting Z2M does not help, it successfully paired, lighting up green aswell and the LQI seems to update regularly, but no other values are populating. I have set the adapter to zstack aswell. I am using the Zigstar UZG-01 with a CC2652P7.

I can also activate the alarm sounds and initiate the self test, but all measurement values show null, except the siren state, which changes when i activate a diffrent alarm sound.
[image]

Nerivec, 2024-05-07:
@Christian81IN Could we get a debug log (of you pairing the device to z2m) from your zstack since it appears you were able to make this device work there?

Christian81IN, 2024-05-07:
Sure, can you point me to a manual how to create the debug log? Or is it just the normal log of z2m addon?

Nerivec, 2024-05-07:
If you are on 1.37.0 it's all in one place yes. Just switch log level to `debug`. See [docs](https://www.zigbee2mqtt.io/guide/usage/debug.html). The log file is in your installation/config folder (depends on your setup).

xF4m3, 2024-05-08:
I have let it paired since my last post and because of the activity here today i checked back if anything changed. And now it seems to report all the values, i am unsure if they updated regularly yet. It definetly took more then 24h to show values since i checked a day after pairing and there was nothing. Only Battery seems to have not reported yet and the Siren state did not reset from what i last set, so that doesnt seem to update aswell. As you can see below it took a few days until it started reporting. very weird, in the bosch app/bridge it took about 30min-2h until it reported values. 

[image]

Not sure if this is important, but i think that was not there the first time i paired it, or atleast there were less entries i think.
[image]

But this is definetly interessting, it took about 5 days until it started to report, good thing i can look at the history graph.
[image]

Mar1usW3, 2024-05-08:
@Nerivec i also had the problem with skyconnect/yellow (ezsp driver) that the pairing worked but no values were reported. I saw multiple post that the issue only occurs with silicon labs chips. Thats also for ZHA...
Till now i didn't test this with ember. I will test this the next days.

kedane2000, 2024-05-08:
I have just tried something else. I flashed the firmware 7.4.2.0 to the Skyconnect and used "ember" as the adapter. Unfortunately, I was unable to connect the Twinguard despite this.

```
Info 2024-05-08 16:35:00Interview for '0x000d6f0018b5a968' started
Info 2024-05-08 16:35:00Starting interview of '0x000d6f0018b5a968'
Warning 2024-05-08 16:35:02Received network/route error ROUTE_ERROR_SOURCE_ROUTE_FAILURE for "46375".
Error 2024-05-08 16:35:29Interview failed for '0x000d6f0018b5a968 with error 'Error: Interview failed because can not get node descriptor ('0x000d6f0018b5a968')'
Error 2024-05-08 16:35:29Failed to interview '0x000d6f0018b5a968', device has not successfully been paired
Info 2024-05-08 16:35:37Interview for '0x000d6f0018b5a968' started
Info 2024-05-08 16:35:37Starting interview of '0x000d6f0018b5a968'
Warning 2024-05-08 16:35:47Received network/route error ROUTE_ERROR_SOURCE_ROUTE_FAILURE for "55506".
Error 2024-05-08 16:36:06Interview failed for '0x000d6f0018b5a968 with error 'Error: Interview failed because can not get node descriptor ('0x000d6f0018b5a968')'
Error 2024-05-08 16:36:06Failed to interview '0x000d6f0018b5a968', device has not successfully been paired
Info 2024-05-08 16:36:22Interview for '0x000d6f0018b5a968' started
Info 2024-05-08 16:36:22Starting interview of '0x000d6f0018b5a968'
Warning 2024-05-08 16:36:32Received network/route error ROUTE_ERROR_SOURCE_ROUTE_FAILURE for "61221".
Warning 2024-05-08 16:36:46Received network/route error ROUTE_ERROR_SOURCE_ROUTE_FAILURE for "43663".
Error 2024-05-08 16:36:51Interview failed for '0x000d6f0018b5a968 with error 'Error: Interview failed because can not get node descriptor ('0x000d6f0018b5a968')'
Error 2024-05-08 16:36:51Failed to interview '0x000d6f0018b5a968', device has not successfully been paired
Info 2024-05-08 16:36:53Interview for '0x000d6f0018b5a968' started
Info 2024-05-08 16:36:53Starting interview of '0x000d6f0018b5a968'
Warning 2024-05-08 16:36:55Received network/route error ROUTE_ERROR_SOURCE_ROUTE_FAILURE for "43663".
Error 2024-05-08 16:37:22Interview failed for '0x000d6f0018b5a968 with error 'Error: Interview failed because can not get node descriptor ('0x000d6f0018b5a968')'
Error 2024-05-08 16:37:22Failed to interview '0x000d6f0018b5a968', device has not successfully been paired
Info 2024-05-08 16:38:12[STACK STATUS] Network closed.
```

Tronix117, 2024-07-24:
I got some success by:
1. pairing with ZBDongle-P
2. migrating to ZBDongle-E

Also by:
1. pairing with ZHA
2. migrating to Zigbee2Mqtt

However it's far from perfect, plus it doesn't seem I can reproduce those manipulations every time with a success... It worked once for ten tries, and I can not figured out what I did specificaly the time it worked. The only difference is that the time it worked, I had some other zigbee routers connected, but I'm hesitant to say it's related.

What is weird is that the pairing is:
* working great with ZBDongle-P
* working great with ZBDongle-E and ZHA (Home Assistant) or Zigpy

And is not working with ZBDongle-E and Zigbee2Mqtt (using adapter: ezsp, it seems I can't use adapter: ember for some reasons... status=HOST_FATAL_ERROR, but it's a different issue).

Nerivec, 2024-07-24:
Can anyone try this again with latest-dev ember and report the logs back?
Also if someone can provide a sniff, that would be great: [latest docs](https://github.com/Koenkk/zigbee2mqtt.io/blob/develop/docs/advanced/zigbee/04_sniff_zigbee_traffic.md#configuring-wireshark)
https://github.com/Nerivec/ember-zli/wiki/Sniff

_@Tronix117 Let me know if you need some help with that ember switch (I check Discord regularly, you can DM me from zigbee2mqtt server, as needed)._

runningman84, 2024-07-27:
I have tried to pair a new Twinguard in 1.39.0 but I get this error:
````
[2024-07-27 17:27:29] ^[[31merror^[[39m:        zh:controller: Interview failed for '0x000d6f001910bf32 with error 'Error: Interview failed because can not get node descriptor ('0x000d6f001910bf32')'
[2024-07-27 17:27:29] ^[[31merror^[[39m:        z2m: Failed to interview '0x000d6f001910bf32', device has not successfully been paired
````
I have a zigstar lan gateway which works fine with all other devices.

Is there anything I can do to help figure this out?

apoc4lyps, 2024-08-02:
I have the same issued, tested with 1.39.1

Adapter firmware version
7.4.2 [GA]

Adapter
Home Assistant SkyConnect  (EmberZNet)

```
Error 2024-08-02 13:11:56zh:controller: Interview failed for '0x000d6f0019104d22 with error 'Error: Interview failed because can not get node descriptor ('0x000d6f0019104d22')'
Error 2024-08-02 13:11:56z2m: Failed to interview '0x000d6f0019104d22', device has not successfully been paired
```

Nerivec, 2024-08-02:
Since the device appears to be in [a loop of join](https://github.com/Koenkk/zigbee2mqtt/issues/22446#issuecomment-2094954177), I'll assume it's expecting something non-standard (maybe something similar to Aqara that only wants to pair if the coordinator reports itself as an Aqara hub, vendor-lock kind of thing).
This is supported by the fact that this device is not Zigbee 3.0 compliant (uses a much less secure protocol version), and appears to be customized on top of that, [per docs](https://www.bosch-smarthome.com/uk/en/service/help/product-help/twinguard-help/):
> Your Twinguard communicates with the Bosch Smart Home system via the encrypted ZigBee HA 1.2 protocol. In addition, manufacturer-specific protocol extensions are used to ensure a secure and reliable connection.

runningman84, 2024-08-09:
I could pair my twinguard today using the install code option with the barcode from the device. The device seems to work now only these things are reported as null:

VOC
CO2
AQI

TheMagnetar, 2024-08-10:
@runningman84 what did you do to successfully pair it?

Tronix117, 2024-08-10:
> @runningman84 what did you do to successfully pair it?

I imagine he used the older chip zbdongle-P, maybe the thread should be renamed by appending "with ZBDongle-E"

TheMagnetar, 2024-08-10:
I think it is a general problem with EmberZNet. I am using a SMLIGHT SLZB-07

runningman84, 2024-08-11:
> @runningman84 what did you do to successfully pair it?

I have just used the install code method. I am using a Zigstar lan Gateway in my setup and have about 40 zigbee devices in my network.

apoc4lyps, 2024-08-13:
I can confirm that the twinguard is pairing with an "old" CC2652 on the same z2m version. 
Is there anything from z2m that can fix this behavior or is this related to emberZnet firmware?

Mar1usW3, 2024-08-15:
> Since the device appears to be in [a loop of join](https://github.com/Koenkk/zigbee2mqtt/issues/22446#issuecomment-2094954177), I'll assume it's expecting something non-standard (maybe something [...]

Understood... but why is it working with an non silicon labs chip?

apoc4lyps, 2024-08-16:
I think I will return the twinguard and use a BSD-2 instead, sadly

S474N, 2024-08-27:
Also problem with pairing. I have 9 pieces of BSD-2. 8 paired OK, but the last has problem.

Coordinator: zStack3x0 (SLZB-06) - CC2652P
Revision: 20240710
Z2M: [1.39.1](https://github.com/Koenkk/zigbee2mqtt/releases/tag/1.39.1) commit: [e132316](https://github.com/Koenkk/zigbee2mqtt/commit/e132316)

```
Info 2024-06-02 18:50:56MQTT publish: topic 'Z2M/0xf082c0fffe0ab753', payload '{"last_seen":"2024-06-02T16:50:56.209Z"}'
...
Error 2024-06-02 18:51:01Interview failed for '0xf082c0fffe0ab753 with error 'Error: Interview failed because can not get node descriptor ('0xf082c0fffe0ab753')'
Error 2024-06-02 18:51:01Failed to interview '0xf082c0fffe0ab753', device has not successfully been paired
Info 2024-06-02 18:51:01MQTT publish: topic 'Z2M/bridge/event', payload '{"data":{"friendly_name":"0xf082c0fffe0ab753","ieee_address":"0xf082c0fffe0ab753","status":"failed"},"type":"device_interview"}'
Info 2024-06-02 18:51:01MQTT publish: topic 'Z2M/bridge/log', payload '{"message":"interview_failed","meta":{"friendly_name":"0xf082c0fffe0ab753"},"type":"pairing"}'
Info 2024-06-02 18:51:01MQTT publish: topic 'Z2M/0xf082c0fffe0ab753', payload '{"last_seen":"2024-06-02T16:51:01.456Z"}'
Info 2024-06-02 18:51:01Interview for '0xf082c0fffe0ab753' started
Info 2024-06-02 18:51:01Starting interview of '0xf082c0fffe0ab753'
Info 2024-06-02 18:51:02MQTT publish: topic 'Z2M/bridge/event', payload '{"data":{"friendly_name":"0xf082c0fffe0ab753","ieee_address":"0xf082c0fffe0ab753","status":"started"},"type":"device_interview"}'
Info 2024-06-02 18:51:02MQTT publish: topic 'Z2M/bridge/log', payload '{"message":"interview_started","meta":{"friendly_name":"0xf082c0fffe0ab753"},"type":"pairing"}'
```

[image]

After clicking to DEV CONSOLE on device:
[image]

mdiezi, 2024-09-01:
Hi, same problem here, tried with skyconnect and then ZBDongle-E (firmware 7.4.3.0), ezps then ember...z2m seems to detect the twinguard but then pairing fails. Just updated Z2M to 1.40.0-1, but that doesn't seem to help...
Any new idea? Thanks in advance

Przemo-999, 2024-09-02:
I have the same issued, tested with 1.40.0
EmberZNet
Adapter
ZBDongle-E (firmware 7.4.3.0)

MarcPot, 2024-09-08:
Same problem, also dongle-E, ember and latest firmware.

Maybe the device should be removed from the supported page? Or have some kind of warning?

Magic01, 2024-09-16:
Hi,
I have also the problem, that I can pair the device but I do not get any values - I paired it already several times but the picture never changed. I get it paired - there is no error but I never get any values like smoke, temperature, humidity and so on.
If I try to set e.g. the heartbeat I get an error:
2024-09-16 08:40:26z2m: Publish 'set' 'heartbeat' to '0x000d6f00196fc41c' failed: 'Error: Cluster with name 'twinguardSetup' does not exist'

I guess more or less same like the others but I **do not have a Dongle-E, I have a P**.
I had a pretty old firmware on it but now I'm on the lastet one:
Zigbee2MQTT-Version 
1.40.1
Coordinator-Typ
zStack3x0
Coordinator-Version
20240710
Frontend-Version
0.7.4
Zigbee Herdsman Konverter Version
20.12.1
Zigbee Herdsman Version
0.57.3

rdillenburger, 2024-10-09:
Hi, 
I also have the same issue with the Bosch Twingaurd devices ([https://www.zigbee2mqtt.io/devices/8750001213.html])

Updated all related Components yesterday evening and still have this issue:
```
Error 2024-10-08 22:07:33zh:controller: Interview failed for 'XY with error 'Error: Interview failed because can not get node descriptor ('XY')'
Error 2024-10-08 22:07:33z2m: Failed to interview 'XY', device has not successfully been paired
```
Entering Installation Code (photo QR-Code on the back of the device) works fine, but the interview fails with the error message above. 
I tried the same with 3 different devices of that type; all with the same issue. Even after hardware Reset no change in the behaviour.

**My current environment:** 

- Home Assistant Core: 2024.10.1
- Supervisor: 2024.10.0
- Operating System: 13.1
- Frontend: 20241002.2

Zigbee2MQTT:

- Version: 1.40.2
- Coordinator Type: EmberZNet (Sonoff E)
- Coordinator Version: 7.4.3 [GA]
- Frontend Version: 0.7.4
- Zigbee Herdsman Konverter Version: 20.21.0
- Zigbee Herdsman Version: 2.1.3

Any ideas to that topic?

ssunny1081, 2024-10-19:
I tried for hours to get the Twinguard connected, there was no sign in the logs that it is even trying to pair, in the End it turned out that ZigBee Channel 26 was the culprit. 

After Switching to Channel 25, it did work right away. Maybe it helps someone..

Twinguard 8750001215

Zigbee2MQTT-Version
[1.40.2]
Coordinator-Typ
zStack3x0
Coordinator-Version
20240710
ZBDongle-P

MarcPot, 2024-10-19:
> I tried for hours to get the Twinguard connected, there was no sign in the logs that it is even trying to pair, in the End it turned out that ZigBee Channel 26 was the culprit.
>
> After Switching to [...]

Dongle-P again. So for users ofthat dongle it’s an option to try.

MOROHEUZ, 2024-10-22:
Same issue with my configuration. 
The dongle E no matter with which firmware or z2m version does not bring a correct result
He joined once, but then he peeped every few minutes as if the battery was empty. 

No problems with the ZStack Dongle P. 
I don't know if this is only the case with the twinguard, or in general with the bosch devices?

Puma7, 2024-11-02:
> I could pair my twinguard today using the install code option with the barcode from the device. The device seems to work now only these things are reported as null:
> 
> VOC CO2 AQI

Hi, did you find a solution for thet issue with VOC CO2 AQI = null?

runningman84, 2024-11-04:
> > I could pair my twinguard today using the install code option with the barcode from the device. The device seems to work now only these things are reported as null:
> > VOC CO2 AQI
> 
> Hi, did you find a solution for thet issue with VOC CO2 AQI = null?
> 
> Edit: In my case it took about 2 hours until the values where shown. So sometimes just wait :)

yes you just have to wait for these values...

TheMagnetar, 2024-11-04:
> Same issue with my configuration. The dongle E no matter with which firmware or z2m version does not bring a correct result He joined once, but then he peeped every few minutes as if the battery was [...]

It is just with this device (as far as I know) and only with the Ember adapter

rdillenburger, 2024-11-06:
> Same issue with my configuration. The dongle E no matter with which firmware or z2m version does not bring a correct result He joined once, but then he peeped every few minutes as if the battery was [...]

It's only that Bosch device. I have several instances of the Bosch Smoke Alarm 2 ([https://www.zigbee2mqtt.io/devices/BSD-2.html](https://www.zigbee2mqtt.io/devices/BSD-2.html)) in my Environment, which have all been paired with my Dongle E using installation code.

Przemo-999, 2024-11-13:
> Hi, same problem here, I have the same issued, tested with 1.40.0 EmberZNet Adapter ZBDongle-E (firmware 7.4.3.0)
> 
> In another location the same Bosch smoke detector paired and works fine, but there I have ZBDongle-P and ezsp

info 2024-11-13 20:32:09zh:ember:ezsp: Received network/route error ROUTE_ERROR_INDIRECT_TRANSACTION_EXPIRY for "53301".
info 2024-11-13 20:32:11zh:ember:ezsp: Received network/route error ROUTE_ERROR_INDIRECT_TRANSACTION_EXPIRY for "53301".
error 2024-11-13 20:32:11zh:controller: Interview failed for '0x000d6f0018b5b2a6 with error 'Error: Interview failed because can not get active endpoints ('0x000d6f0018b5b2a6')'
error 2024-11-13 20:32:12z2m: Failed to interview '0x000d6f0018b5b2a6', device has not successfully been paired
info 2024-11-13 20:32:12z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x000d6f0018b5b2a6","ieee_address":"0x000d6f0018b5b2a6","status":"failed"},"type":"device_interview"}'
info 2024-11-13 20:32:15zh:controller: Interview for '0x000d6f0018b5b2a6' started
info 2024-11-13 20:32:15z2m: Starting interview of '0x000d6f0018b5b2a6'
info 2024-11-13 20:32:15z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x000d6f0018b5b2a6","ieee_address":"0x000d6f0018b5b2a6","status":"started"},"type":"device_interview"}'
info 2024-11-13 20:32:15zh:ember:ezsp: Received network/route error ROUTE_ERROR_INDIRECT_TRANSACTION_EXPIRY for "53301".
info 2024-11-13 20:32:17zh:ember:ezsp: Received network/route error ROUTE_ERROR_INDIRECT_TRANSACTION_EXPIRY for "53301".
info 2024-11-13 20:32:24zh:ember:ezsp: Received network/route error ROUTE_ERROR_INDIRECT_TRANSACTION_EXPIRY for "3453".
error 2024-11-13 20:32:25zh:controller: Interview failed for '0x000d6f0018b5b2a6 with error 'Error: Interview failed because can not get active endpoints ('0x000d6f0018b5b2a6')'
error 2024-11-13 20:32:25z2m: Failed to interview '0x000d6f0018b5b2a6', device has not successfully been paired
info 2024-11-13 20:32:25z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x000d6f0018b5b2a6","ieee_address":"0x000d6f0018b5b2a6","status":"failed"},"type":"device_interview"}'
info 2024-11-13 20:32:26zh:ember:ezsp: Received network/route error ROUTE_ERROR_INDIRECT_TRANSACTION_EXPIRY for "3453".
info 2024-11-13 20:32:30zh:controller: Interview for '0x000d6f0018b5b2a6' started
info 2024-11-13 20:32:30z2m: Starting interview of '0x000d6f0018b5b2a6'
info 2024-11-13 20:32:30z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x000d6f0018b5b2a6","ieee_address":"0x000d6f0018b5b2a6","status":"started"},"type":"device_interview"}'
info 2024-11-13 20:32:31zh:ember:ezsp: Received network/route error ROUTE_ERROR_INDIRECT_TRANSACTION_EXPIRY for "3453".
error 2024-11-13 20:32:41zh:controller: Interview failed for '0x000d6f0018b5b2a6 with error 'Error: Interview failed because can not get active endpoints ('0x000d6f0018b5b2a6')'
error 2024-11-13 20:32:41z2m: Failed to interview '0x000d6f0018b5b2a6', device has not successfully been paired
info 2024-11-13 20:32:41z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x000d6f0018b5b2a6","ieee_address":"0x000d6f0018b5b2a6","status":"failed"},"type":"device_interview"}'
info 2024-11-13 20:32:44zh:controller: Interview for '0x000d6f0018b5b2a6' started
info 2024-11-13 20:32:44z2m: Starting interview of '0x000d6f0018b5b2a6'
info 2024-11-13 20:32:45z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x000d6f0018b5b2a6","ieee_address":"0x000d6f0018b5b2a6","status":"started"},"type":"device_interview"}'
info 2024-11-13 20:32:53zh:ember:ezsp: Received network/route error ROUTE_ERROR_INDIRECT_TRANSACTION_EXPIRY for "53301".
error 2024-11-13 20:32:58zh:controller: Interview failed for '0x000d6f0018b5b2a6 with error 'Error: Interview failed because can not get active endpoints ('0x000d6f0018b5b2a6')'
error 2024-11-13 20:32:58z2m: Failed to interview '0x000d6f0018b5b2a6', device has not successfully been paired
info 2024-11-13 20:32:59z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x000d6f0018b5b2a6","ieee_address":"0x000d6f0018b5b2a6","status":"failed"},"type":"device_interview"}'
info 2024-11-13 20:32:59zh:ember:ezsp: Received network/route error ROUTE_ERROR_INDIRECT_TRANSACTION_EXPIRY for "53301".
info 2024-11-13 20:32:59zh:controller: Interview for '0x000d6f0018b5b2a6' started
info 2024-11-13 20:32:59z2m: Starting interview of '0x000d6f0018b5b2a6'
info 2024-11-13 20:33:00z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x000d6f0018b5b2a6","ieee_address":"0x000d6f0018b5b2a6","status":"started"},"type":"device_interview"}'
info 2024-11-13 20:33:00zh:ember:ezsp: Received network/route error ROUTE_ERROR_INDIRECT_TRANSACTION_EXPIRY for "53301".
error 2024-11-13 20:33:13zh:controller: Interview failed for '0x000d6f0018b5b2a6 with error 'Error: Interview failed because can not get active endpoints ('0x000d6f0018b5b2a6')'
error 2024-11-13 20:33:13z2m: Failed to interview '0x000d6f0018b5b2a6', device has not successfully been paired
info 2024-11-13 20:33:13z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x000d6f0018b5b2a6","ieee_address":"0x000d6f0018b5b2a6","status":"failed"},"type":"device_interview"}'
info 2024-11-13 20:33:14zh:controller: Interview for '0x000d6f0018b5b2a6' started
info 2024-11-13 20:33:14z2m: Starting interview of '0x000d6f0018b5b2a6'
info 2024-11-13 20:33:15z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x000d6f0018b5b2a6","ieee_address":"0x000d6f0018b5b2a6","status":"started"},"type":"device_interview"}'

Przemo-999, 2024-11-13:
> info 2024-11-13 20:32:09zh:ember:ezsp: Received network/route error ROUTE_ERROR_INDIRECT_TRANSACTION_EXPIRY for "53301". info 2024-11-13 20:32:11zh:ember:ezsp: Received network/route error [...]

PLEASE HELP Koenkk [Bosch 8750001213](https://www.zigbee2mqtt.io/devices/8750001213.html#bosch-8750001213) with SONOFF_Zigbee_3.0_USB_Dongle_Plus_V2_20230525180906-if00 adapter: ember

merquel, 2024-12-28:
Has anyone been able to resolve this? Is the only solution to replace the SkyConnect dongle?

ashleysmithgpu, 2025-01-14:
Just another data point: I also cannot get the twinguard to pair.

Twinguard Mat/N: 8750001215
Coordinator: zStack3x0 (SONOFF Zigbee 3.0 USB CC2652P)
Revision: 20240710
Z2M: 2.0.0

juliangeiges, 2025-01-18:
As another data point. Count pair any new Twinguards. After setting Z2M back to a prior 2 version. Pairing works fine.

Coordinator: zStack3x0 Texas Instruments LAUNCHXL-CC1352P-2

[25 more replies]

closed 2025-02-10 by Koenkk