Koenkk/zigbee2mqtt #20170: [New device support]: Centralite 3405-L Keypad

Scags104, 2023-12-11:
### Link

https://pdf.lowes.com/useandcareguides/812489023049_use.pdf

### Database entry

{"id":15,"type":"EndDevice","ieeeAddr":"0x000d6f000e2dc47d","nwkAddr":40261,"manufId":4174,"manufName":"CentraLite","powerSource":"Battery","modelId":"3405-L","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":1025,"inClusterList":[0,1,3,32,1026,1280,1281,2821,64516,64517],"outClusterList":[25,1281],"clusters":{"genBasic":{"attributes":{"hwVersion":0}},"ssIasZone":{"attributes":{"iasCieAddr":"0x00124b002b484eb6","zoneState":1}},"genPollCtrl":{"attributes":{"checkinInterval":6480}}},"binds":[{"cluster":32,"type":"endpoint","deviceIeeeAddress":"0x00124b002b484eb6","endpointID":1}],"configuredReportings":[],"meta":{}}},"stackVersion":2,"hwVersion":0,"zclVersion":1,"interviewCompleted":true,"meta":{},"lastSeen":1702328410820,"defaultSendRequestWhen":"active","checkinInterval":1620}

### Comments

Coming over from ZHA this is the only device i cant get working out of the box

Scags104, 2024-06-09:
Would still like to see support here.

modrica, 2024-12-06:
Anyone got it to work?

gruberdieter, 2025-01-02:
I am in the same boat. Switched over from ZHA and this is one of the last missing. Thanks if that can be made possible.

closed 2025-03-11 by github-actions[bot]