Koenkk/zigbee2mqtt #20172: LELLKI WP33-EU/WP34-EU - Publish 'set' 'power_on_behavior' failed:

mrmaximas, 2023-12-11:
### What happened?

After update to dev branch i have errors with some my old devices. Only one for begin

2023-12-12 02:27:32 Publish 'set' 'power_on_behavior' to 'Power Strip' failed: 'Error: Write 0xa4c1381a6adee8f4/1 genOnOff({"startUpOnOff":255}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')'

Error 2023-12-12 02:27:34 Failed to configure 'Power Strip', attempt 3 (Error: Read 0xa4c1381a6adee8f4/1 genOnOff(["startUpOnOff"], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE') at Endpoint.checkStatus (/Users/max/zigbee2mqtt/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:321:28) at Endpoint.read (/Users/max/zigbee2mqtt/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:463:22) at setupAttributes (/Users/max/zigbee2mqtt/node_modules/zigbee-herdsman-converters/src/lib/modernExtend.ts:43:9) at configure (/Users/max/zigbee2mqtt/node_modules/zigbee-herdsman-converters/src/lib/modernExtend.ts:59:13) at Object.configure (/Users/max/zigbee2mqtt/node_modules/zigbee-herdsman-converters/src/index.ts:138:25) at Configure.configure (/Users/max/zigbee2mqtt/lib/extension/configure.ts:117:13))

### How to reproduce it (minimal and precise)

Just start z2m dev

### Zigbee2MQTT version

1.34.0-dev commit: 7c81eda4

### Adapter firmware version

20221226

### Adapter

zStack3x0

### Debug log

DB record {"id":4,"type":"Router","ieeeAddr":"0xa4c1381a6adee8f4","nwkAddr":54302,"manufId":4417,"manufName":"_TZ3000_air9m6af","powerSource":"Mains (single phase)","modelId":"TS011F","epList":[1,2,3,4,5,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":266,"inClusterList":[3,4,5,6,57344,57345,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65506":48,"65508":0,"65534":0,"modelId":"TS011F","manufacturerName":"_TZ3000_air9m6af","powerSource":1,"zclVersion":3,"appVersion":67,"stackVersion":0,"hwVersion":1,"dateCode":""}},"genOnOff":{"attributes":{"32768":0,"onOff":1,"onTime":0,"offWaitTime":0,"tuyaBacklightMode":1,"moesStartUpOnOff":2,"tuyaBacklightSwitch":1}},"manuSpecificTuya_3":{"attributes":{"powerOnBehavior":2,"switchType":0}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b002a1aa3c4","endpointID":1}],"configuredReportings":[{"cluster":6,"attrId":0,"minRepIntval":0,"maxRepIntval":65000,"repChange":1,"manufacturerCode":null}],"meta":{}},"2":{"profId":260,"epId":2,"devId":266,"inClusterList":[4,5,6,57345],"outClusterList":[],"clusters":{"genOnOff":{"attributes":{"onOff":1,"onTime":0,"offWaitTime":0}},"manuSpecificTuya_3":{"attributes":{"powerOnBehavior":2}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b002a1aa3c4","endpointID":1}],"configuredReportings":[{"cluster":6,"attrId":0,"minRepIntval":0,"maxRepIntval":65000,"repChange":1,"manufacturerCode":null}],"meta":{}},"3":{"profId":260,"epId":3,"devId":266,"inClusterList":[4,5,6,57345],"outClusterList":[],"clusters":{"genOnOff":{"attributes":{"onOff":1,"onTime":0,"offWaitTime":0}},"manuSpecificTuya_3":{"attributes":{"powerOnBehavior":2}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b002a1aa3c4","endpointID":1}],"configuredReportings":[{"cluster":6,"attrId":0,"minRepIntval":0,"maxRepIntval":65000,"repChange":1,"manufacturerCode":null}],"meta":{}},"4":{"profId":260,"epId":4,"devId":266,"inClusterList":[4,5,6,57345],"outClusterList":[],"clusters":{"genOnOff":{"attributes":{"onOff":0,"onTime":0,"offWaitTime":0}},"manuSpecificTuya_3":{"attributes":{"powerOnBehavior":2}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b002a1aa3c4","endpointID":1}],"configuredReportings":[{"cluster":6,"attrId":0,"minRepIntval":0,"maxRepIntval":65000,"repChange":1,"manufacturerCode":null}],"meta":{}},"5":{"profId":260,"epId":5,"devId":266,"inClusterList":[4,5,6,57345],"outClusterList":[],"clusters":{"genOnOff":{"attributes":{"onOff":1,"onTime":0,"offWaitTime":0}},"manuSpecificTuya_3":{"attributes":{"powerOnBehavior":2}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b002a1aa3c4","endpointID":1}],"configuredReportings":[{"cluster":6,"attrId":0,"minRepIntval":0,"maxRepIntval":65000,"repChange":1,"manufacturerCode":null}],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":67,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":-599931283},"lastSeen":1702337254000,"defaultSendRequestWhen":"immediate"}

Koenkk, 2023-12-12:
Fixed!

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

closed 2023-12-12 by Koenkk

mrmaximas, 2023-12-19:
Hi!
I tried using the device console and found that it works (r/w):
2023-12-20 00:32:45 Read result of 'manuSpecificTuya_3': {"powerOnBehavior":2}
2023-12-20 00:34:19 Wrote '{"powerOnBehavior":2}' to 'manuSpecificTuya_3'
2023-12-20 00:35:40 Wrote '{"powerOnBehavior":1}' to 'manuSpecificTuya_3'
2023-12-20 00:35:53 Read result of 'manuSpecificTuya_3': {"powerOnBehavior":1}
2023-12-20 00:36:06 Wrote '{"powerOnBehavior":2}' to 'manuSpecificTuya_3'
powerOnBehavior":2 - it's a Last State??

Koenkk, 2023-12-20:
Added!

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

mrmaximas, 2023-12-20:
> Added!
> 
> Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

it's works! thank you!