Koenkk/zigbee2mqtt #20174: TuYa TS011F_plug_1 - duplicate messages in the log.

mrmaximas, 2023-12-11:
### What happened?

All devices of this type always have the same double entries in the log at the same time. 
I'm not sure that this was not also the case in the master branch
https://www.zigbee2mqtt.io/devices/TS011F_plug_1.html#tuya-ts011f_plug_1

### Zigbee2MQTT version

1.34.0-dev commit: 7c81eda

### Adapter firmware version

20221226

### Adapter

zStack3x0

### Debug log

Info 2023-12-12 02:41:12MQTT publish: topic 'zigbee2mqtt/Spare Plug 3', payload '{"child_lock":"UNLOCK","current":0,"energy":12.3,"indicator_mode":"off/on","linkquality":32,"power":0,"power_outage_memory":"restore","state":"OFF","voltage":233}'
Info 2023-12-12 02:41:12MQTT publish: topic 'zigbee2mqtt/Spare Plug 3', payload '{"child_lock":"UNLOCK","current":0,"energy":12.3,"indicator_mode":"off/on","linkquality":40,"power":0,"power_outage_memory":"restore","state":"OFF","voltage":233}'
Info 2023-12-12 02:41:23MQTT publish: topic 'zigbee2mqtt/Spare Plug 3', payload '{"child_lock":"UNLOCK","current":0,"energy":12.3,"indicator_mode":"off/on","linkquality":36,"power":0,"power_outage_memory":"restore","state":"OFF","voltage":233}'
Info 2023-12-12 02:41:23MQTT publish: topic 'zigbee2mqtt/Spare Plug 3', payload '{"child_lock":"UNLOCK","current":0,"energy":12.3,"indicator_mode":"off/on","linkquality":36,"power":0,"power_outage_memory":"restore","state":"OFF","voltage":233}'
Info 2023-12-12 02:41:39MQTT publish: topic 'zigbee2mqtt/Spare Plug 2', payload '{"child_lock":"UNLOCK","current":0.03,"energy":15.5,"indicator_mode":"off/on","linkquality":178,"power":5,"power_outage_memory":"restore","state":"ON","voltage":237}'
Info 2023-12-12 02:41:47MQTT publish: topic 'zigbee2mqtt/Spare Plug 1', payload '{"child_lock":"UNLOCK","current":0,"energy":12.9,"indicator_mode":"off/on","linkquality":109,"power":0,"power_outage_memory":"restore","state":"ON","voltage":233}'
Info 2023-12-12 02:41:47MQTT publish: topic 'zigbee2mqtt/Spare Plug 1', payload '{"child_lock":"UNLOCK","current":0,"energy":12.9,"indicator_mode":"off/on","linkquality":109,"power":0,"power_outage_memory":"restore","state":"ON","voltage":233}'
Info 2023-12-12 02:41:58MQTT publish: topic 'zigbee2mqtt/Spare Plug 1', payload '{"child_lock":"UNLOCK","current":0,"energy":12.9,"indicator_mode":"off/on","linkquality":109,"power":0,"power_outage_memory":"restore","state":"ON","voltage":233}'
Info 2023-12-12 02:41:58MQTT publish: topic 'zigbee2mqtt/Spare Plug 1', payload '{"child_lock":"UNLOCK","current":0,"energy":12.9,"indicator_mode":"off/on","linkquality":109,"power":0,"power_outage_memory":"restore","state":"ON","voltage":233}'

mrmaximas, 2023-12-11:
Info 2023-12-12 02:48:23MQTT publish: topic 'zigbee2mqtt/Ipad 9 Power', payload '{"current":0,"energy":4.4,"indicator_mode":"off/on","linkquality":200,"power":0,"power_outage_memory":"restore","state":"OFF","voltage":232}'
Info 2023-12-12 02:48:23MQTT publish: topic 'zigbee2mqtt/Ipad 9 Power', payload '{"current":0,"energy":4.4,"indicator_mode":"off/on","linkquality":204,"power":0,"power_outage_memory":"restore","state":"OFF","voltage":232}'
Info 2023-12-12 02:48:34MQTT publish: topic 'zigbee2mqtt/Ipad 9 Power', payload '{"current":0,"energy":4.4,"indicator_mode":"off/on","linkquality":204,"power":0,"power_outage_memory":"restore","state":"OFF","voltage":232}'
Info 2023-12-12 02:48:34MQTT publish: topic 'zigbee2mqtt/Ipad 9 Power', payload '{"current":0,"energy":4.4,"indicator_mode":"off/on","linkquality":204,"power":0,"power_outage_memory":"restore","state":"OFF","voltage":232}'

kevincaradant, 2024-02-08:
Hi @mrmaximas and @Koenkk 

I'm facing to the same crazy issue. I'm not sure since when it's started. I thought it was because on my CC1352P-2 I just up to date the firmware to 2023 instead of a version of 2021.

But I don't think it was this. 

I have 85 devices, so it's not very stable by moment because of these messages which are too many I think 

[image]

[image]

Any idea how to fix it quickly to avoid spamming my network ? 

Thank you in advance ;)

mrmaximas, 2024-02-09:
> Edit : after to disable 'last seen' option. It's seems to be okay.

it's seems but not ok if you turn-on debug log ;)

kevincaradant, 2024-02-09:
Oh okay I didn't try the debug log. 

I have 10 of these plugs. All my zigbee network is instable. You have the same issue in the moment ?

I hope a solution will be find quickly to back on a normal behavior.

mrmaximas, 2024-02-09:
> Oh okay I didn't try the debug log.
> 
> I have 10 of these plugs. All my zigbee network is instable. You have the same issue in the moment ?
> 
I have currently removed these outlets from my house, but I have a similar problem with spam from another device that I can't remove yet ;) https://github.com/Koenkk/zigbee2mqtt/issues/21226
FW 20221226

Koenkk, 2024-02-14:
Could you provide the herdsman debug log of this?

See [this](https://www.zigbee2mqtt.io/guide/usage/debug.html) on how to enable the herdsman debug logging. Note that this is only logged to STDOUT and not to log files.

kevincaradant, 2024-02-28:
I haven't forgotten about you, Koenkk. Unfortunately, the issue seems to have disappeared since I rebooted with herdsman debug. It's really strange. So, if @mrmaximas is still experiencing the issue, we can keep this matter open. Otherwise, I'll leave it to mrmaximas to close it.

closed 2024-09-26 by github-actions[bot]

Myslawek, 2025-06-27:
Necroing. I too have issues with Tuya TS011F. It worked fine for 2 days and now it sends all messages duplicated.