Koenkk/zigbee2mqtt #20178: Zigbee2MQTT fails to start with "Error: spawn udevadm ENOEN"

sebastianlorenzen, 2023-12-12:
### What happened?

Fresh home assistant installation on a Raspi 4. (everything updated to latest version)
Sonoff Zigbee P dongle on USB hub with external Power Supply
Dongle is recognized by the OS
Mosquitto addon installed with default config
Zigbee2MQTT addon installed and configured as follows:

`data_path: /config/zigbee2mqtt
socat:
  enabled: false
  master: pty,raw,echo=0,link=/tmp/ttyZ2M,mode=777
  slave: tcp-listen:8485,keepalive,nodelay,reuseaddr,keepidle=1,keepintvl=1,keepcnt=5
  options: "-d -d"
  log: false
mqtt:
  server: mqtt://core-mosquitto:1883
serial:
  Port: >-
    /dev/serial/by-id/usb-ITead_Sonoff_Zigbee_3.0_USB_Dongle_Plus_0080d1c5a69ded11a86dd1a5a7669f5d-if00-port0
homeassistant: true
permit_join: false
zigbee_herdsman_debug: true`

when I try to start the Zigbee2MQTT addon, it fails with "Error: spawn udevadm ENOENT"

foll log:
`[09:54:05] INFO: Preparing to start...
[09:54:06] INFO: Socat not enabled
[09:54:06] INFO: Zigbee Herdsman debug logging enabled
[09:54:07] INFO: Starting Zigbee2MQTT...
Zigbee2MQTT:debug 2023-12-12 09:54:15: Can't load state from file /config/zigbee2mqtt/state.json (doesn't exist)
Zigbee2MQTT:info  2023-12-12 09:54:15: Logging to console and directory: '/config/zigbee2mqtt/log/2023-12-12.09-54-15' filename: log.txt
Zigbee2MQTT:info  2023-12-12 09:54:16: Starting Zigbee2MQTT version 1.34.0 (commit #unknown)
Zigbee2MQTT:info  2023-12-12 09:54:16: Starting zigbee-herdsman (0.25.0)
Zigbee2MQTT:debug 2023-12-12 09:54:16: Using zigbee-herdsman with settings: '{"adapter":{"concurrent":null,"delay":null,"disableLED":false},"backupPath":"/config/zigbee2mqtt/coordinator_backup.json","databaseBackupPath":"/config/zigbee2mqtt/database.db.backup","databasePath":"/config/zigbee2mqtt/database.db","network":{"channelList":[11],"extendedPanID":[202,185,94,134,44,110,16,106],"networkKey":"HIDDEN","panID":26334},"serialPort":{}}'
2023-12-12T08:54:16.563Z zigbee-herdsman:adapter No path provided, auto detecting path
Zigbee2MQTT:error 2023-12-12 09:54:16: Error while starting zigbee-herdsman
Zigbee2MQTT:error 2023-12-12 09:54:16: Failed to start zigbee
Zigbee2MQTT:error 2023-12-12 09:54:16: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
Zigbee2MQTT:error 2023-12-12 09:54:16: Exiting...
Zigbee2MQTT:error 2023-12-12 09:54:16: Error: spawn udevadm ENOENT
    at Process.ChildProcess._handle.onexit (node:internal/child_process:285:19)
    at onErrorNT (node:internal/child_process:485:16)
    at processTicksAndRejections (node:internal/process/task_queues:83:21)`

### Zigbee2MQTT version

1.34.0-1

### Adapter firmware version

unknown

### Adapter

Sonoff Zigbee 3.0 P

### Debug log

debug 2023-12-12 09:54:15: Can't load state from file /config/zigbee2mqtt/state.json (doesn't exist)
info  2023-12-12 09:54:15: Logging to console and directory: '/config/zigbee2mqtt/log/2023-12-12.09-54-15' filename: log.txt
info  2023-12-12 09:54:16: Starting Zigbee2MQTT version 1.34.0 (commit #unknown)
info  2023-12-12 09:54:16: Starting zigbee-herdsman (0.25.0)
debug 2023-12-12 09:54:16: Using zigbee-herdsman with settings: '{"adapter":{"concurrent":null,"delay":null,"disableLED":false},"backupPath":"/config/zigbee2mqtt/coordinator_backup.json","databaseBackupPath":"/config/zigbee2mqtt/database.db.backup","databasePath":"/config/zigbee2mqtt/database.db","network":{"channelList":[11],"extendedPanID":[202,185,94,134,44,110,16,106],"networkKey":"HIDDEN","panID":26334},"serialPort":{}}'
error 2023-12-12 09:54:16: Error while starting zigbee-herdsman
error 2023-12-12 09:54:16: Failed to start zigbee
error 2023-12-12 09:54:16: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
error 2023-12-12 09:54:16: Exiting...
error 2023-12-12 09:54:16: Error: spawn udevadm ENOENT
    at Process.ChildProcess._handle.onexit (node:internal/child_process:285:19)
    at onErrorNT (node:internal/child_process:485:16)
    at processTicksAndRejections (node:internal/process/task_queues:83:21)

sebastianlorenzen, 2023-12-12:
I have reinstalled Mosquitto and Zigbe2MQTT, issue still there.
I have tested ZHA and I can see the stick without issues. Have removed ZHA again.

sebastianlorenzen, 2024-01-01:
I tried it with a conbe2 USB stick and it worked. Is my stick broken?

sebastianlorenzen, 2024-01-24:
I switched from a Raspi to a Nuc and now it works.... strange

hyperlite75, 2024-02-15:
Sajnos nekem nem indul el a Zigbee2mqtt....Error: spawn udevadm ENOENT hibát dob fel.   

 • Install Zigbee2MQTT THE RIGHT WAY in ...  alapján telepítettem....
Napló:[22:57:39] INFO: Preparing to start...
[22:57:39] INFO: Socat not enabled
[22:57:39] INFO: Starting Zigbee2MQTT...
Zigbee2MQTT:info  2024-02-15 22:57:40: Logging to console and directory: '/config/zigbee2mqtt/log/2024-02-15.22-57-40' filename: log.txt
Zigbee2MQTT:info  2024-02-15 22:57:40: Starting Zigbee2MQTT version 1.35.3 (commit #unknown)
Zigbee2MQTT:info  2024-02-15 22:57:40: Starting zigbee-herdsman (0.33.8)
Zigbee2MQTT:error 2024-02-15 22:57:40: Error while starting zigbee-herdsman
Zigbee2MQTT:error 2024-02-15 22:57:40: Failed to start zigbee
Zigbee2MQTT:error 2024-02-15 22:57:40: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
Zigbee2MQTT:error 2024-02-15 22:57:40: Exiting...
Zigbee2MQTT:error 2024-02-15 22:57:40: Error: spawn udevadm ENOENT
    at Process.ChildProcess._handle.onexit (node:internal/child_process:284:19)
    at onErrorNT (node:internal/child_process:477:16)
    at processTicksAndRejections (node:internal/process/task_queues:82:21)

NUC és conbee3 az eszköz

enabled: false
master: pty,raw,echo=0,link=/tmp/ttyZ2M,mode=777
slave: tcp-listen:8485,keepalive,nodelay,reuseaddr,keepidle=1,keepintvl=1,keepcnt=5
options: "-d -d"
log: false
mqtt:
  server: mqtt://core-mosquitto
serial:
  port: /dev/ttyUSB0
  adapter: deconz

Mit rontok el?