Koenkk/zigbee2mqtt #20182: [Feature request]: Exclude device from logging

mrmaximas, 2023-12-12:
### Is your feature request related to a problem? Please describe

I have one device that sends reports very often. It's normal for this device types.
https://www.zigbee2mqtt.io/devices/SPM01-D2TZ.html 
I think it would be appropriate to provide for excluding devices from logging, at your own risk of course.
Thanks!

### Describe the solution you'd like

/

### Describe alternatives you've considered

/

### Additional context

Info 2023-12-12 18:26:54MQTT publish: topic 'zigbee2mqtt/Power Monitor', payload '{"current":5.63,"energy":133.3,"linkquality":112,"power":1199,"produced_energy":0.15,"voltage":230.4}'
Info 2023-12-12 18:26:54MQTT publish: topic 'zigbee2mqtt/Power Monitor', payload '{"current":5.63,"energy":133.3,"linkquality":116,"power":1199,"produced_energy":0.15,"voltage":230.4}'
Info 2023-12-12 18:26:54MQTT publish: topic 'zigbee2mqtt/Power Monitor', payload '{"current":5.66,"energy":133.3,"linkquality":116,"power":1196,"produced_energy":0.15,"voltage":230.3}'
Info 2023-12-12 18:26:59MQTT publish: topic 'zigbee2mqtt/Power Monitor', payload '{"current":5.66,"energy":133.3,"linkquality":112,"power":1196,"produced_energy":0.15,"voltage":230.3}'
Info 2023-12-12 18:26:59MQTT publish: topic 'zigbee2mqtt/Power Monitor', payload '{"current":5.66,"energy":133.3,"linkquality":112,"power":1196,"produced_energy":0.15,"voltage":230.3}'
Info 2023-12-12 18:26:59MQTT publish: topic 'zigbee2mqtt/Power Monitor', payload '{"current":5.65,"energy":133.3,"linkquality":112,"power":1192,"produced_energy":0.15,"voltage":230.3}'
Info 2023-12-12 18:27:04MQTT publish: topic 'zigbee2mqtt/Power Monitor', payload '{"current":5.65,"energy":133.3,"linkquality":112,"power":1192,"produced_energy":0.15,"voltage":230.3}'
Info 2023-12-12 18:27:04MQTT publish: topic 'zigbee2mqtt/Power Monitor', payload '{"current":5.65,"energy":133.3,"linkquality":112,"power":1192,"produced_energy":0.15,"voltage":230.3}'
Info 2023-12-12 18:27:04MQTT publish: topic 'zigbee2mqtt/Power Monitor', payload '{"current":5.72,"energy":133.3,"linkquality":112,"power":1207,"produced_energy":0.15,"voltage":230.4}'
Info 2023-12-12 18:27:09MQTT publish: topic 'zigbee2mqtt/Power Monitor', payload '{"current":5.72,"energy":133.3,"linkquality":112,"power":1207,"produced_energy":0.15,"voltage":230.4}'
Info 2023-12-12 18:27:09MQTT publish: topic 'zigbee2mqtt/Power Monitor', payload '{"current":5.72,"energy":133.3,"linkquality":116,"power":1207,"produced_energy":0.15,"voltage":230.4}'
Info 2023-12-12 18:27:09MQTT publish: topic 'zigbee2mqtt/Power Monitor', payload '{"current":5.65,"energy":133.3,"linkquality":112,"power":1194,"produced_energy":0.15,"voltage":231.2}'

mrmaximas, 2023-12-14:
> You might want to mess with your reporting thresholds (interval / change) in the device->Reporting tab (it would appear you have these set to quite an aggressive threshold) - this should reduce the amount the device sends and therefore reduce your log message count.

Thanks for you answer, page with Reporting tab looks like empty
[image]

danieledwardgeorgehitchcock, 2023-12-14:
Thanks for checking! This device does look particularly chatty! I can look at adding some debounce logic, but I don't think that'd help much given the logs that you have added above.

I am not sure device specific log exclusions is a good route to go. If you were to have an issue with the device, how would you know it without logging?

mrmaximas, 2023-12-14:
@danieledwardgeorgehitchcock this is the reason why I mention the requested feature - at the user's own risk ))

danieledwardgeorgehitchcock, 2023-12-14:
But users have a habit of forgetting which could lead to a nightmare of false-fault finding on here. Appreciate what you are saying though :-)

mrmaximas, 2023-12-14:
@danieledwardgeorgehitchcock this device is very chatty, that's normal for, but 95% of my log is filled with this device events.

evelant, 2023-12-17:
I think there's a bug in z2m somewhere that causes this unrelated to the actually zigbee portion. I've had this happen too. Z2M started started endlessly sending these power monitor messages to mqtt every ~1s for a plug until I restarted z2m.

mrmaximas, 2024-01-18:
> i'm using this workaround:
> 
> ```
>   '0xXXXXXX':
>     friendly_name: energy_input
>     debounce: 5
>     debounce_ignore:
>       - energy
>       - produced_energy
> ```

thanks!

closed 2024-08-19 by github-actions[bot]