Koenkk/zigbee2mqtt #20186: Zigbee2MQTT requires node version ^18 || ^20 || ^21, you are running v16.20.2!

sdotter, 2023-12-12:
### What happened?

Zigbee2MQTT requires node version ^18 || ^20 || ^21, you are running v16.20.2!

Running latest dev

### Zigbee2MQTT version

Latest

### Adapter firmware version

6.10.3.0

### Adapter

Sonoff dongle e

sjorge, 2023-12-12:
Exactly what it  says, the version of node you are using is too old. Might be you're running an older OS.

sdotter, 2023-12-12:
Yeah sorry forget to mention that this is on home assistants z2m addon (docker sizzle stuff)

sjorge, 2023-12-12:
Probably better to open an issue here https://github.com/zigbee2mqtt/hassio-zigbee2mqtt if that is the case.

sdotter, 2023-12-12:
Yeah you are right 👍 thanks jorge
This issue is closed

closed 2023-12-12 by sdotter