Koenkk/zigbee2mqtt #20187: BTicino K4003C/L4003C/N4003C/NT4003C not exposing the switch to Home Assistant

MattiaEffendi, 2023-12-12:
### What happened?

In Zigbee2Mqtt i see the "state" switch and it works as expected on the device.

[image]

Altough, in Home Assistant, the device is not exposing the switch to commute the state

[image]

The device is setup via the MQTT discovery function. I have two of these switches and they both have the same problem. They worked fine a few months ago and now this happens.

Zigbee2mqtt is running as a Home Assistant addon

### What did you expect to happen?

I expect to see the switch

### Zigbee2MQTT version

 1.34.0

### Adapter firmware version

20190425

### Adapter

CC2531

Koenkk, 2023-12-13:
Your HA version is probably outdated, try updating it

MattiaEffendi, 2023-12-13:
@Koenkk i'm running HA 2023.12.1.

Koenkk, 2023-12-13:
Does anything show up in the HA logs?

MattiaEffendi, 2023-12-13:
@Koenkk it is giving me some errors:
```
2023-12-12 20:26:39.812 ERROR (MainThread) [homeassistant.util.logging] Exception in async_discover when dispatching 'mqtt_discovery_new_switch_mqtt': ({'availability': [{'topic': 'zigbee2mqtt/bridge/state', 'value_template': '{{ value_json.state }}'}], 'command_topic': 'zigbee2mqtt/Corridoio/set', 'device': {'identifiers': ['zigbee2mqtt_0x0004740000b30bf9'], 'manufacturer': 'BTicino', 'model': 'Light switch with neutral (K4003C/L4003C/N4003C/NT4003C)', 'name': 'Corridoio', 'sw_version': '00b\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00 \x02\x00\x01\x00 \x02\x00\x02\x00!\x12\x00\x00\x00\x00��\x03\x00\x10\x02\x01\x04\x00\x18\x02\x00��!'}, 'json_attributes_topic': 'zigbee2mqtt/Corridoio', 'name': None, 'object_id': 'corridoio', 'origin': {'name': 'Zigbee2MQTT', 'sw': '1.34.0', 'url': 'https://www.zigbee2mqtt.io'}, 'payload_off': 'OFF', 'payload_on': 'ON', 'state_topic': 'zigbee2mqtt/Corridoio', 'unique_id': '0x0004740000b30bf9_switch_zigbee2mqtt', 'value_template': '{{ value_json.state }}', 'platform': 'mqtt'},)
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/mqtt/mixins.py", line 316, in async_discover
    config: DiscoveryInfoType = discovery_schema(discovery_payload)
  File "/usr/local/lib/python3.10/site-packages/voluptuous/schema_builder.py", line 272, in __call__
    return self._compiled([], data)
  File "/usr/local/lib/python3.10/site-packages/voluptuous/schema_builder.py", line 595, in validate_dict
    return base_validate(path, iteritems(data), out)
  File "/usr/local/lib/python3.10/site-packages/voluptuous/schema_builder.py", line 433, in validate_mapping
    raise er.MultipleInvalid(errors)
voluptuous.error.MultipleInvalid: string value is None for dictionary value @ data['name']
```

Koenkk, 2023-12-13:
That's the error I would expect with 2023.8.0, can you double check that in HA -> settings -> about shows 2023.12.1?

MattiaEffendi, 2023-12-13:
@Koenkk i'm sorry for making you lose time.
I manage quite a few home assistant installs and I thought this was one that I updated when I do my batch update sessions. Turns out this one is still running 2023.3.6 (yes, I will definitely update).

Thank you for your quick support and sorry again

closed 2023-12-14 by MattiaEffendi