Koenkk/zigbee2mqtt #20189: [New device support]: Moes ZTRV-801-EN

Appoxus, 2023-12-13:
### Link

https://de.aliexpress.com/item/1005006194184981.html?src=google&aff_fcid=d9fb02d4926e4ccfb968ddb2709d14fa-1701894068736-04266-UneMJZVf&aff_fsk=UneMJZVf&aff_platform=aaf&sk=UneMJZVf&aff_trace_key=d9fb02d4926e4ccfb968ddb2709d14fa-1701894068736-04266-UneMJZVf&terminal_id=37aa72aa1a514423b2841296bbaae943&afSmartRedirect=y

### Database entry

{"id":11,"type":"EndDevice","ieeeAddr":"0xa4c1389685b118e8","nwkAddr":21500,"manufId":4417,"manufName":"_TZE200_rtrmfadk","powerSource":"Battery","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[4,5,61184,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"\u000b{\u000b-i\f{\u000b-i\r{\u000b-i\u000e{\u000b-iS~\u000b-iU~\u000b-iV~\u000b-i","65506":56,"65508":0,"65534":0,"modelId":"TS0601","manufacturerName":"_TZE200_rtrmfadk","powerSource":3,"zclVersion":3,"appVersion":72,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":72,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":-708457359},"lastSeen":1702447427274,"defaultSendRequestWhen":"immediate"}

### Comments

I bought some of these on Aliexpress and they get detected as TS0601 by Z2M. They're a ripoff of the Bosch Thermostat and there's a Wifi version as well.

Setting the temperature, child lock and window mode via HA seems to work, the schedule won't work.
The Debug log throws some errors. Could you please have a look at this?

Here's the Debug log after pairing and trying to set temperatures and the schedule via HA. FYI I wanted to remove the schedule, since I plan to do the scheduling via HA.

[debug_log_TRV801.txt](https://github.com/Koenkk/zigbee2mqtt/files/13657462/debug_log_TRV801.txt)

Thank you for your support!

Appoxus, 2023-12-14:
Idk if it helps, but here's the Manual. It seems to be possible to switch between energy saving and comfort mode (PID controlled).

The TRVs seem to be doing their Job OK atm. Only over night the valves dropped down to 5 degrees, I think its related to the schedule.

[image]

[image]

Appoxus, 2023-12-15:
Something seems to be wrong with the calibration offset. It randomly jumps to -429496721.5
I'm using this Blueprint to control the valves:
[https://community.home-assistant.io/t/advanced-heating-control/469873](url)
But I'm seeing the same behavior with Better Thermostat. I can't tell if it's a Z2M error or a failure on the automation side.

Giexperience, 2023-12-16:
Hello, I'm Raul. I have a new TRV device, and I am its manufacturer. The company's name is Giexperience. Can I send you a sample? Could you assist me in creating support for it on '2MQTT (New device support)'?Best regards
GIEXSMART@126.COM

Giexperience, 2023-12-16:
[image]
[image]
[image]
[image]
[image]

Giexperience, 2023-12-16:
It’s this TRV. can you help me?

Koenkk, 2023-12-16:
@Giexperience does it use the same protocol?

Appoxus, 2023-12-16:
@Koenkk is it possible to deactivate the schedule?

I'm currerntly setting the schedule for every day as following:

03:00/16.0 03:00/16.0 03:00/16.0 03:00/16.0

The TRV ist set to Manual, but Z2M Shows the mode as auto, See attached screenshot.

[image]

The temperature calibration is also not working as intended. Sometimes it jumps to bogus values like -42000 or something like that, resulting in showing 30 degree measured temperature on the TRV.

Even on those valves that are not controlled by an automation.

Appoxus, 2023-12-16:
Temperature calibration went off again, see attached screenshot.

[image]

Koenkk, 2023-12-17:
@Appoxus I know little about the behaviour of this device (don't have it)

Giexperience, 2023-12-18:
> @Giexperience它使用相同的协议吗？

Yes. They use the same protocol and the same firmware.

Giexperience, 2023-12-18:
> @Giexperience它使用相同的协议吗？

They are Tuya ZigBee. But we are different manufacturers. The firmware is different.

chrigu1981, 2024-05-29:
is this device now supported? i paired it successfully but z2m says unsupported.

Giexperience, 2024-05-30:
Hello,

Thank you for reaching out.

Many users have successfully used this device with Z2M. I apologize, as I am not an expert in this area. However, here is a post from another user that might be helpful to you: https://github.com/Koenkk/zigbee-herdsman-converters/pull/4929#issuecomment-1996569692

I hope this helps. Please let me know if you need any further assistance.

Best regards,

GIEXSMART-Raul

| |
GIEX-CINLIN-RAUL
|
|
***@***.***
|
---- Replied Message ----
| From | Chrigu ***@***.***> |
| Date | 5/29/2024 20:48 |
| To | ***@***.***> |
| Cc | ***@***.***> ,
***@***.***> |
| Subject | Re: [Koenkk/zigbee2mqtt] [New device support]: Moes ZTRV-801-EN (Issue #20189) |

is this device now supported? i paired it successfully but z2m says unsupported.

maksbit, 2024-08-23:
Has anyone managed to connect an external converter for this TRV valve? I also bought this model: but it does not support Z2M.

maksbit, 2024-09-01:
> ### Link
> https://de.aliexpress.com/item/1005006194184981.html?src=google&aff_fcid=d9fb02d4926e4ccfb968ddb2709d14fa-1701894068736-04266-UneMJZVf&aff_fsk=UneMJZVf&aff_platform=aaf&sk=UneMJZVf&aff_trace_key=d9fb02d4926e4ccfb968ddb2709d14fa-1701894068736-04266-UneMJZVf&terminal_id=37aa72aa1a514423b2841296bbaae943&afSmartRedirect=y
> 
> ### Database entry
> {"id":11,"type":"EndDevice","ieeeAddr":"0xa4c1389685b118e8","nwkAddr":21500,"manufId":4417,"manufName":"_TZE200_rtrmfadk","powerSource":"Battery","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[4,5,61184,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"\u000b{\u000b-i\f{\u000b-i\r{\u000b-i\u000e{\u000b-iS~\u000b-iU~\u000b-iV~\u000b-i","65506":56,"65508":0,"65534":0,"modelId":"TS0601","manufacturerName":"_TZE200_rtrmfadk","powerSource":3,"zclVersion":3,"appVersion":72,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":72,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":-708457359},"lastSeen":1702447427274,"defaultSendRequestWhen":"immediate"}
> 
> ### Comments
> ![2023-12-13 08_58_14-Moes tuya smart zigbee thermostat isches kühler ventil trv programmier bare app ](https://private-user-images.githubusercontent.com/50029453/290116595-efea2bef-e730-4c07-bb00-413c9e1d98c7.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjUyMjA5NTUsIm5iZiI6MTcyNTIyMDY1NSwicGF0aCI6Ii81MDAyOTQ1My8yOTAxMTY1OTUtZWZlYTJiZWYtZTczMC00YzA3LWJiMDAtNDEzYzllMWQ5OGM3LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA5MDElMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwOTAxVDE5NTczNVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWQ1NjIyNmU4MTVhNjQxY2M1MGMyMzdlMTFiZjAyY2U2MzkxYjI4MmFhNGI2ZDhlN2Q0YTYyOTBkZmJhMzA3ZDImWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.zAdy39Yk41OwnNIMQSY3v3e0lGwTYGwHQif8xwuZDLk)
> 
> I bought some of these on Aliexpress and they get detected as TS0601 by Z2M. They're a ripoff of the Bosch Thermostat and there's a Wifi version as well.
> 
> Setting the temperature, child lock and window mode via HA seems to work, the schedule won't work. The Debug log throws some errors. Could you please have a look at this?
> 
> Here's the Debug log after pairing and trying to set temperatures and the schedule via HA. FYI I wanted to remove the schedule, since I plan to do the scheduling via HA.
> 
> [debug_log_TRV801.txt](https://github.com/Koenkk/zigbee2mqtt/files/13657462/debug_log_TRV801.txt)
> 
> Thank you for your support!
> 
> ### External converter
> _No response_
> 
> ### Supported color modes
> _No response_
> 
> ### Color temperature range
> _No response_

Hello, did you manage to run this thermostat in Z2M?

Murada99, 2024-10-02:
I use the following external converter https://github.com/Koenkk/zigbee2mqtt/issues/20476
You have to replace "_TZE200_rtrmfadk" by "_TZE204_rtrmfadk"

Here are the available states:
{
    "battery": 97,
    "child_lock": "LOCK",
    "current_heating_setpoint": 9,
    "last_seen": "2024-10-02T10:30:03.943Z",
    "linkquality": 80,
    "local_temperature": 18,
    "local_temperature_calibration": 0,
    "max_temperature": 22,
    "min_temperature": 9,
    "mode": "eco",
    "position": 0,
    "preset": "manual",
    "running_state": "idle",
    "schedule_friday": "06:00/9.0 12:00/9.0 18:00/9.0 22:00/9.0",
    "schedule_monday": "06:00/9.0 12:00/9.0 18:00/9.0 22:00/9.0",
    "schedule_saturday": "06:00/9.0 12:00/9.0 18:00/9.0 22:00/9.0",
    "schedule_sunday": "06:00/9.0 12:00/9.0 18:00/9.0 22:00/9.0",
    "schedule_thursday": "06:00/9.0 12:00/9.0 18:00/9.0 22:00/9.0",
    "schedule_tuesday": "06:00/9.0 12:00/9.0 18:00/9.0 22:00/9.0",
    "schedule_wednesday": "06:00/9.0 12:00/9.0 18:00/9.0 22:00/9.0",
    "screen_orientation": "down",
    "system_mode": "auto"
}

maksbit, 2024-10-02:
> Я использую следующий внешний конвертер [#20476.](https://github.com/Koenkk/zigbee2mqtt/issues/20476) Вам нужно заменить "_TZE200_rtrmfadk" на "_TZE204_rtrmfadk"
> 
> Вот доступные состояния: { "battery": 97, "child_lock": "LOCK", "current_heating_setpoint": 9, "last_seen": "2024-10-02T10:30:03.943Z", "linkquality": 80, "local_temperature": 18, "local_temperature_calibration": 0, "max_temperature": 22, "min_temperature": 9, "mode": "eco", "position": 0, "preset": "manual", "running_state": "idle", "schedule_friday": "06:00/9.0 12:00/9.0 18:00/9.0 22:00/9.0", "schedule_monday": "06:00/9.0 12:00/9.0 18:00/9.0 22:00/9.0", "schedule_saturday": "06:00/9.0 12:00/9.0 18:00/9.0 22:00/9.0", "schedule_sunday": "06:00/9.0 12:00/9.0 18:00/9.0 22:00/9.0", "schedule_thursday": "06:00/9.0 12:00/9.0 18:00/9.0 22:00/9.0", "schedule_tuesday": "06:00/9.0 12:00/9.0 18:00/9.0 22:00/9.0", "schedule_wednesday": "06:00/9.0 12:00/9.0 18:00/9.0 22:00/9.0", "ориентация_экрана": "вниз", "режим_системы": "авто" }

I tried this external converter, also replaced it with **_TZE204_9mjy74mp**. But it still doesn't want to support the device. I deleted and added it many times.
[image]

maksbit, 2024-10-02:
> I use the following external converter #20476 You have to replace "_TZE200_rtrmfadk" by "_TZE204_rtrmfadk"
>
> Here are the available states: { "battery": 97, "child_lock": "LOCK", [...]

I tried this external converter, also replaced it with **_TZE204_9mjy74mp**. But it still doesn't want to support the device. I deleted and added it many times.
[image]

Murada99, 2024-10-05:
Strange !
I have this model:
https://fr.aliexpress.com/item/1005006068601747.html

It seems to be the same model as yours, but as you can see the zigbee manufacturer is _TZE204_rtrmfadk

[image]

maksbit, 2024-10-05:
> Strange ! I have this model: https://fr.aliexpress.com/item/1005006068601747.html
>
> It seems to be the same model as yours, but as you can see the zigbee manufacturer is _TZE204_rtrmfadk [...]

yes, I have exactly the same model! But yes, it is strange that I change the model name and the converter simply does not pull up.

copernic-us, 2024-10-21:
I have the same model but no support at z2m for now. Dunno if I should return it to seller

DonaldChung-HK, 2024-10-23:
Hi, I have made a converter script [z2m-moes-trv-801-converter](https://github.com/DonaldChung-HK/z2m-moes-trv-801-converter/) for this valve and eveything should be functional. I think they changed `manufacturerName` for different batch which have a slightly different firmware but should be fine if you just change it to match your valve.

Murada99, 2024-11-03:
@DonaldChung-HK , thanks ! Your script works well. The state 'mode' was missing in my old script.

4ug3r, 2024-11-28:
> Hello, can someone please make a video manual to make this TRV work. This one of mine is supported in Z2M, but as TRV601. Unfortunately this doesn't work. The explanation via a converter is not clear to me how to get started. They are very solid TRVs and it would be a shame not to be able to use them. Thank you in advance.

Tutorial: Adding a Custom Converter for Your Moes TRV Device

    Save the File
    Download the file from the following link: https://raw.githubusercontent.com/DonaldChung-HK/z2m-moes-trv-801-converter/refs/heads/main/moes_valve_ZTRV801.js

    Edit the File
    Open the file and modify the manufacturerName field to match your device.
    By default, it is set as: manufacturerName: '_TZE200_9mjy74mp'

Change it to the manufacturerName of your device. For example, in my case, it is: manufacturerName: '_TZE204_rtrmfadk'

How to Find Your Manufacturer Name

    Pair your TRV device with Zigbee2MQTT.
    Open Zigbee2MQTT, go to Devices, and select your TRV device.
    Navigate to About > Zigbee Manufacturer to find the correct manufacturerName.

Save the Edited File
Save the modified moes_valve_ZTRV801.js file to the following location on your Zigbee2MQTT setup:

/config/zigbee2mqtt/external_converters

Add the Converter in Zigbee2MQTT

    Open Zigbee2MQTT, go to Settings > External Converters.
    Click Add New One (+) and enter the path to your file:

/config/zigbee2mqtt/external_converters/moes_valve_ZTRV801.js

Restart Zigbee2MQTT
After adding the converter, restart Zigbee2MQTT to apply the changes.

mwargocki, 2024-12-05:
I bought two of these valves from the same seller. They look the same, but the instructions are slightly different and so is the valve's behavior. One: powered by tuya, second: works with tuya. For the first one, this script works great, but with the second one, the values ​​are not displayed correctly and when I try to set some data in zigbee2mqtt nothing happened.

moonfall84, 2024-12-06:
> Hi, I have made a converter script [z2m-moes-trv-801-converter](https://github.com/DonaldChung-HK/z2m-moes-trv-801-converter/) for this valve and eveything should be functional. I think they changed [...]

I'm testing your script. For now looks OK and I my device is visible.
I just had to change the code with name:
from: _TZE200_9mjy74mp'
to: _TZE204_qyr2m29i
because I checked the log and first equipment was with that id/name. 

[2024-12-06 13:56:49] warning: 	z2m: Device '0xa4c1384e869b9992' with Zigbee model 'TS0601' and manufacturer name '_TZE204_qyr2m29i' is NOT supported, please follow https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html

Need to check next devices.

moonfall84, 2024-12-06:
> > > Hi, I have made a converter script [z2m-moes-trv-801-converter](https://github.com/DonaldChung-HK/z2m-moes-trv-801-converter/) for this valve and eveything should be functional. I think they changed `manufacturerName` for different batch which have a slightly different firmware but should be fine if you just change it to match your valve.
> > 
> > 
> > I'm testing your script. For now looks OK and I my device is visible. I just had to change the code with name: from: _TZE200_9mjy74mp' to: _TZE204_qyr2m29i because I checked the log and first equipment was with that id/name.
> > [2024-12-06 13:56:49] warning: z2m: Device '0xa4c1384e869b9992' with Zigbee model 'TS0601' and manufacturer name '_TZE204_qyr2m29i' is NOT supported, please follow https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html
> > Need to check next devices.
> > EDIT: I have to add that I can manage the brightness level only from Home Assistant. All other functions are not corresponding. I think also that the window sensor is not included in that device right? So why it's in the code?
> 
> Have both versions, and for: _TZE200_9mjy74mp - script works for: _TZE204_qyr2m29i - script doesn't work
> 
> If someone has a solution for that it would be great.

I think you have make a copy of "moes_valve_ZTRV801.js" (e.g. moes_valve_ZTRV801v2.js) and then:
change the "_TZE200_9mjy74mp" for "_TZE204_qyr2m29i" in the file

    fingerprint: [                                                                                                                                               
        {                                                                                                                                                        
            // The model ID from: Device with modelID 'TS0601' is not supported                                                                                  
            // You may need to add \u0000 at the end of the name in some cases                                                                                   
            modelID: 'TS0601',                                                                                                                                   
            // This can be different as they change it for different batches                                                                      
            manufacturerName: '_TZE204_qyr2m29i',                                                                                                                
        },  
        
 Because you will have 2 files it's needed to add also one line in \config\zigbee2mqtt\configuration.yaml:
external_converters:
  - data/moes_valve_ZTRV801.js
  - data/moes_valve_ZTRV801v2.js

mwargocki, 2024-12-06:
Yes, 
I've added two files with different manufacturerName but it doesn't work for _TZE204_qyr2m29i. This valve has different settings.

moonfall84, 2024-12-06:
> Yes, 
> 
> I've added two files with different manufacturerName but it doesn't work for _TZE204_qyr2m29i. This valve has different settings than _TZE200_9mjy74mp. 

Write me a message to moonfall84@gmail.com

pimvanelsen, 2024-12-20:
[pjyzigbee 801.pdf](https://github.com/user-attachments/files/18207869/pjyzigbee.801.pdf)
[ZigBee Smart TRV_10_23.xlsx](https://github.com/user-attachments/files/18207875/ZigBee.Smart.TRV_10_23.xlsx)

received the DPIDs and UART protocol from vendor (for my version "TS0601" manufacturer "_TZE204_qyr2m29i". hopes this helps

moonfall84, 2025-01-04:
Great it works now 😀 
Does anyone find out how to replace the internal temperature sensor by external one (e.g. Xiaomi temp/himidity device)?

copernic-us, 2025-01-04:
> Great it works now 😀 
> Does anyone find out how to replace the internal temperature sensor by external one (e.g. Xiaomi temp/himidity device)?

Have You tried Advanced climate control? You can setup external sensor in it.

peletonduwer, 2025-01-05:
TRV801 is apparently now supported in the new version of zigb2mqtt, but it still shows up as TRV601 and still doesn't work.

https://www.zigbee2mqtt.io/devices/TRV801.html

moonfall84, 2025-01-05:
> TRV801 is apparently now supported in the new version of zigb2mqtt, but it still shows up as TRV601 and still doesn't work.
> 
> 
> 
> https://www.zigbee2mqtt.io/devices/TRV801.html

For me works perfectly to be honest. 

And it's recognized as TRV801 with new picture etc.
Check again your configuration and delete all manual inputs you made before.

peletonduwer, 2025-01-05:
> For me works perfectly to be honest.
>
> And it's recognized as TRV801 with new picture etc. Check again your configuration and delete all manual inputs you made before.

I didn't do any manual inputs at all. I simply connected with zigbee2mqtt and when it didn't work I simply removed the device and waited for the zigbee2mqtt update as I did now and then reconnected. Result, it is redisplayed as TRV601

mwargocki, 2025-01-05:
I have 2 models of this valve: 
_TZE204_9mjy74mp (discovered as TRV801 model (TS0601 zigbee model))
and 
_TZE204_qyr2m29i (discovered as TRV801Z model (TS0601 zigbee model))

Everything works since I've updated to z2m 2.0.0

moonfall84, 2025-01-05:
Now I'm thinking that maybe there is another model/version of this device? Because both that you have (I have the second version only) are supported right now.

mwargocki, 2025-01-05:
It is very likely that there is a third one. Both of my models require different converters (they could not work on the same one due to different mapping). I see that someone has already raised the topic and I saw the model: _TZE204_rtrmfadk, but I did not read much because mine are working at the moment.

moonfall84, 2025-01-05:
@peletonduwer -
Can you check what is your "code" version of your device?

peletonduwer, 2025-01-05:
> @peletonduwer - Can you check what is your "code" version of your device?

On the TRV it says TRV801Z and in zigbee2mqtt _TZE204_rtrmfadk (TSO601)
[image]
[image]

moonfall84, 2025-01-05:
> > @peletonduwer - Can you check what is your "code" version of your device?
> 
> 
> 
> On the TRV it says TRV801Z  TRV 241016801Z and in zigbee2mqtt _TZE204_rtrmfadk (TSO601)
> 
> [image]
> 
> [image]
> 
> 

That's why it's not supported yet :/it's third version as I thought.

peletonduwer, 2025-01-23:
Anyone have an idea to get this third version working? 

On the TRV it says TRV801Z TRV 241016801Z and in zigbee2mqtt _TZE204_rtrmfadk (TSO601) , see above

copernic-us, 2025-01-23:
> Anyone have an idea to get this third version working? 
> 
> On the TRV it says TRV801Z TRV 241016801Z and in zigbee2mqtt _TZE204_rtrmfadk (TSO601) , see above

Have You tried external converter with changed fingerprint?

hradocan, 2025-01-24:
My MOES TRV801 discovered wrongly as: Zigbee Model TS0601 Zigbee Manufacturer _TZE200_rtrmfadk on fresh installation [2.0.0](https://github.com/Koenkk/zigbee2mqtt/releases/tag/2.0.0) commit: [060ae99cff715cc9ef6a0cfa7cae0cdcb17f0c8a](https://github.com/Koenkk/zigbee2mqtt/commit/060ae99cff715cc9ef6a0cfa7cae0cdcb17f0c8a)

In tuya.ts:
     {
        fingerprint: tuya.fingerprint('TS0601', ['_TZE204_9mjy74mp', **'_TZE200_rtrmfadk'**, '_TZE200_9mjy74mp']),
        model: 'TRV602',
        vendor: 'Tuya',
        description: 'Thermostatic radiator valve.',
        whiteLabel: [tuya.whitelabel('Moes', 'TRV801', 'Thermostatic radiator valve', ['_TZE204_9mjy74mp', '_TZE200_9mjy74mp'])],
        onEvent: tuya.onEventSetLocalTime,
        fromZigbee: [tuya.fz.datapoints],
        toZigbee: [tuya.tz.datapoints],

Maybe could be solved by external convertor, but I don't know how.

peletonduwer, 2025-01-25:
> My MOES TRV801 discovered wrongly as: Zigbee Model TS0601 Zigbee Manufacturer _TZE200_rtrmfadk on fresh installation [2.0.0](https://github.com/Koenkk/zigbee2mqtt/releases/tag/2.0.0) commit: [...]

Hopefully there will be a zigbee2mqtt update containing the third version

rozwell, 2025-01-26:
I also have 4x **TRV801Z**  / ***_TZE204_rtrmfadk*** detected as **TRV601** but they seem to be working rather fine, with minor inconveniences like missing screen brightness or manual mode shown as auto. Schedule seems to be working fine.

[image]

peletonduwer, 2025-01-26:
> I also have 4x **TRV801Z** / _**_TZE204_rtrmfadk**_ detected as **TRV601** but they seem to be working rather fine, with minor inconveniences like missing screen brightness or manual mode shown as auto. Schedule seems to be working fine.
> 
> [image]

It doesn't work here, only on and off works. the thermostat dashboard and better thermostat do not work.

rozwell, 2025-01-26:
> It doesn't work here, only on and off works. the thermostat dashboard and better thermostat do not work.

I'm running Home Assistant 2025.1.4 on Raspberry Pi 5 with ConBee 3.
Zigbee2MQTT 2.0.0-2
Mosquitto broker 6.5.0

My entire setup is only a few weeks old and valves worked out of the box as TRV601 from the beginning.
Maybe that will give you a hint?

peletonduwer, 2025-02-07:
Everything is up to date, but the TRV801Z still does not work. It says supported, but doesn't respond to commands
My MOES TRV801Z discovered wrongly as: Zigbee Model TS0601 Zigbee Manufacturer _TZE200_rtrmfadk

jamiehook, 2025-02-25:
I also am facing this issue. TRV801Z detected as TRV601

rozwell, 2025-04-27:
So it's still waiting for a proper support...
I'm using only 3 out of 4 units, so when I'll find some extra time, I'll see what I can do 😉

peletonduwer, 2025-04-28:
indeed, still no support for this type (TRV801Z / _TZE204_rtrmfadk). It is listed as supported, but no command works.

closed 2025-07-05 by github-actions[bot]

mcanteri, 2025-08-19:
This is a pity because the valve is really good and works smoothly and silently.

rozwell, 2025-09-05:
I just noticed that after recent update these new features showed up:
- Display brightness: works
- Screen orientation: works
- Switch deviation eco: didn't have a chance to test it

DarkFeniXZp, 2025-09-10:
TRV801Z does not work correctly. As I understand, Preset includes not only off, on and auto, but also antifrost from Frost protection, and eco and comfort from Hysteresis. When I try to control the device, eco and comfort are constantly set in Preset by themselves.