Koenkk/zigbee2mqtt #20196: Unable to push "set" to device getting error message

pdefoy, 2023-12-13:
### What happened?

After upgdating too latest version, I am no longer able to "set" configurations to devices.
Example: when trying to set the trigger_indicator on an Aqara motion sensor, I get the following error.
Publish 'set' 'trigger_indicator' to 'Motion 4' failed: 'Error: Expected one of: true, false, got: 'true''

This is working on version 1.33.2

### What did you expect to happen?

The config change gets published to the device

### How to reproduce it (minimal and precise)

Change the value of an exposed config on a device

### Zigbee2MQTT version

1.34.0

### Adapter firmware version

0x26580700

### Adapter

Conbee2

### Debug log

Publish 'set' 'trigger_indicator' to 'Motion 4' failed: 'Error: Expected one of: true, false, got: 'true''

tropicalappspoland, 2023-12-13:
Same here:
Publish 'set' 'trigger_indicator' to 'czujka' failed: 'Error: Expected one of: true, false, got: 'true''

Current version: 1.34.0-1

I use Aqara motion sensor P1.

Koenkk, 2023-12-13:
Fixed!

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

closed 2023-12-13 by Koenkk

pdefoy, 2023-12-13:
As always, exceptional support and turn around time.
Thanks @Koenkk