Koenkk/zigbee2mqtt #20202: Failed to configure NodOn SIN-4-1-20 / failed (Status 'UNSUPPORTED_ATTRIBUTE')

merlinpimpim, 2023-12-13:
### What happened?

This device doesn't work well with Z2M, I regularly get this error message:
`Zigbee2MQTT:error 2023-13-12 20:59:09: Failed to configure 'ALLÉE - Portail - Module Nodon', attempt 1 (Error: Read 0x9035eafffed8d440/1 genOnOff(["nodonTransitionTime"], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE') at Endpoint.checkStatus (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:421:28) at Endpoint.read (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:563:22) at processTicksAndRejections (node:internal/process/task_queues:96:5) at Object.configure (/app/node_modules/zigbee-herdsman-converters/src/devices/nodon.ts:55:13) at Configure.configure (/app/lib/extension/configure.ts:117:13) at Immediate.<anonymous> (/app/lib/extension/configure.ts:69:17))`

It's this reference for information:
https://www.zigbee2mqtt.io/devices/SIN-4-1-20.html

And maybe there's a specific problem with this model because Z2M constantly offers a higher OTA fw but it never applies and the official Android Nodon app, that works via Bluetooth, says I already have the latest firmware:
[image]

### What did you expect to happen?

No error ! :)

### How to reproduce it (minimal and precise)

Nothing special, the error message appears regularly once the module has been paired.

### Zigbee2MQTT version

1.34.0-1

### Adapter firmware version

7.3.1.0 build 0

### Adapter

Skyconnect

cropse, 2023-12-14:
Same or similiar issue I believe:

## Zigbee2MQTT version
1.34.0-1

## Adapter firmware version
Sonoff ZHABridge
Tasmota 13.3.0

## Adapter
Ewelink 

## Debug log
```
Failed to configure '0xf082c0fffe18b702', attempt 3 (TypeError: Cannot read properties of null (reading 'configure') at Object.getConfigureKey (/app/node_modules/zigbee-herdsman-converters/src/lib/configureKey.ts:19:21) at Configure.configure (/app/lib/extension/configure.ts:120:45) at runMicrotasks (<anonymous>) at processTicksAndRejections (node:internal/process/task_queues:96:5))
```

## Reference device:
https://www.zigbee2mqtt.io/devices/TM-YKQ004.html

Koenkk, 2023-12-15:
Should already be fixed in the dev branch.

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

closed 2023-12-15 by Koenkk

merlinpimpim, 2024-01-08:
Z2M 1.35.1:
--> Seems to have been resolved, except for the problem of the phantom firmware (543) that doesn't exist in real life (the official Android Nodon app, that works via Bluetooth, says I already have the latest firmware for this Nodon module).

merlinpimpim, 2024-01-08:
Just in case this Nodon SIN-4-1-20 module is also the one mentioned here: https://github.com/Koenkk/zigbee2mqtt/issues/20253

merlinpimpim, 2024-01-09:
Warning 2024-01-09 15:16:26 OTA: Firmware on '0x9035eafffed8d440' is newer than latest firmware online.