Koenkk/zigbee2mqtt #20206: Route is ignored, devices always connect to coordinator

Stooovie, 2023-12-14:
### What happened?

I have a network of roughly 40 devices, with 15 routers (Ikea, Tuya, Aqara). One of my Ikea LED2003G10 bulbs is flaky, only sporadically responding. Map says it's connected directly to the Slae.sh CC2562 stick like 12 meters away. 

I removed it from the network and paired via one of the other Ikea bulbs mid-way. But Map shows it's still connected directly to the coordinator. I did tap the specific "Permit join (Dining Ikea)" button, not `permit join (all)`. Tried multiple times with multiple routers, same thing.

### What did you expect to happen?

A new device to connect to a specified router, not the coordinator.

### Zigbee2MQTT version

1.34.0-1 

### Adapter firmware version

20220219

### Adapter

Slae'sh 2652RB

danieledwardgeorgehitchcock, 2023-12-14:
Zigbee is self-healing therefore, it will pick the route it things best.

This is not something that is controlled by Z2M and you have ruled out device incompatibility by using same-manufacturer routers.

Stooovie, 2023-12-14:
I understand, but then why is the option to pair with a specific router there?

danieledwardgeorgehitchcock, 2023-12-14:
This is just to assist with the initial pairing of the device and can sometimes help with devices which break the Zigbee standard (i.e. TuYa)

closed 2024-07-12 by github-actions[bot]