Koenkk/zigbee2mqtt #20211: Bosch Twintguard was not paired

boldiebold, 2023-12-14:
### What happened?

z2m work so far on my HA Yellow device.
I was able to pair 106 devices. (73 routers, 33 end-devices)

Now I´m fighting with my Bosch Twintguards.
- Installation code was added.
- Paring was activated just for coordinator

Device joined, interview was started but every time it will stop with an error successful paring not possible.
Twintguard were factory reseted.

### Zigbee2MQTT version

1.34.0

### Adapter firmware version

7.3.1.0 build 176

### Adapter

EZSP v12

### Debug log

[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/13674353/log.txt)

Mikkooooo, 2023-12-14:
> ### What happened?
> z2m work so far on my HA Yellow device. I was able to pair 106 devices. (73 routers, 33 end-devices)
>
> Now I´m fighting with my Bosch Twintguards.
>
> * Installation code was added.
> * [...]

Hi, I have these problems too with the Sonoff Dongle E.
Some members write here that it is the Sonoff Dongle E that causes the issues.
 Last week I successfully paired a Bosch BSD-2 smoke detector, also with installation code.
I tried everything possible to pair the Twinguard - all without success.
Do you also have the dongle E (eszp v12)?
Do you have the FW on?
7.3.1.0 build 176 flashed?

danieledwardgeorgehitchcock, 2023-12-14:
Duplicate of #20169

vaenror, 2024-01-02:
Same Problem here with a Sonoff Dongle-E. BSD-2 with install code works just fine. I bought the E exclusively to pair devices with install codes and now the TwinGuard doesn’t pair/interview :(

S474N, 2024-06-02:
Also problem with pairing. I have 9 pieces of BSD-2. 8 paired OK, but the last has problem.

```
2024-06-02 18:41:30Interview failed for '0xf082c0fffe0ab753 with error 'Error: Interview failed because can not get node descriptor ('0xf082c0fffe0ab753')'
Error 2024-06-02 18:41:30Failed to interview '0xf082c0fffe0ab753', device has not successfully been paired

```

[image]

closed 2025-01-01 by github-actions[bot]

remidebette, 2025-01-03:
Hi @Nerivec, is this still tracked in https://github.com/Koenkk/zigbee2mqtt/issues/20169 ?