Koenkk/zigbee2mqtt #20212: Sengled E21-N1EA Blue Flash

blakefrain, 2023-12-14:
### What happened?

Using the Sengled E21-N1EA bulbs when you turn them on using the exposes tab of the device it turns blue fist then to white. This also happens any time you turn it on through HA I have three of these bulbs and they all do it. When I make the transition longer I can see the blue color for longer then it will change to white.

### What did you expect to happen?

The bulb to turn on to white.

### How to reproduce it (minimal and precise)

Join Sengled E21-N1EA to zigbee network and turn on the light

### Zigbee2MQTT version

1.34.0

### Adapter firmware version

0x26660700

### Adapter

ConBee2

### Debug log

Debug 2023-12-14 10:09:50Received MQTT message on 'zigbee2mqtt/Bedroom Light 1/set' with data '{"state":"OFF"}'
Debug 2023-12-14 10:09:50Publishing 'set' 'state' to 'Bedroom Light 1'
Info 2023-12-14 10:09:50MQTT publish: topic 'zigbee2mqtt/Bedroom Light 1', payload '{"brightness":71,"color":{"h":42,"hue":42,"s":38,"saturation":38,"x":0.3636,"y":0.3656},"color_mode":"color_temp","color_temp":226,"energy":0.05,"level_config":{"on_level":"previous"},"linkquality":255,"power":4.3,"state":"OFF","update":{"installed_version":-1,"latest_version":-1,"state":"idle"},"update_available":false}'
Debug 2023-12-14 10:09:51Received MQTT message on 'zigbee2mqtt/Bedroom Light 1/set' with data '{"state":"ON"}'
Debug 2023-12-14 10:09:51Publishing 'set' 'state' to 'Bedroom Light 1'
Info 2023-12-14 10:09:51MQTT publish: topic 'zigbee2mqtt/Bedroom Light 1', payload '{"brightness":71,"color":{"h":42,"hue":42,"s":38,"saturation":38,"x":0.3636,"y":0.3656},"color_mode":"color_temp","color_temp":226,"energy":0.05,"level_config":{"on_level":"previous"},"linkquality":255,"power":4.3,"state":"ON","update":{"installed_version":-1,"latest_version":-1,"state":"idle"},"update_available":false}'

evelant, 2023-12-17:
I have a ton of the E21-N1EA bulbs. Unfortunately they're just... finicky. I suspect that they have firmware problems which Sengled will never fix (there's never been an update for them via Sengled's hub AFAIK and they don't publicly publish their firmware files or any change  notes). Mine will often inexplicably fail to respond, or get stuck at a slightly wrong color temperature different from the other ones. They also don't like to route nicely, often refusing to route via strong nearby routers and instead routing directly to the coordinator despite abysmal LQI. Power cycling them sometimes fixes these issues.

You might try giving the affected bulbs a factory reset. It's a pain with these unless they're hooked up to a physical old style toggle switch. You have to power cycle them 10 times _fast_ after which they will cycle through red, green, blue, then are reset and will be ready to pair again. This has fixed issues with the bulbs for me in the past.  If they're not on a dumb toggle switch I've had luck resetting them by twisting them in the fixture quickly to power cycle 10 times. Also if you don't get the power cycles exactly right for the red-green-blue reset sometimes the bulbs go into some other undocumented bulb where they just flash white endlessly.

As for stability in the network these are configured by default to report power usage every 5 seconds in z2m. That leads to a lot of traffic. I went to the reporting tab for all of mine and removed reporting for power usage. That seems to make them a bit more stable and less chatty.

Hope that helps a bit.

blakefrain, 2023-12-17:
So after some messing around and resetting bulbs it looks like the issue is with certain toggle color temp values. I have it set to 370 mireds and that seems to be working I will close out this issue and hope this helps others. Thanks!

closed 2023-12-17 by blakefrain