Koenkk/zigbee2mqtt #20214: Bticino L4027C remains blinking green light after successful pairing

abnd-hub, 2023-12-14:
### What happened?

After successful pairing, the green light remains blinking

### What did you expect to happen?

After pairing, the green light should turn off and a fixed blue light or alternatively no light should appear (depends on the "LED in the dark" configuration).

### How to reproduce it (minimal and precise)

- Unpair the device (the light on device should turn red) if you have paired with bticino gateway
- Click on reset button of device to show green pairing light
- Click on "permit join all" in Zigbee2MQTT in Home Assistant
- The device will appear in the device list of Zigbee2MQTT and is paired successfully
- Click on "permit join all" in Zigbee2MQTT in Home Assistant to disable join
- In my case, green light remain blinking and not fixed

### Zigbee2MQTT version

1.34.0-1

### Adapter firmware version

20210708

### Adapter

ITead Sonoff Zigbee 3.0 Dongle Plus - Model ZBDongle-P

### Debug log

[0x0004740001593fed.log](https://github.com/Koenkk/zigbee2mqtt/files/13678769/0x0004740001593fed.log)

Lollos60, 2023-12-29:
I solved this trying to pair directly to the closest Bticino/Legrand device.
Try to "Allow Pair" from the pull down menu and select the closest Bticino/Legrand device. Reset the blind controller ---> Red light ----> push briefly the configuration button another time ----> green blinking ----> fixed green light.
Try this way several time, should fix the issue that seems to appear only with last firmware versions.

Xavinooo, 2024-01-05:
I have the same problem with all my Legrand 067776A (with firmware 2c and 3c).
The @Lollos60 didn't work in my case.

The shutter switch is working perflectly, but is staying in pairing mode.
When removing/restoring power, they leave the network.

Xavinooo, 2024-01-06:
Found a solution, but only working with v1.34.0 : [https://github.com/Koenkk/zigbee2mqtt/issues/20617](https://github.com/Koenkk/zigbee2mqtt/issues/20617). The repro step should allow you to pair it properly.

abnd-hub, 2024-01-08:
With @Lollos60  answer I solved it.
Below is the detail of what I did:
On Home Assistant go to the Zigbee2MQTT panel.
Deleted all bticino's devices that were causing the problem and then clicked on "Permit Join (All)".
For each Bticino device to be configured you need to do:
- held down reset button until it was disassociated (red light)
- then click again to reset button to show the green light (association mode).
Note: I had to do these two steps several times for each device. On some devices only 2 times, on others even 15 or more times.
Sometimes it associated correctly and with a fixed green light but with an alert on HA in Zigbee2MQTT for device (icon warning on device). In that case I continued with the two steps, resetting and re-associating.
Finally, I would like to point out that sometimes, especially with devices that required more than 15 attempts, the HA crashed and restarted.

closed 2024-08-06 by github-actions[bot]