Koenkk/zigbee2mqtt #20216: [New device support]: Namron Zigbee Thermostat 16A White

ThaProZac, 2023-12-15:
### Link

https://www.namron.com/products/namron-zigbee-termostat-16a-hvit/

### Database entry

{"id":4,"type":"Router","ieeeAddr":"0x70ac08fffe6d0475","nwkAddr":49234,"manufId":4098,"manufName":"Namron AS","powerSource":"Mains (single phase)","modelId":"4512758","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":769,"inClusterList":[0,3,4,5,6,513,516,1794,2820],"outClusterList":[3,10,25],"clusters":{"genOnOff":{"attributes":{"onOff":1}},"genBasic":{"attributes":{"modelId":"4512758","manufacturerName":"Namron AS","appVersion":0,"stackVersion":0,"hwVersion":0}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":0,"stackVersion":0,"hwVersion":0,"dateCode":"20230711","swBuildId":"1.07","zclVersion":8,"interviewCompleted":true,"meta":{},"lastSeen":1702597418555,"defaultSendRequestWhen":"immediate"}

### Comments

Completely new to HA and this Thermostat had way too many options. For now i just added on/off as I need it and leaving tomorrow and need to control a heating pipe to a well due to freezing. 
Tried to get energy readings info from it by copying some info from Namron touch thermostat, but no luck.

### External converter

```shell
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['4512758'], // The model ID from: Device with modelID 'lumi.sens' is not supported.
    model: '4512758', // Vendor model number, look on the device for a model number
    vendor: 'Namron', // Vendor of the device (only used for documentation and startup logging)
    description: 'Thermostat 16A', // Description of the device, copy from vendor site. (only used for documentation and startup logging)
    fromZigbee: [fz.on_off, fz.metering, fz.electrical_measurement], // We will add this later
    toZigbee: [tz.on_off], // Should be empty, unless device can be controlled (e.g. lights, switches).
    exposes: [
        e.switch(),
        e.power(), e.current(), e.voltage(), e.energy(),
        e.enum('floor_sensor_type', ea.ALL, ['10k', '12k', '15k', '22k', '33k', '47k'])
                .withDescription('Type of the external floor sensor.  Default: NTC 10K/25.')] // Defines what this device exposes, used for e.g. Home Assistant discovery and in the frontend
};

module.exports = definition;
```

aregart, 2024-01-12:
Hi 😊
Have you worked more on this device support lately? I have a few of these and can help with testing/verifying etc. But the programming part of this to begin with looks a bit too large for my level.
I see that the latest version of Z2M includes support for the "Touch termostat 2.0" (PR # 6811) and it seems to have (at least in large part) the same functions. Can this be used as a starting point somehow?

ThaProZac, 2024-01-15:
I was not able to get it to work, but several other smart people have been poking at this. 
Updated the config file above, make not that for this to work you need to modify z2m's yaml file aswell. 
The author of this code is still working on it. I gave him the link to this post in hopes he can update, if not I will if i see him updating it.

aregart, 2024-01-15:
Thanks! 
With the updated version showing now, are there some functions not working but basics are ok, or is it just not working in general?

audun-v, 2024-01-16:
I have tested the code and  it's working great for me.

Just do step 2.1 as described here, and paste the  code from the top.
https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html

This part does not work. When I add this to z2m configuration.yaml I just get an error and z2m won't start.
Can someone find out what's wrong with the action_template?

>Make note, for this to work you also need to modify the zigbee2mqtt conf yaml file and add the following after friendly name:
homeassistant:
climate:
action_template: |-
{% if value_json.system_mode != 'off' or value_json.power > 0 %}
{% set values = {'off': 'idle','idle':'idle','heat':'heating','cool':'cooling'} %}
{{ values[value_json.running_mode] }}
{% else %}
off
{% endif %}

It works just fine  without this in the config.

aregart, 2024-01-16:
Thanks - I will test this myself tomorrow. Just the code above as a regular external converter in z2m...

What is the purpose of the code that is intended for configuration.yaml ?

aregart, 2024-01-17:
Thanks again @audun-v and @ThaProZac - the converter seems to work well.
A couple of minor issues I have after some testing:

- Both the "frost" and "window_check" parameters seems to be a binary (0/1) value from the device according to Z2M logs. But the converter expects enabled/disabled. These two device settings are not working in my HA.

- There is an "energy" parameter being sent from the device, but it is apparently not configured in the converter code. This would be great to get to a separate entity to be able to trace energy usage over time.

I could absolutely be wrong here, I am primarily not a programmer. This is just what I understand from the logs and the converter code, and it is my attempt to contribute a bit.

Can this be sorted out easily, or maybe some hint on how to fix?

ThaProZac, 2024-01-29:
Sorry for late replies, notifications was somehow off.
I have no error with the edit in the main z2m conf.yaml file so you must have copied wrong or pasted it in the wrong place maybe? It needs to be after the friendly name and from what the author told me, it's to get the correct state of the thermostat (off, heat, idle).
Some users have had issues with the P mode (regulator) when in use with a timer (on for x min, off for x min) that it suddenly don't turn on again, so be aware if you are testing this. Not had this problem myself as mine is always on atm.

aregart, 2024-01-29:
No worries for the delay :)

I don't really see the connection with what I had issues with, to the code ment for configuration.yaml. I had no issues with the HVAC state or P-mode funcion (regulator not being used in my case, only thermostat mode). I pasted in that code as described and saw the HVAC state reported correctly from then on.

But since last time I commented I actually managed to edit the converter code to successfully bring in power measurement and fix the two sensors which was not working. I can paste my code here later if it useful to someone. It contains only a few minor changes.

Bottom line - got everything working now - thanks again 👍

audun-v, 2024-02-01:
> I have no error with the edit in the main z2m conf.yaml file so you must have copied wrong or pasted it in the wrong place maybe? It needs to be after the friendly name and from what the author told me, it's to get the correct state of the thermostat (off, heat, idle).

I managed to get it working. Wrong identation I guess.

audun-v, 2024-02-01:
> No worries for the delay :)
>
> I don't really see the connection with what I had issues with, to the code ment for configuration.yaml. I had no issues with the HVAC state or P-mode funcion (regulator [...]

Could you please paste your modified code, that would be very helpful. :) Thank you!

DurinMusicspear, 2024-02-01:
> No worries for the delay :)
>
> I don't really see the connection with what I had issues with, to the code ment for configuration.yaml. I had no issues with the HVAC state or P-mode funcion (regulator [...]

I would also like to have the modified code, thank you!

aregart, 2024-02-02:
Here is my converter. Now, my contribution here is minimal, all respect to those who made most of this. I edited "frost" and "window_check" readings, and added energy reading.

There might be more functionality the thermostat supports through zigbee, I have not looked into that yet. But it seems to me that most functions are in place now.

Just to be clear: There is a bit of code that must be added to configuration.yaml (the one inside zigbee2mqtt folder). That is to make the thermostat correctly report active HVAC status (or so I believe). The code can be copied from the first post in this thread, below is a screenshot from my file to show how it should look after pasting in.

[image]

Attempting to attach the converter file itself inside a zip file:

[namron-4512758.zip](https://github.com/Koenkk/zigbee2mqtt/files/14137865/namron-4512758.zip)

DurinMusicspear, 2024-02-02:
Thank you, that works better! The only problem i have now is that the P mode does not work, if i understand correctly it should use the "Pi heating demand" to set the power level manually in the P mode? Have anyone got that working?

aregart, 2024-02-02:
Not really sure about regulator mode, someone else here might answer better.

My understanding of it is like this: Sensor mode must be "P" to engage regulator mode. The setting "Pi heating demand" is a percentage of time when the heating is active. So if it is set at the default at 20, heating will be active 12 minutes each hour (20%).

And then there is the "Regulator" setting (goes from 0 to 30), I don't understand exactly what it does. Attached screenshot from my manual - there must be something wrong with how that is written...

[image]

audun-v, 2024-02-03:
> Here is my converter. Now, my contribution here is minimal, all respect to those who made most of this. I edited "frost" and "window_check" readings, and added energy reading.
> 
> There might be more functionality the thermostat supports through zigbee, I have not looked into that yet. But it seems to me that most functions are in place now.
> 
> Just to be clear: There is a bit of code that must be added to configuration.yaml (the one inside zigbee2mqtt folder). That is to make the thermostat correctly report active HVAC status (or so I believe). The code can be copied from the first post in this thread, below is a screenshot from my file to show how it should look after pasting in.
> 
> ![094713](https://private-user-images.githubusercontent.com/147825883/301787059-9c274c67-c9fa-4a15-9d75-eb60680e56f6.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDY5NDczMjksIm5iZiI6MTcwNjk0NzAyOSwicGF0aCI6Ii8xNDc4MjU4ODMvMzAxNzg3MDU5LTljMjc0YzY3LWM5ZmEtNGExNS05ZDc1LWViNjA2ODBlNTZmNi5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwMjAzJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDIwM1QwNzU3MDlaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT01OGJmMmMxYjFlZmQ1ZjA1Y2M2NTAyOTI0ZGJjNmI1NTI3NjY2ODgzZmI3N2ViODJjMWZlMmJmZjc1ZTQxZjMyJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.9Z_BstBVN3z-giFWGn57gouQuWLM_HIfoGPkIKNVYKM)
> 
> Attempting to attach the converter file itself inside a zip file:
> 
> [namron-4512758.zip](https://github.com/Koenkk/zigbee2mqtt/files/14137865/namron-4512758.zip)

Thank you!
The action template is working now.
I added your modified code as well. The thermostat is working, but I regularly get an error in the z2m logs.
Do you have this as well?

```
Debug 2024-02-03 08:56:30Received Zigbee message from 'Termostat bad nede', type 'attributeReport', cluster 'seMetering', data '{"currentSummDelivered":[0,21447]}' from endpoint 1 with groupID 0
Error 2024-02-03 08:56:30Failed to apply calibration to 'energy': 'energy_precision' is not a number, got string ()
Info 2024-02-03 08:56:30MQTT publish: topic 'zigbee2mqtt/Termostat bad nede', payload '{"abs_max_cool_setpoint_limit":40,"abs_max_heat_setpoint_limit":40,"abs_min_cool_setpoint_limit":5,"abs_min_heat_setpoint_limit":5,"away_mode":"OFF","backlight":10,"child_lock":"LOCK","current":2.06,"display_auto_off_enabled":"disabled","energy":214.47,"fault":"none","floor_sensor_calibration":null,"frost":0,"hysterersis":null,"linkquality":69,"local_temperature":28,"local_temperature_calibration":0,"occupied_cooling_setpoint":20,"occupied_heating_setpoint":28.5,"outdoor_temperature":null,"pi_heating_demand":20,"power":488,"programming_operation_mode":"setpoint","regulator":30,"running_mode":"heat","running_state":null,"sensor_mode":"f","state":"ON","system_mode":"heat","update":{"state":"idle"},"window_check":0,"window_open_check":null,"window_state":"closed"}'
```
[image]

aregart, 2024-02-03:
Sorry, I don't have those errors. I don't know what that means...

DurinMusicspear, 2024-02-03:
> > Here is my converter. Now, my contribution here is minimal, all respect to those who made most of this. I edited "frost" and "window_check" readings, and added energy reading.
> > There might be more functionality the thermostat supports through zigbee, I have not looked into that yet. But it seems to me that most functions are in place now.
> > Just to be clear: There is a bit of code that must be added to configuration.yaml (the one inside zigbee2mqtt folder). That is to make the thermostat correctly report active HVAC status (or so I believe). The code can be copied from the first post in this thread, below is a screenshot from my file to show how it should look after pasting in.
> 
> Thank you! The action template is working now. I added your modified code as well. The thermostat is working, but I regularly get an error in the z2m logs. Do you have this as well?
> 
> ```
> Debug 2024-02-03 08:56:30Received Zigbee message from 'Termostat bad nede', type 'attributeReport', cluster 'seMetering', data '{"currentSummDelivered":[0,21447]}' from endpoint 1 with groupID 0
> Error 2024-02-03 08:56:30Failed to apply calibration to 'energy': 'energy_precision' is not a number, got string ()
> Info 2024-02-03 08:56:30MQTT publish: topic 'zigbee2mqtt/Termostat bad nede', payload '{"abs_max_cool_setpoint_limit":40,"abs_max_heat_setpoint_limit":40,"abs_min_cool_setpoint_limit":5,"abs_min_heat_setpoint_limit":5,"away_mode":"OFF","backlight":10,"child_lock":"LOCK","current":2.06,"display_auto_off_enabled":"disabled","energy":214.47,"fault":"none","floor_sensor_calibration":null,"frost":0,"hysterersis":null,"linkquality":69,"local_temperature":28,"local_temperature_calibration":0,"occupied_cooling_setpoint":20,"occupied_heating_setpoint":28.5,"outdoor_temperature":null,"pi_heating_demand":20,"power":488,"programming_operation_mode":"setpoint","regulator":30,"running_mode":"heat","running_state":null,"sensor_mode":"f","state":"ON","system_mode":"heat","update":{"state":"idle"},"window_check":0,"window_open_check":null,"window_state":"closed"}'
> ```
> 
> ![image](https://private-user-images.githubusercontent.com/56632644/302022313-07409a0f-1c3e-481e-99e7-e6951708f816.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDY5NTMzNTUsIm5iZiI6MTcwNjk1MzA1NSwicGF0aCI6Ii81NjYzMjY0NC8zMDIwMjIzMTMtMDc0MDlhMGYtMWMzZS00ODFlLTk5ZTctZTY5NTE3MDhmODE2LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAyMDMlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMjAzVDA5MzczNVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWJmZDA1NWZlNDMwMzFhMDQ0YTEyNDRjZDg5Y2VjYjMyNTE1Njk5YzYxYTM1Njg0ZjhiMzg5MDU2OGRlZTg0OGUmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.B9TtDqNK98SIAXLRtdLNRJcrYCbQ9PW29rQyc_wt2ow)

Try setting "Energy calibration" to 0:
[image]

DurinMusicspear, 2024-02-03:
> Not really sure about regulator mode, someone else here might answer better.
> 
> My understanding of it is like this: Sensor mode must be "P" to engage regulator mode. The setting "Pi heating demand" is a percentage of time when the heating is active. So if it is set at the default at 20, heating will be active 12 minutes each hour (20%).
> 
> And then there is the "Regulator" setting (goes from 0 to 30), I don't understand exactly what it does. Attached screenshot from my manual - there must be something wrong with how that is written...
> 
> ![Skjermbilde 2024-02-02 145516](https://private-user-images.githubusercontent.com/147825883/301865148-47d8e529-c7db-4c7d-8b15-a1a0b4dfb56d.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDY5NTMzNTUsIm5iZiI6MTcwNjk1MzA1NSwicGF0aCI6Ii8xNDc4MjU4ODMvMzAxODY1MTQ4LTQ3ZDhlNTI5LWM3ZGItNGM3ZC04YjE1LWExYTBiNGRmYjU2ZC5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwMjAzJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDIwM1QwOTM3MzVaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1mMWUwY2Y0NzMzN2M1Yjg4M2JiZmM4M2IzN2VlYjIxZDE5ODQ1NmZhZjM2YWRlMzY3M2RiMjQ2NWI4ODc1NjUyJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.np0KqKLNyUY14PT5nvotnNBg-h8gXGXV_aNyWsoM9UQ)
> 
> Update: Someone suggested that last setting might be what time period the percentage should be calculated from. So the default of 30 would mean the thermostat turns on for 6 min every 30 min. Setting it to 15 will make it turn on for 3 min every 15 min. Given the first setting is still on 20%. This is unverified...

Based on my testing it seems like "Pi heating demand" is the percentage of time the heating is active, and "Regulator" is the timeframe, so if "Regulator" is set to 30 min and "Pi heating demand" is 20% it wil heat for 6 mins and then be idle for 24 mins. And then repeat that cycle every 30 mins.

The problem i have is that i cannot set those configuration values from Z2M, if i try to change them they will just be 0:
[image]

lordradon, 2024-04-08:
> Here is my converter. Now, my contribution here is minimal, all respect to those who made most of this. I edited "frost" and "window_check" readings, and added energy reading.
>
> There might be more [...]

Hi, I've done all this (probably missing something since not working..). Added the text in config..yaml, added the external converter. I feel like everything is in, but it still says unsupported device and only get State, Power, Current in z2m. No way to control the thermostat with heat and temperature.
I have 4 of these, and really need some help getting them to work.

Anything you can see from this error perhaps? Let me know what else I need to provide. Noob at this
[image]

aregart, 2024-04-09:
I have not checked your code from configuration.yaml in detail, but from your log error I think you might have the same issue I got after upgrading Z2M plugin to the latest version recently. Worked fine again after restoring previos version from backup. I got it permanently fixed by removing/disabling one of the lines in the start of the converter file.

Screenshot from my current converter file below, working fine with thermostat controls and statistics. See the one line disabled with double slashes in the beginning.

[image]

lordradon, 2024-04-09:
> I have not checked your code from configuration.yaml in detail, but from your log error I think you might have the same issue I got after upgrading Z2M addon to the latest version recently. Worked fine again after restoring previos version from backup. I got it permanently fixed by removing/disabling one of the lines in the start of the converter file.
> 
> Screenshot from my current converter file below, working fine with thermostat controls and statistics. See the one line disabled with double slashes in the beginning.
> 

Thanks for the reply. I don't have any addons, I'm not using HA, but z2m then into homey (for now anyway)
But I tried removing // the text you said. Restarted z2m, but nothing  really changed for me. 🤷

aregart, 2024-04-09:
I think your errors are different from what I struggled with after my Z2M upgrade, sorry I don't know how to help with this...

lordradon, 2024-04-09:
Thanks for trying anyway 😊

LazyFunker, 2024-04-14:
Here is an updated version.
Regulator percentage works, and I also fixed the running state in a "hacky" way.
The old/legacy extend is also fixed.

```
const {} = require('zigbee-herdsman-converters/lib/modernExtend');
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const constants = require('zigbee-herdsman-converters/lib/constants');
const e = exposes.presets;
const ea = exposes.access;

const utils = require('zigbee-herdsman-converters/lib/utils');
const {
    Zcl
} = require('zigbee-herdsman');

const local = {
    fz: {
        namron_thermostat_new: {
            cluster: 'hvacThermostat',
            type: ['readResponse', 'attributeReport'],
            convert: (model, msg, publish, options, meta) => {
                const result = KeyValue = {};
                const data = msg.data;
                if (data.hasOwnProperty(0x8)) { // regulator_percentage
                    result['pi_heating_demand'] = data[0x8];
                }
                if (data.hasOwnProperty('runningMode')) { // runningMode
                    const lookup = {0: 'off', 3: 'cool', 4: 'heat', 16: 'idle'};
                    result['running_mode'] = utils.getFromLookup(data['runningMode'], lookup);
                    const lookup2 = {0: 'idle', 3: 'cool', 4: 'heat', 16: 'idle'};
                    result['running_state'] = utils.getFromLookup(data['runningMode'], lookup2);
                }
                if (data.hasOwnProperty(0x8000)) { // windowCheck
                    result['window_check'] = data[0x8000];
                }
                if (data.hasOwnProperty(0x8001)) { // frost
                    result['frost'] = data[0x8001];
                }
                if (data.hasOwnProperty(0x8002)) { // windowState
                    result['window_state'] = data[0x8002] ? 'opened' : 'closed';
                }
                if (data.hasOwnProperty(0x8004)) { // sensorMode
                    const lookup = { 0: 'a', 1: 'f', 2: 'af', 3: 'a2', 4: 'a2f', 5: 'fp', 6: 'p'};
                    result['sensor_mode'] = utils.getFromLookup(data[0x8004], lookup);
                }
                if (data.hasOwnProperty(0x8005)) { // backlight
                    result['backlight'] = parseFloat(data[0x8005]);
                }
                if (data.hasOwnProperty(0x8006)) { // fault
                    const lookup = {0: 'none', 1: 'er1', 2: 'er2', 4: 'er3', 8: 'er4', 16: 'er5', 32: 'er6', 64: 'er7', 128: 'er8'};
                    result['fault'] = utils.getFromLookup(data[0x8006], lookup);
                }
                if (data.hasOwnProperty(0x8007)) { // regulator
                    result['regulator'] = data[0x8007];
                }
                return result;
            },
        },
    },

    tz: {
        namron_thermostat_new: {
            key: ['pi_heating_demand', 'running_mode', 'window_check', 'frost', 'window_state',
            'sensor_mode', 'backlight', 'fault', 'regulator'],
            convertSet: async(entity, key, value, meta) => {
                if (key === 'pi_heating_demand') {
[... 177 lines cut ...]
            attribute: {ID: 0x8005, type: 32},
            minimumReportInterval: 5,
            maximumReportInterval: constants.repInterval.HOUR,
            reportableChange: null}],
        options);
        await endpoint.configureReporting('hvacThermostat', [{ // fault
            attribute: {ID: 0x8006, type: 24},
            minimumReportInterval: 0,
            maximumReportInterval: constants.repInterval.HOUR,
            reportableChange: null}],
        options);
        await endpoint.configureReporting('hvacThermostat', [{ // regulator
            attribute: {ID: 0x8007, type: 32},
            minimumReportInterval: 0,
            maximumReportInterval: constants.repInterval.HOUR,
            reportableChange: null}],
        options);

        await endpoint.read('hvacThermostat', ['systemMode', 'runningMode', 'occupiedHeatingSetpoint']);
        await endpoint.read('hvacThermostat', [0x8000, 0x8001, 0x8002, 0x8004, 0x8005, 0x8006, 0x8007]);
    },
};
 
module.exports = definition;
```

lordradon, 2024-04-14:
[image]
How do I know if it actually runs the namron  js? I did what it said here https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html

But I cant get it to show supported (or at least show other than On/Off, Watt. 

@LazyFunker do you mean you dont need this part anymore in config?

> homeassistant:
climate:
action_template: |-
{% if value_json.system_mode != 'off' or value_json.power > 0 %}
{% set values = {'off': 'idle','idle':'idle','heat':'heating','cool':'cooling'} %}
{{ values[value_json.running_mode] }}
{% else %}
off
{% endif %}

LazyFunker, 2024-04-14:
> ![image](https://private-user-images.githubusercontent.com/29264129/322295984-6c37b60e-33ac-4025-b291-6ad97a63eeaa.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3 [...]

Have you also restarted Z2M?
Could try to repair it if that doesn't work.

Yes, you can just have it there, but it should work without it.

lordradon, 2024-04-14:
@LazyFunker Yeah, restarted. Is repair the Reconfigure button? Cant find a direct repair button. 

Any of this mean anything to you?
> Error 2024-04-14 21:04:31Failed to configure 'Basement Bathroom Thermostat', attempt 4 (Error: ZCL command 0x70ac08fffe6cf3ba/1 haElectricalMeasurement.read(["acVoltageDivisor","acVoltageMultiplier"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE') at Endpoint.checkStatus (C:\zigbee2mqtt\node_modules\zigbee-herdsman\src\controller\model\endpoint.ts:319:28) at Endpoint.zclCommand (C:\zigbee2mqtt\node_modules\zigbee-herdsman\src\controller\model\endpoint.ts:762:22) at Endpoint.read (C:\zigbee2mqtt\node_modules\zigbee-herdsman\src\controller\model\endpoint.ts:404:24) at result.configure (C:\zigbee2mqtt\node_modules\zigbee-herdsman-converters\src\lib\modernExtend.ts:1234:29) at Object.configure (C:\zigbee2mqtt\node_modules\zigbee-herdsman-converters\src\index.ts:162:21) at Configure.configure (C:\zigbee2mqtt\lib\extension\configure.ts:122:13) at Configure.onMQTTMessage (C:\zigbee2mqtt\lib\extension\configure.ts:55:21) at EventEmitter.wrappedCallback (C:\zigbee2mqtt\lib\eventBus.ts:174:17))

LazyFunker, 2024-04-14:
@lordradon No, you should probably try to remove it, and then readd it.
I can't see directly what's wrong on that error code no.

lordradon, 2024-04-14:
Same thing happened when re-adding device. Is there any way to tell if the .js file has been opened by zigbee2mqtt?
Just feels like that might be the issue

LazyFunker, 2024-04-14:
Have you replaced the old file with the updated version (And remembered to save)?
Have you also placed that file under: `/homeassistant/zigbee2mqtt/` with the name you have in the configuration.yaml file for Z2M?

lordradon, 2024-04-14:
Wow, in the "Support new devices" page it says

> To extend the generated external definition, save it next to the Zigbee2MQTT configuration.yaml

So I had it in the data folder where I have my config. But tested placing it in the main z2m folder, and that seems to have done the trick :) 

Thanks for troubleshooting with me!

DurinMusicspear, 2024-04-14:
> Here is an updated version. Regulator percentage works, and I also fixed the running state, so no need for the edit in configuration.yaml file. The old/legacy extend is also fixed.
> 
> ```
> const {} = require('zigbee-herdsman-converters/lib/modernExtend');
> const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
> const tz = require('zigbee-herdsman-converters/converters/toZigbee');
> const exposes = require('zigbee-herdsman-converters/lib/exposes');
> const reporting = require('zigbee-herdsman-converters/lib/reporting');
> const constants = require('zigbee-herdsman-converters/lib/constants');
> const e = exposes.presets;
> const ea = exposes.access;
> 
> const utils = require('zigbee-herdsman-converters/lib/utils');
> const {
>     Zcl
> } = require('zigbee-herdsman');
> 
> const local = {
>     fz: {
>         namron_thermostat_new: {
>             cluster: 'hvacThermostat',
>             type: ['readResponse', 'attributeReport'],
>             convert: (model, msg, publish, options, meta) => {
>                 const result = KeyValue = {};
>                 const data = msg.data;
>                 if (data.hasOwnProperty(0x8)) { // regulator_percentage
>                     result['pi_heating_demand'] = data[0x8];
>                 }
>                 if (data.hasOwnProperty('runningMode')) { // runningMode
>                     const lookup = {0: 'off', 3: 'cool', 4: 'heat', 16: 'idle'};
>                     result['running_mode'] = utils.getFromLookup(data['runningMode'], lookup);
>                     const lookup2 = {0: 'idle', 3: 'cool', 4: 'heat', 16: 'idle'};
>                     result['running_state'] = utils.getFromLookup(data['runningMode'], lookup2);
>                 }
>                 if (data.hasOwnProperty(0x8000)) { // windowCheck
>                     result['window_check'] = data[0x8000];
>                 }
>                 if (data.hasOwnProperty(0x8001)) { // frost
>                     result['frost'] = data[0x8001];
>                 }
>                 if (data.hasOwnProperty(0x8002)) { // windowState
>                     result['window_state'] = data[0x8002] ? 'opened' : 'closed';
>                 }
>                 if (data.hasOwnProperty(0x8004)) { // sensorMode
>                     const lookup = { 0: 'a', 1: 'f', 2: 'af', 3: 'a2', 4: 'a2f', 5: 'fp', 6: 'p'};
>                     result['sensor_mode'] = utils.getFromLookup(data[0x8004], lookup);
>                 }
>                 if (data.hasOwnProperty(0x8005)) { // backlight
>                     result['backlight'] = parseFloat(data[0x8005]);
>                 }
>                 if (data.hasOwnProperty(0x8006)) { // fault
>                     const lookup = {0: 'none', 1: 'er1', 2: 'er2', 4: 'er3', 8: 'er4', 16: 'er5', 32: 'er6', 64: 'er7', 128: 'er8'};
>                     result['fault'] = utils.getFromLookup(data[0x8006], lookup);
>                 }
>                 if (data.hasOwnProperty(0x8007)) { // regulator
>                     result['regulator'] = data[0x8007];
>                 }
>                 return result;
>             },
>         },
>     },
> 
>     tz: {
>         namron_thermostat_new: {
>             key: ['pi_heating_demand', 'running_mode', 'window_check', 'frost', 'window_state',
>             'sensor_mode', 'backlight', 'fault', 'regulator'],
>             convertSet: async(entity, key, value, meta) => {
>                 if (key === 'pi_heating_demand') {
>                     const payload = {0x8: {value: value, type: Zcl.DataType.uint8}};
>                     await entity.write('hvacThermostat', payload);
[... 179 lines cut ...]
>         options);
>         await endpoint.configureReporting('hvacThermostat', [{ // fault
>             attribute: {ID: 0x8006, type: 24},
>             minimumReportInterval: 0,
>             maximumReportInterval: constants.repInterval.HOUR,
>             reportableChange: null}],
>         options);
>         await endpoint.configureReporting('hvacThermostat', [{ // regulator
>             attribute: {ID: 0x8007, type: 32},
>             minimumReportInterval: 0,
>             maximumReportInterval: constants.repInterval.HOUR,
>             reportableChange: null}],
>         options);
> 
>         await endpoint.read('hvacThermostat', ['systemMode', 'runningMode', 'occupiedHeatingSetpoint']);
>         await endpoint.read('hvacThermostat', [0x8000, 0x8001, 0x8002, 0x8004, 0x8005, 0x8006, 0x8007]);
>     },
> };
>  
> module.exports = definition;
> ```

This works great, thanks!

I just noticed you are missing the the energy report, just adding e.energy(), seems to fix that.

aregart, 2024-04-15:
Thanks @LazyFunker 👍 The code bits in configuration.yaml is not needed anymore. I also added energy reading like @DurinMusicspear mentioned.

I also had to edit the "window_check" and "frost" readings, as I did in the previous version. I don't know if there is something specific in my system that causes the need for that. The lines under "tz:" part for those two should look like this on both:

`const lookup = {'0': false, '1': true};`

trikinov, 2024-04-23:
Will this device be added to official versjon - and be supported out of the box in near future ?

Gonnar86, 2024-09-03:
> Will this device be added to official versjon - and be supported out of the box in near future ?

Guess I have the same question. I have tried several times doing the above guide, but I get an idiotic syntax error on the copy pasted config.. 
Starting to lose patience with myself.. haha

eireit, 2024-10-02:
Somone can comfirm if the converter is working so we can submit a pull request to [zigbee-herdsman-converters](https://github.com/Koenkk/zigbee-herdsman-converters) so this device is supported out of the box by Zigbee2MQTT. This can be done by adding the definition to the namron.ts file.

aregart, 2024-10-05:
> Somone can comfirm if the converter is working so we can submit a pull request to [zigbee-herdsman-converters](https://github.com/Koenkk/zigbee-herdsman-converters) so this device is supported out of the box by Zigbee2MQTT.

I can confirm it is working fine. One error with the converter is that it gives the thermostat a heat/cool/off selector (screenshot below), but the cooling option should have been removed. The heat and off options turns the thermostat on and off as expected.

[image]

eireit, 2024-10-22:
@aregart, I dont have any readings for power and current. I have readings for energy but that is not available as a entity in HA.
{
    "abs_max_cool_setpoint_limit": 40,
    "abs_max_heat_setpoint_limit": 40,
    "abs_min_cool_setpoint_limit": 5,
    "abs_min_heat_setpoint_limit": 5,
    "backlight": 10,
    "child_lock": "UNLOCK",
    **"current": 0,**
    **"energy": 53.96,**
    "fault": "none",
    "frost": 0,
    "linkquality": 100,
    "local_temperature": 25,
    "local_temperature_calibration": 0,
    "occupied_cooling_setpoint": 26,
    "occupied_heating_setpoint": 25,
    "pi_heating_demand": 20,
    **"power": 0,**
    "programming_operation_mode": "setpoint",
    "regulator": 30,
    "running_mode": "idle",
    "running_state": "idle",
    "sensor_mode": "f",
    "system_mode": "heat",
    "window_check": 1,
    "window_state": "closed"
}

aregart, 2024-10-23:
@eireit - When the thermostat is idle, as you see in the "running_mode" or "running_state" towards the end there, then it is expected to see "current" and "power" to be zero. Those two values are momentary electrical values of what the thermostat is doing. The "energy" counter will increment upwards, counting energy usage when the thermostat is active.

Test this by turning the heat up beyond the temperature in the room, see the thermostat going active with the indicator on the display, and check the changed values in Z2M after a few secounds.

Missing entity in HA _can_ be because some entities are disabled by default. Go to the device in HA and check under controls/sensors/diagnostics and check if there are some sensors that can be enabled. In my example below I have manually enabled "power" but not "current" if I remember correctly what I did a while back...

[image]

eireit, 2024-10-23:
@aregart Hi, my mistake, ofcaurse the power and current is 0 when idle. Then it's only the energy that is missing. I see that you also miss that one.

[image]

aregart, 2024-10-23:
@eireit - I can see the energy sensor under category "sensors" on the device. My screenshot was from "diagnostics" further down. Ideally you can enable yours there too 😊

eireit, 2024-10-23:
@aregart I only have these alternatives under sensors:
[image]

aregart, 2024-10-23:
@eireit - These are mine:

[image]

No idea what to do with that, I'm afraid...
Does the "Energy" data show under "exposes" on the device in Z2M?

[image]

eireit, 2024-10-23:
@aregart no energy is shown under exposes.

[image]

aregart, 2024-10-23:
@eireit - At least I think your thermostat is working as it should, the data is sent correctly since you can see it under "state". I do not see energy under "reporting" no, but I have some "attrib10", "attrib11" etc which I don't know what represents.

Sharing my running converter below. One day I think i will mess about and look at the "cooling" option, just so that option is removed. If that is done properly, I suspect we are only missing an icon before someone can submit a PR for built-in support in Z2M.

```
const {} = require('zigbee-herdsman-converters/lib/modernExtend');
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const constants = require('zigbee-herdsman-converters/lib/constants');
const e = exposes.presets;
const ea = exposes.access;

const utils = require('zigbee-herdsman-converters/lib/utils');
const {
    Zcl
} = require('zigbee-herdsman');

const local = {
    fz: {
        namron_thermostat_new: {
            cluster: 'hvacThermostat',
            type: ['readResponse', 'attributeReport'],
            convert: (model, msg, publish, options, meta) => {
                const result = KeyValue = {};
                const data = msg.data;
                if (data.hasOwnProperty(0x8)) { // regulator_percentage
                    result['pi_heating_demand'] = data[0x8];
                }
                if (data.hasOwnProperty('runningMode')) { // runningMode
                    const lookup = {0: 'off', 3: 'cool', 4: 'heat', 16: 'idle'};
                    result['running_mode'] = utils.getFromLookup(data['runningMode'], lookup);
                    const lookup2 = {0: 'idle', 3: 'cool', 4: 'heat', 16: 'idle'};
                    result['running_state'] = utils.getFromLookup(data['runningMode'], lookup2);
                }
                if (data.hasOwnProperty(0x8000)) { // windowCheck
                    result['window_check'] = data[0x8000];
                }
                if (data.hasOwnProperty(0x8001)) { // frost
                    result['frost'] = data[0x8001];
                }
                if (data.hasOwnProperty(0x8002)) { // windowState
                    result['window_state'] = data[0x8002] ? 'opened' : 'closed';
                }
                if (data.hasOwnProperty(0x8004)) { // sensorMode
                    const lookup = { 0: 'a', 1: 'f', 2: 'af', 3: 'a2', 4: 'a2f', 5: 'fp', 6: 'p'};
                    result['sensor_mode'] = utils.getFromLookup(data[0x8004], lookup);
                }
                if (data.hasOwnProperty(0x8005)) { // backlight
                    result['backlight'] = parseFloat(data[0x8005]);
                }
                if (data.hasOwnProperty(0x8006)) { // fault
                    const lookup = {0: 'none', 1: 'er1', 2: 'er2', 4: 'er3', 8: 'er4', 16: 'er5', 32: 'er6', 64: 'er7', 128: 'er8'};
                    result['fault'] = utils.getFromLookup(data[0x8006], lookup);
                }
                if (data.hasOwnProperty(0x8007)) { // regulator
                    result['regulator'] = data[0x8007];
                }
                return result;
            },
        },
    },

    tz: {
        namron_thermostat_new: {
            key: ['pi_heating_demand', 'running_mode', 'window_check', 'frost', 'window_state',
            'sensor_mode', 'backlight', 'fault', 'regulator'],
            convertSet: async(entity, key, value, meta) => {
                if (key === 'pi_heating_demand') {
[... 177 lines cut ...]
            attribute: {ID: 0x8005, type: 32},
            minimumReportInterval: 5,
            maximumReportInterval: constants.repInterval.HOUR,
            reportableChange: null}],
        options);
        await endpoint.configureReporting('hvacThermostat', [{ // fault
            attribute: {ID: 0x8006, type: 24},
            minimumReportInterval: 0,
            maximumReportInterval: constants.repInterval.HOUR,
            reportableChange: null}],
        options);
        await endpoint.configureReporting('hvacThermostat', [{ // regulator
            attribute: {ID: 0x8007, type: 32},
            minimumReportInterval: 0,
            maximumReportInterval: constants.repInterval.HOUR,
            reportableChange: null}],
        options);

        await endpoint.read('hvacThermostat', ['systemMode', 'runningMode', 'occupiedHeatingSetpoint']);
        await endpoint.read('hvacThermostat', [0x8000, 0x8001, 0x8002, 0x8004, 0x8005, 0x8006, 0x8007]);
    },
};
 
module.exports = definition;
```

eireit, 2024-10-23:
@aregart Thanks, after adding your JS file, the energy is shown.... Thanks...

aregart, 2024-10-23:
@eireit Great to hear 👍

Xine72, 2024-10-23:
Hi. How do I know the external converter file is picked up by the thermostat in Z2M? Had it working a few weeks ago, but suddenly lost all exposures. Also, is there a specific naming structure that needs to be matched between naming of the .js file and some other file?

aregart, 2024-10-23:
@Xine72 The file name must match what you add in Z2M configuration, and the file must be placed in the correct folder. Other than that, I can't help out...

[image]

[image]

stiglinga, 2024-11-19:
There are just a few places needed to remove cool in the code to remove this option.

                if (data.hasOwnProperty('runningMode')) { // runningMode
                    const lookup = {0: 'off', 4: 'heat', 16: 'idle'}; -> **removed 'cool'**
                    result['running_mode'] = utils.getFromLookup(data['runningMode'], lookup);
                    const lookup2 = {0: 'idle', 4: 'heat', 16: 'idle'}; -> **removed 'cool'**

                else if (key === 'running_mode') {
                    const lookup = {'off': 0, 'heat': 4, 'idle': 16}; -> **removed 'cool'**
                    const payload = {'runningMode': {value: utils.getFromLookup(value, lookup), type: Zcl.DataType.enum8}};
                    await entity.write('hvacThermostat', payload);

                    const lookup2 = {'idle': 0, 'heat': 4, 'idle': 16}; -> **removed 'cool'**
                    const payload2 = {'runningState': {value: utils.getFromLookup(value, lookup2), type: Zcl.DataType.enum8}};
                    await entity.write('hvacThermostat', payload2);

    exposes: [
        e.climate()
            .withSetpoint('occupied_heating_setpoint', 5, 40, 0.5)
            .withLocalTemperature()
            .withLocalTemperatureCalibration(-3, 3, 0.1)
            .withSystemMode(['off', 'heat']) -> **removed 'cool'**
            .withRunningState(['idle', 'heat']) -> **removed 'cool'**

Here is an update of the converter with cool removed.

`const {} = require('zigbee-herdsman-converters/lib/modernExtend');
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const constants = require('zigbee-herdsman-converters/lib/constants');
const e = exposes.presets;
const ea = exposes.access;

const utils = require('zigbee-herdsman-converters/lib/utils');
const {
    Zcl
} = require('zigbee-herdsman');

const local = {
    fz: {
        namron_thermostat_new: {
            cluster: 'hvacThermostat',
            type: ['readResponse', 'attributeReport'],
            convert: (model, msg, publish, options, meta) => {
                const result = KeyValue = {};
                const data = msg.data;
                if (data.hasOwnProperty(0x8)) { // regulator_percentage
                    result['pi_heating_demand'] = data[0x8];
                }
                if (data.hasOwnProperty('runningMode')) { // runningMode
                    const lookup = {0: 'off', 4: 'heat', 16: 'idle'};
                    result['running_mode'] = utils.getFromLookup(data['runningMode'], lookup);
                    const lookup2 = {0: 'idle', 4: 'heat', 16: 'idle'};
                    result['running_state'] = utils.getFromLookup(data['runningMode'], lookup2);
                }
                if (data.hasOwnProperty(0x8000)) { // windowCheck
                    result['window_check'] = data[0x8000];
                }
                if (data.hasOwnProperty(0x8001)) { // frost
                    result['frost'] = data[0x8001];
                }
                if (data.hasOwnProperty(0x8002)) { // windowState
                    result['window_state'] = data[0x8002] ? 'opened' : 'closed';
                }
                if (data.hasOwnProperty(0x8004)) { // sensorMode
                    const lookup = { 0: 'a', 1: 'f', 2: 'af', 3: 'a2', 4: 'a2f', 5: 'fp', 6: 'p'};
                    result['sensor_mode'] = utils.getFromLookup(data[0x8004], lookup);
                }
[... 197 lines cut ...]
        await endpoint.configureReporting('hvacThermostat', [{ // backlight
            attribute: {ID: 0x8005, type: 32},
            minimumReportInterval: 5,
            maximumReportInterval: constants.repInterval.HOUR,
            reportableChange: null}],
        options);
        await endpoint.configureReporting('hvacThermostat', [{ // fault
            attribute: {ID: 0x8006, type: 24},
            minimumReportInterval: 0,
            maximumReportInterval: constants.repInterval.HOUR,
            reportableChange: null}],
        options);
        await endpoint.configureReporting('hvacThermostat', [{ // regulator
            attribute: {ID: 0x8007, type: 32},
            minimumReportInterval: 0,
            maximumReportInterval: constants.repInterval.HOUR,
            reportableChange: null}],
        options);

        await endpoint.read('hvacThermostat', ['systemMode', 'runningMode', 'occupiedHeatingSetpoint']);
        await endpoint.read('hvacThermostat', [0x8000, 0x8001, 0x8002, 0x8004, 0x8005, 0x8006, 0x8007]);
    },
};
 
module.exports = definition;`

stiglinga, 2024-11-19:
There are just a few places needed to remove cool in the code to remove this option.

> if (data.hasOwnProperty('runningMode')) { // runningMode  
>    const lookup = {0: 'off', 4: 'heat', 16: 'idle'}; -> **removed 'cool**  
>    result['running_mode'] = utils.getFromLookup(data['runningMode'], lookup);  
>    const lookup2 = {0: 'idle', 4: 'heat', 16: 'idle'}; -> **removed 'cool**  

> else if (key === 'running_mode') {  
>     const lookup = {'off': 0, 'heat': 4, 'idle': 16}; -> **removed 'cool**  
>     const payload = {'runningMode': {value: utils.getFromLookup(value, lookup), type: Zcl.DataType.enum8}};  
>     await entity.write('hvacThermostat', payload);  

> const lookup2 = {'idle': 0, 'heat': 4, 'idle': 16}; -> **removed 'cool**  
> const payload2 = {'runningState': {value: utils.getFromLookup(value, lookup2), type: Zcl.DataType.enum8}};  
> await entity.write('hvacThermostat', payload2);  

> exposes: [
>     e.climate()
>         .withSetpoint('occupied_heating_setpoint', 5, 40, 0.5)
>         .withLocalTemperature()
>         .withLocalTemperatureCalibration(-3, 3, 0.1) [...]

Here is an update of the converter with cool removed.

```javascript
const {} = require('zigbee-herdsman-converters/lib/modernExtend');
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const constants = require('zigbee-herdsman-converters/lib/constants');
const e = exposes.presets;
const ea = exposes.access;

const utils = require('zigbee-herdsman-converters/lib/utils');
const {
    Zcl
} = require('zigbee-herdsman');

const local = {
    fz: {
        namron_thermostat_new: {
            cluster: 'hvacThermostat',
            type: ['readResponse', 'attributeReport'],
            convert: (model, msg, publish, options, meta) => {
                const result = KeyValue = {};
                const data = msg.data;
                if (data.hasOwnProperty(0x8)) { // regulator_percentage
                    result['pi_heating_demand'] = data[0x8];
                }
                if (data.hasOwnProperty('runningMode')) { // runningMode
                    const lookup = {0: 'off', 4: 'heat', 16: 'idle'};
                    result['running_mode'] = utils.getFromLookup(data['runningMode'], lookup);
                    const lookup2 = {0: 'idle', 4: 'heat', 16: 'idle'};
                    result['running_state'] = utils.getFromLookup(data['runningMode'], lookup2);
                }
                if (data.hasOwnProperty(0x8000)) { // windowCheck
                    result['window_check'] = data[0x8000];
                }
                if (data.hasOwnProperty(0x8001)) { // frost
                    result['frost'] = data[0x8001];
                }
                if (data.hasOwnProperty(0x8002)) { // windowState
                    result['window_state'] = data[0x8002] ? 'opened' : 'closed';
                }
                if (data.hasOwnProperty(0x8004)) { // sensorMode
                    const lookup = { 0: 'a', 1: 'f', 2: 'af', 3: 'a2', 4: 'a2f', 5: 'fp', 6: 'p'};
                    result['sensor_mode'] = utils.getFromLookup(data[0x8004], lookup);
                }
                if (data.hasOwnProperty(0x8005)) { // backlight
[... 196 lines cut ...]
        await endpoint.configureReporting('hvacThermostat', [{ // backlight
            attribute: {ID: 0x8005, type: 32},
            minimumReportInterval: 5,
            maximumReportInterval: constants.repInterval.HOUR,
            reportableChange: null}],
        options);
        await endpoint.configureReporting('hvacThermostat', [{ // fault
            attribute: {ID: 0x8006, type: 24},
            minimumReportInterval: 0,
            maximumReportInterval: constants.repInterval.HOUR,
            reportableChange: null}],
        options);
        await endpoint.configureReporting('hvacThermostat', [{ // regulator
            attribute: {ID: 0x8007, type: 32},
            minimumReportInterval: 0,
            maximumReportInterval: constants.repInterval.HOUR,
            reportableChange: null}],
        options);

        await endpoint.read('hvacThermostat', ['systemMode', 'runningMode', 'occupiedHeatingSetpoint']);
        await endpoint.read('hvacThermostat', [0x8000, 0x8001, 0x8002, 0x8004, 0x8005, 0x8006, 0x8007]);
    },
};
 
module.exports = definition;

closed 2025-01-27 by github-actions[bot]