Koenkk/zigbee2mqtt #20217: Tuya bulbs changed in last update are not working properly

diegocjorge, 2023-12-15:
### What happened?

Hey @Koenkk, I'm having some issues with the devices _TZ3210_r5afgmkl and _TZ3210_jd3z4yig from Tuya after the latest update. 

Before the update, they were recognized as the device TS0505B_1 (https://www.zigbee2mqtt.io/devices/TS0505B_1.html) and worked just fine, including do_not_disturb and color_power_on_behavior options.
But now they are detected as TS0505B_1_1 model (https://www.zigbee2mqtt.io/devices/TS0505B_1_1.html) and this messed up their RGB color control. 

For example, when they are in a group, turned on with color temperature (they are RGBCCT bulbs) and I try to set an RGB color from HA they only change to the last color I had set from Z2M Dashboard.
So, if the last color I set for each individual bulb in the Z2M Dashboard was “blue” and I try to set the group color as “orange” from HA they will all turn blue everytime.

Also, when I try to refresh their attributes, I get these error messages:
Publish 'get' 'color' to 'bulb_name' failed: 'Error: Cluster 'lightingColorCtrl' has no attribute 'currentLevel''
Publish 'get' 'brightness' to 'bulb_name' failed: 'Error: Cluster 'lightingColorCtrl' has no attribute 'currentLevel''

Is there any way you can change the _TZ3210_r5afgmkl and _TZ3210_jd3z4yig back to use the same converter from https://www.zigbee2mqtt.io/devices/TS0505B_1.html? Thanks a lot!

### Zigbee2MQTT version

1.34.0

### Adapter firmware version

20230507

### Adapter

Sonoff Zigbee 3.0 USB Dongle Plus P

Koenkk, 2023-12-15:
Done!

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

closed 2023-12-15 by Koenkk

welderpb, 2023-12-18:
Hi @Koenkk. Same issue with _TZ3210_mja6r5ix device. it is detected as TS0505B_1_1 in 1.34.0 version. dev branch doesn't help..

Koenkk, 2023-12-19:
Probably not running the latest dev branch, make sure that frontend -> settings -> about -> zigbee-herdsman-converters version is at least 15.93.0

welderpb, 2024-01-19:
Hi @Koenkk 
issue still there.. Latest version 1.35.1
[image]

It is correctly detected in version 1.33.1 and lower

Koenkk, 2024-01-20:
@felipejfc since you moved `_TZ3210_mja6r5ix` in https://github.com/Koenkk/zigbee-herdsman-converters/pull/6266, are you sure this is correct?

mixedd69, 2024-05-13:
@Koenkk will add to others, as still have issues with those bulbs
Z2M version: 1.37.1
Bulb:
[image]

Receiving strange errors about current state on these, and it can't determine if it's mains powered device or not ("?" under Power)
`2024-05-13 09:31:57Publish 'set' 'brightness' to 'living-room/ceiling-light-a' failed: 'Error: ZCL command 0xa4c1387cbae34227/1 genLevelCtrl.moveToLevel({"level":254,"transtime":450}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 3146 - 1 - 119 - 8 - 11 after 10000ms)'`