Koenkk/zigbee2mqtt #20221: BTicino devices error on latest OTA update: Timeout: device did not request any image blocks

syntesys87, 2023-12-15:
### What happened?

When I try to update any BTicino device, i receive t he error: Timeout: device did not request any image blocks

### Zigbee2MQTT version

1.34.0

### Adapter firmware version

 7.3.1.0 build 176

### Adapter

EZSP v12 SONOFF_Zigbee_3.0_USB_Dongle_Plus_V2

### Debug log

[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/13684355/log.txt)

davidepanato, 2023-12-18:
same error on my side with these devices --> https://www.zigbee2mqtt.io/devices/412171.html#legrand-412171

some of them were updated after multiple tries, some others keep failing.

essegi65, 2023-12-21:
same problem with bticino livingnow series (n° 3) : [https://www.zigbee2mqtt.io/devices/067775_741811.html#legrand-067775%252F741811]

syntesys87, 2023-12-29:
This is another log after the fimware update released today:
[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/13793959/log.txt)

philippk75, 2024-01-07:
smae problem for me, skyconnect stick and zigbee2mqtt

closed 2024-08-06 by github-actions[bot]