Koenkk/zigbee2mqtt #20222: [New device support]: Danfoss Icon2 Room Sensor (088U2121)

JohnConnett, 2023-12-15:
### Link

https://assets.danfoss.com/documents/354092/AI380762595805en-010301.pdf

### Database entry

{"id":2,"type":"EndDevice","ieeeAddr":"0x8cf681fffe824a00","nwkAddr":26671,"manufId":4678,"manufName":"Danfoss","powerSource":"Battery","modelId":"0x8040","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":769,"inClusterList":[0,1,3,32,513,516,1026,1029,2821],"outClusterList":[3,25],"clusters":{"genBasic":{"attributes":{"modelId":"0x8040","manufacturerName":"Danfoss","powerSource":3,"zclVersion":8,"appVersion":12,"stackVersion":9,"hwVersion":1,"dateCode":"YYYYMMDD","swBuildId":"01.12\u0000"}},"genPollCtrl":{"attributes":{"checkinInterval":14400}},"hvacThermostat":{"attributes":{"localTemp":2156,"danfossRoomStatusCode":0}},"genPowerCfg":{"attributes":{"batteryPercentageRemaining":162}}},"binds":[{"cluster":32,"type":"endpoint","deviceIeeeAddress":"0xe0798dfffe0a8ce9","endpointID":1},{"cluster":1,"type":"endpoint","deviceIeeeAddress":"0xe0798dfffe0a8ce9","endpointID":1},{"cluster":513,"type":"endpoint","deviceIeeeAddress":"0xe0798dfffe0a8ce9","endpointID":1}],"configuredReportings":[{"cluster":513,"attrId":0,"minRepIntval":0,"maxRepIntval":3600,"repChange":10,"manufacturerCode":null}],"meta":{}}},"appVersion":12,"stackVersion":9,"hwVersion":1,"dateCode":"YYYYMMDD","swBuildId":"01.12\u0000","zclVersion":8,"interviewCompleted":true,"meta":{},"lastSeen":1702631450006,"defaultSendRequestWhen":"active","checkinInterval":3600}

### Comments

This is a revival of [#17958](https://github.com/Koenkk/zigbee2mqtt/issues/17958).

I have set up a standalone installation using just a [Home Assistant SkyConnect](https://www.home-assistant.io/skyconnect/) and a single Icon2. The external converter (attached) is almost identical to that provided by [Koenkk](https://github.com/Koenkk) in the Jun 17 comment of #17958.

This appears to have been partially successful. The _About_ entry is as follows:
```
Friendly name           0x8cf681fffe824a00
Description
Last seen               N/A
Availability            Disabled
Device type             EndDevice
Zigbee Model            0x8040
Zigbee Manufacturer     Danfoss
Description             Icon2™ RT Room Thermostat
Support status          Supported
IEEE Address            0x8cf681fffe824a00
Network address         0x682F
Firmware build date     YYYYMMDD
Firmware version0       1.12
Manufacturer            Danfoss
Model                   088U2121
Power                   Battery ?
Interview completed     True

```
The log contains messages like:
```
info  2023-12-15 09:06:40: MQTT publish: topic 'zigbee2mqtt/0x8cf681fffe824a00', payload '{"linkquality":255,"local_temperature":21.56}'
```
and
```
info  2023-12-15 10:15:01: MQTT publish: topic 'zigbee2mqtt/0x8cf681fffe824a00', payload '{"linkquality":255,"local_temperature":21.9,"max_heat_setpoint_limit":35,"min_heat_setpoint_limit":4,"occupied_heating_setpoint":21}'
```
Which is promising.

However, the log also contains:
```
info  2023-12-14 17:18:34: Configuring '0x8cf681fffe824a00'
error 2023-12-14 17:18:35: Failed to configure '0x8cf681fffe824a00', attempt 3 (Error: ConfigureReporting 0x8cf681fffe824a00/1 hvacThermostat([{"attribute":"runningState","minimumReportInterval":0,"maximumReportInterval":3600,"reportableChange":0}], {"sendWhen":"active","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')
    at Endpoint.checkStatus (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:421:28)
    at Endpoint.configureReporting (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:762:22)
    at Object.thermostatRunningState (/opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/src/lib/reporting.ts:176:5)
    at Object.configure (/opt/zigbee2mqtt/data/extension/externally-loaded.js:40:9)
    at Configure.configure (/opt/zigbee2mqtt/lib/extension/configure.ts:117:13))
```
How can I identify the **'UNSUPPORTED_ATTRIBUTE'** that is causing the failure?

I am keen for this device to be fully supported. Unfortunately, I am not yet fluent in [TypeScript](https://www.typescriptlang.org/) or the finer details of Zigbee devices.

Is there a way to query a Zigbee device to obtain a full set of its attributes?

Can anything useful be obtained from the [Compliance Document](https://api.knack.com/v1/applications/54e658034b4f44e42fb18201/download/asset/6374a069a6f0650021e3c379/pics.zip) in the entry for the [Danfoss Icon2 Room Thermostat](https://csa-iot.org/csa_product/danfoss-icon2-room-thermostat/) on the CSA website?

I am happy to provide further information and to test any suggested changes on the same configuration.

### External converter

```shell
jrc@skylon:~$ cat /opt/zigbee2mqtt/data/0x8040.js 
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const legacy = require('zigbee-herdsman-converters/lib/legacy');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const utils = require('zigbee-herdsman-converters/lib/utils');
const globalStore = require('zigbee-herdsman-converters/lib/store');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['0x8040'],
    model: '088U2121',
    vendor: 'Danfoss',
    description: 'Icon2™ RT Room Thermostat',
    fromZigbee: [fz.thermostat, fz.hvac_user_interface],
    toZigbee: [
        tz.thermostat_local_temperature,
        tz.thermostat_system_mode,
        tz.thermostat_occupied_heating_setpoint,
        tz.thermostat_unoccupied_heating_setpoint,
        tz.thermostat_occupied_cooling_setpoint,
        tz.thermostat_running_state,
    ],
    exposes: [
        e.climate()
            .withSetpoint('occupied_heating_setpoint', 7, 28, 0.5)
            .withSetpoint('unoccupied_heating_setpoint', 7, 28, 0.5)
            .withLocalTemperature()
            .withSystemMode(['off', 'heat', 'auto'])
            .withRunningState(['idle', 'heat']),
    ],
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(1);
        await reporting.bind(endpoint, coordinatorEndpoint, ['genPowerCfg', 'hvacThermostat']);
        await reporting.thermostatTemperature(endpoint);
        await reporting.thermostatRunningState(endpoint);
        await reporting.thermostatOccupiedHeatingSetpoint(endpoint);
        await reporting.thermostatUnoccupiedHeatingSetpoint(endpoint);
    },
};

module.exports = definition;
```

JohnConnett, 2023-12-15:
Had a closer look at the file `pics/HA_clusters.xml` in the [Compliance Document](https://api.knack.com/v1/applications/54e658034b4f44e42fb18201/download/asset/6374a069a6f0650021e3c379/pics.zip). I assume that the entries that end with `TRUE</attribute>` are the attributes supported by the Icon2.

Wrote this Python program to dig them out:
```import xmltodict
import json
file=open("pics/HA_clusters.xml","r")
xml_string=file.read()
python_dict=xmltodict.parse(xml_string)
homeAutomationPICs = python_dict['homeAutomationPICs']
cluster = homeAutomationPICs['cluster']
for item in cluster:
    print(item['name'])
    for attribute in item['attribute']:
        if attribute['#text'] == 'TRUE':
            print(f"  name=\"{attribute['@name']}\", code={attribute['@code']}, {attribute['#text']}")
    for attribute in item['attribute']:
        if attribute['#text'] == 'FALSE':
            print(f"  name=\"{attribute['@name']}\", code={attribute['@code']}, {attribute['#text']}")
file.close()
```
This was the output:
```
Relative Humidity Measurement
  name="measured value", code=0x0000, TRUE
  name="min measured value", code=0x0001, TRUE
  name="max measured value", code=0x0002, TRUE
  name="tolerance", code=0x0003, TRUE
Thermostat
  name="local temperature", code=0x0000, TRUE
  name="occupied heating setpoint", code=0x0012, TRUE
  name="min heat setpoint limit", code=0x0015, TRUE
  name="max heat setpoint limit", code=0x0016, TRUE
  name="control sequence of operation", code=0x001B, TRUE
  name="system mode", code=0x001C, TRUE
  name="outdoor temperature", code=0x0001, FALSE
  name="occupancy", code=0x0002, FALSE
  name="abs min heat setpoint limit", code=0x0003, FALSE
  name="abs max heat setpoint limit", code=0x0004, FALSE
  name="abs min cool setpoint limit", code=0x0005, FALSE
  name="abs max cool setpoint limit", code=0x0006, FALSE
  name="pi cooling demand", code=0x0007, FALSE
  name="pi heating demand", code=0x0008, FALSE
  name="HVAC system type configuration", code=0x0009, FALSE
  name="local temperature calibration", code=0x0010, FALSE
  name="occupied cooling setpoint", code=0x0011, FALSE
  name="unoccupied cooling setpoint", code=0x0013, FALSE
  name="unoccupied heating setpoint", code=0x0014, FALSE
  name="min cool setpoint limit", code=0x0017, FALSE
  name="max cool setpoint limit", code=0x0018, FALSE
  name="min setpoint dead band", code=0x0019, FALSE
  name="remote sensing", code=0x001A, FALSE
  name="alarm mask", code=0x001D, FALSE
  name="thermostat running mode", code=0x001E, FALSE
  name="start of week", code=0x0020, FALSE
  name="number of weekly transitions", code=0x0021, FALSE
  name="number of daily transitions", code=0x0022, FALSE
  name="temperature setpoint hold", code=0x0023, FALSE
  name="temperature setpoint hold duration", code=0x0024, FALSE
  name="thermostat programming operation mode", code=0x0025, FALSE
  name="hvac relay state", code=0x0029, FALSE
  name="setpoint change source", code=0x0030, FALSE
  name="setpoint change amount", code=0x0031, FALSE
  name="setpoint change source timestamp", code=0x0032, FALSE
  name="occupied setback", code=0x0034, FALSE
  name="occupied setback min", code=0x0035, FALSE
  name="occupied setback max", code=0x0036, FALSE
  name="unoccupied setback", code=0x0037, FALSE
  name="unoccupied setback min", code=0x0038, FALSE
  name="unoccupied setback max", code=0x0039, FALSE
  name="emergency heat delta", code=0x003A, FALSE
  name="ac type", code=0x0040, FALSE
  name="ac capacity", code=0x0041, FALSE
  name="ac refrigerant type", code=0x0042, FALSE
  name="ac compressor", code=0x0043, FALSE
  name="ac error code", code=0x0044, FALSE
  name="ac louver position", code=0x0045, FALSE
  name="ac coil temperature", code=0x0046, FALSE
  name="ac capacity format", code=0x0047, FALSE
```
Note that "hvac relay state", code=0x0029 is FALSE. This would seem to correspond with this line:
```
            runningState: { ID: 41, type: dataType_1.default.bitmap16 },
```
from `node_modules/zigbee-herdsman/dist/zcl/definition/cluster.js`.

This might be the **'UNSUPPORTED_ATTRIBUTE'** that is causing the failure.

JohnConnett, 2023-12-15:
Looking at the database entry `inClusterList` and `outClusterList`:
```
"inClusterList":[0,1,3,32,513,516,1026,1029,2821]
    ID: 0 genBasic
    ID: 1 genPowerCfg
    ID: 3 genIdentify
    ID: 32 genPollCtrl
    ID: 513 hvacThermostat
    ID: 516 hvacUserInterfaceCfg
    ID: 1026 msTemperatureMeasurement
    ID: 1029 msRelativeHumidity
    ID: 2821 haDiagnostic
"outClusterList":[3,25]
    ID: 3 genIdentify
    ID: 25 genOta
```
Should more of these clusters appear in the `bind`? Currently it just has `genPowerCfg` and `hvacThermostat`.

JohnConnett, 2023-12-16:
For reference, the [Home Assistant SkyConnect](https://www.home-assistant.io/skyconnect/) is running **Zigbee (EZSP) 7.3.2.0 build 212** (with **SDK 4.3.2**).

Luigi8723, 2024-01-08:
Hi @JohnConnett Thanks for reviving this earlier issue.

I also still do not have them supported in Zigbee2MQTT.
hope there is someone who can help out in this last piece?

JohnConnett, 2024-01-09:
I will take another look when time allows. My house has nine [Danfoss 014G2461](https://www.zigbee2mqtt.io/devices/014G2461.html#danfoss-014g2461) Ally radiator valves. I'm interested in using **Icon2 Room Sensors** with a few of them. Given that they share the same heritage it is possible I will learn something useful from looking at the radiator valve code.

Luigi8723, 2024-01-09:
> I will take another look when time allows. My house has nine [Danfoss 014G2461](https://www.zigbee2mqtt.io/devices/014G2461.html#danfoss-014g2461) Ally radiator valves. I'm interested in using [...]

I hope so, my knowledge about this backend things is zero. So i am sorry i cannot help with this.

About the reports of the icon2 thermostat. It is creating an error in the logs. But is it reporting the right temp? And setpoint to HA?
And can you change the setpoint or are these things not possible due to errors?

Luigi8723, 2024-01-12:
Also the Humidy is not reporting in my HAos system.

All the other things seems to work, execept the configuration errors

JohnConnett, 2024-01-14:
Decided to try a different approach. Working from [Support new devices](https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html). Here is my current external converter:
```
jrc@topaz:~$ cat /opt/zigbee2mqtt/data/088U2121.js
const {batteryPercentage, temperature, humidity, identify} = require('zigbee-herdsman-converters/lib/modernExtend');
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const utils = require('zigbee-herdsman-converters/lib/utils');
const globalStore = require('zigbee-herdsman-converters/lib/store');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['0x8040'],
    model: '0x8040',
    vendor: 'Danfoss',
    description: 'Icon2™ RT Room Thermostat',
    extend: [batteryPercentage(), temperature(), humidity(), identify()],
    fromZigbee: [fz.linkquality_from_basic, fz.power_source, fz.battery, ifz.identify, fz.checkin_presence, fz.danfoss_thermostat, fz.hvac_user_interface, fz.temperature, fz.humidity, fz.danfoss_icon_regulator],
    toZigbee: [],
    exposes: [e.battery(), e.temperature(), e.humidity()],
};

module.exports = definition;
jrc@topaz:~$
```
For the `fromZigbee: ` I took the `"inClusterList": ` from the database entry and looked in `zigbee-herdsman-converters/src/converters/fromZigbee.ts` for items that appeared to match. If there was a Danfoss specific item I chose that in preference to the standard one as follows:
```
       0 0x0000 genBasic                 fz.linkquality_from_basic
                                         fz.power_source
       1 0x0001 genPowerCfg              fz.battery
       3 0x0003 genIdentify              fz.identify
      32 0x0020 genPollCtrl              fz.checkin_presence
     513 0x0201 hvacThermostat           fz.danfoss_thermostat
     516 0x0204 hvacUserInterfaceCfg     fz.hvac_user_interface
    1026 0x0402 msTemperatureMeasurement fz.temperature
    1029 0x0405 msRelativeHumidity       fz.humidity
    2821 0x0B05 haDiagnostic             fz.danfoss_icon_regulator
```
No errors in the log. Haven't tried using MQTT to get anything other than `battery`, `temperature` and `humidity` which can be requested through the exposes page of the web UI.

The regular published messages are of the form:
```
info  2024-01-14 12:44:47: MQTT publish: topic 'zigbee2mqtt/0x8cf681fffe824a00', payload '{"battery":75,"humidity":30,"linkquality":255,"temperature":-327.68}'
```
`battery`, `humidity` and `linkquality` look reasonable (`humidity` matches the value on the display). However, `temperature` is wildly wrong (below absolute zero!).

Would appreciate any comments on this approach!

Next steps are to understand why the `temperature` is wrong and to add the `toZigbee: ` entries.

Not clear what should happen for the `exposes:` and `configure: ` entries.

JohnConnett, 2024-01-14:
Was looking at this [Danfoss Ally eTRV0100 - other features than just temperature and setpoint](https://github.com/zigpy/zha-device-handlers/issues/1726) under **zigpy / zha-device-handlers** where a comment from **stachjankowski** has a link to [Feature Catalogue Ally™ electronic Radiator Thermostat (eTRV)](https://assets.danfoss.com/documents/202524/AU417130778872en-000101.pdf).

A quick glance at the document suggests that it provides a theory-of-operation for the eTRV0100. If the equivalent was available for the **Icon2 Room Sensor** it would be extremely useful! I will try asking Danfoss if one exists.

JohnConnett, 2024-01-15:
For comparison, here is what **Zigbee Home Automation: SkyConnect v1.0** reports from **Home Assistant**:

**Device info** 0x8040 by Danfoss Firmware: 0x0000010c
    IEEE: 8c:f6:81:ff:fe:82:4a:00
    Nwk: 0xc938
    Device Type: EndDevice
    LQI: 255
    RSSI: -33
    Last seen: 2024-01-15T16:33:34
    Power source: Battery or Unknown
**Controls**
    Thermostat Heat 21 °C Currently: 24.2 °C
**Sensors**
    Humidity 26.0%
    HVAC action Unknown
    Temperature -327.7 °C
**Diagnostic**
    Battery 100%
    Identity PRESS
    LQI
    RSSI
**Logbook**
    January 15, 2024
    _Danfoss 0x8040 Identify_ changed to January 15, 2024 at 4:20 PM triggered by service Button: Press
    4:20:48 PM - 25 minutes ago - John Connett

Note that the **Thermostat** temperature is correct (24.2 °C) but the **Sensors** temperature is wrong (-327.7 °C) much as with zigbee2mqtt.

Luigi8723, 2024-01-16:
Is this -327,7 not a value which is calculated to Currently: 24.2 °C ??

I will send this report to a co-worker ( i am working at Danfoss Netherlands, for another productline ).

JohnConnett, 2024-01-16:
> Is this -327,7 not a value which is calculated to Currently: 24.2 °C ??

My best guess is that there are two temperatures reported, one from the **hvacThermostat** cluster and another from the **msTemperatureMeasurement** cluster. Seems likely that they both result from the same physical sensor within the device.

I note that -327.68 is -(2^15 / 100). I have not seen any other values. Might be significant?

> I will send this report to a co-worker ( i am working at Danfoss Netherlands, for another productline ).

I have asked Danfoss about more documentation (case number 14816049). Will see what results.

Longer term, I'm not sure if I will be using zigbee2mqtt or ZHA with Home Assistant. I will also look into getting this device fully supported under ZHA.

Luigi8723, 2024-01-16:
Just had a small dicussion with my co-worker about this.

this is reporting out of the Icon2 sensor
```
 "battery": 81,
    "humidity": 54,
    "linkquality": 87,
    "local_temperature": 20.3,
    "max_heat_setpoint_limit": 35,
    "min_heat_setpoint_limit": 4,
    "occupied_heating_setpoint": 16.5,
    "temperature": -327.68,
    "running_state": null,
    "system_mode": "heat"
```

and this the ally
```
adaptation_run_control": "none",
    "adaptation_run_settings": true,
    "adaptation_run_status": "none",
    "algorithm_scale_factor": 10,
    "battery": 52,
    "day_of_week": "tuesday",
    "external_measured_room_sensor": -8000,
    "heat_available": true,
    "heat_required": true,
    "keypad_lockout": "lock1",
    "linkquality": 87,
    "load_balancing_enable": true,
    "load_estimate": -1,
    "load_room_mean": -8000,
    "local_temperature": 16.96,
    "mounted_mode_active": false,
    "mounted_mode_control": false,
    "occupied_heating_setpoint": 19.5,
    "occupied_heating_setpoint_scheduled": 19.5,
    "pi_heating_demand": 100,
    "preheat_status": false,
    "radiator_covered": false,
    "running_state": "heat",
    "setpoint_change_source": "externally",
    "system_mode": "heat",
    "trigger_time": 660,
    "update": {
        "installed_version": 276,
        "latest_version": 284,
        "state": "available"
    },
    "window_open_external": false,
    "window_open_feature": true,
    "window_open_internal": "closed"
```

our guess is at the moment the  "temperature": -327.68," is a fault error due to the missing link to external room sensor.

this is also with the ally     "load_room_mean": -8000," 

Danfoss has also a roomsensor which can be coupled to an ally, but not jet to a Icon2 sensor. Maby the software is prepared for this.

The Danfoss gateway is not able to see the -327 temp also.

So it seems like an future function

JohnConnett, 2024-01-18:
> Danfoss has also a roomsensor which can be coupled to an ally, but not jet to a Icon2 sensor. Maby the software is prepared for this.

The [Danfoss Ally™ Room Sensor (014G2480)](https://store.danfoss.com/en/Climate-Solutions-for-heating/Smart-heating/Danfoss-Ally-%E2%80%93-smart-heating/Thermostats-and-Sensors/Danfoss-Ally%E2%84%A2-Room-Sensor%2C-Zigbee/p/014G2480)? I suspect it is a badge-engineered version of the [TuYa TS0201](https://www.zigbee2mqtt.io/devices/TS0201.html).

See [Danfoss Ally Room Thermostat has wrong picture, vendor, model and description
](https://github.com/Koenkk/zigbee2mqtt/issues/20243).

Luigi8723, 2024-01-18:
> The [Danfoss Ally™ Room Sensor [...]

Yes thats the one.
Seems to be a rebranded product indeed.

Danfoss had this product to have a good temp of the room when the radiator is covered. And this was before the icon 2 was there.

JohnConnett, 2024-01-25:
I was wondering if the [Danfoss Icon2™ Featured Room Thermostat (088U2122)](https://store.danfoss.com/en/Climate-Solutions-for-heating/Hydronic-floor-heating/Room-controls/Floor-Heating-Controls%2C-Danfoss-Icon2%E2%84%A2%2C-Attachable/p/088U2122) which has an infrared floor sensor reports that temperature using the **msTemperatureMeasurement** cluster?

It would make sense that the non-Featured variant would report the temperature as unknown.

JohnConnett, 2024-02-02:
> I have asked Danfoss about more documentation (case number 14816049). Will see what results.

Received a reply: _Sorry no. We cannot answer questions regarding API’s, developing or 3. Party apps/systems. Customers has to go to [developer.danfoss.com](https://developer.danfoss.com/), create an account, and place his question there._

Followed this advice and under the heading **Support** found: _Questions or problems? Please check out our FAQ page for the answers to frequently asked questions about the Danfoss developer portal. For questions regarding specific APIs, please refer to the contacts listed on the API page._ 

As the answer isn't on the FAQ page and the question isn't regarding a specific API I seem to have arrived at a dead end.

Maxxis20, 2024-02-22:
> I was wondering if the [Danfoss Icon2™ Featured Room Thermostat [...]

Hi,

I have also been doing some development on this same topic.

And i can confirm that:
 "local_temperature": 20.3 = The temperature of the sensor
  "temperature": -327.68 = The external temperature, ie. the infrared for the featured or then the external sensor for the 24V thermostat.

mmirannalti, 2024-03-11:
Hi everyone. 
I just bought the icon2 controller and thermostat.
Is it possible to pair them with z2m? Or any other way to make them visible to homeassistant? 
I have seen a long trail of issues but I could not understand the status.

I am happy to help and contribute.

JohnConnett, 2024-03-11:
I only have the **Icon2™ RT 088U2121** wireless thermostat. Basic support for this (and probably its close wireless relatives) is possible. Only features not yet available are the Danfoss manufacturer specific options which will require information from Danfoss. They have provided similar detailed information for their **Ally™ Radiator Thermostats** so it's probably a case of finding the right person to ask. These are **Zigbee** devices and should work with **Zigbee2MQTT** or **ZHA** using any **Home Assistant** supported coordinator.

I don't have an **Icon2™ Controller**. However, I think it operates as the Zigbee coordinator for the wireless thermostats. For a LAN connection I think an **Ally™ Gateway** is required so a [Danfoss API](https://developer.danfoss.com/get-started) would be the route to a **Home Assistant** Integration. Not among the community integrations but there is [Danfoss Ally](https://github.com/MTrab/danfoss_ally) in [HACS](https://hacs.xyz/). Hope that helps.

Maxxis20, 2024-03-11:
Dont bother with the Ally Gateway. 
The Icon2 can join Z2M directly. 
there is another thread for it 🙃

JohnConnett, 2024-03-11:
> Dont bother with the Ally Gateway.
> The Icon2 can join Z2M directly.
> there is another thread for it 🙃

Sorry for the confusion. The **Ally™ Gateway** is mentioned after 3:38 in [Installation guide - Icon2™ floor heating system](https://www.youtube.com/watch?v=Ma4EB1T4mSU) and after 12:24 in [Danfoss Icon2 Review underfloor heating cooling control system](https://www.youtube.com/watch?v=52wKnb-YC9A).

The second video mentions that the **Icon2™ Main Controller** has two connectors for optional external antennas, one for **Icon2™ Wireless Room Thermostats** and the other for an **Ally™ Gateway**. My guess is that it can support two Zigbee networks: one as coordinator for the room thermostats; and the other as an end device to an **Ally™ Gateway**, **Zigbee2MQTT** or **ZHA**. The second video also mentions that an **Ally™ Gateway** can control multiple **Icon2™ Main Controllers** which implies to me that they can act as end devices.

Couldn't find any mention of **Icon2™ Main Controller** being supported in [Zigbee2MQTT Supported Devices](https://www.zigbee2mqtt.io/supported-devices/); [ZHA Device Handlers](https://github.com/zigpy/zha-device-handlers/issues?q=Icon2) or [Zigbee Device Compatibility Repository](https://zigbee.blakadder.com/search.html).

Again, apologies for the confusion. It would be interesting to know what information the **Icon2™ Main Controller** makes available as a Zigbee end device. However, that is drifting off topic for this thread.

JohnConnett, 2024-03-12:
> So there is another thread, or a couple actually regarding support for the Icon2 system. Now at the computer i got the link: [#19982 (reply in thread)](https://github.com/Koenkk/zigbee2mqtt/discussions/19982#discussioncomment-8178764)

Thanks for the link. It would be useful to raise a **New device support** issue for the **Icon2™ Main Controller** so that it is easier for others to find.

The information on install codes may be useful for another of my projects: [Philips Hue Secure Floodlight Camera EU](https://github.com/Koenkk/zigbee2mqtt/issues/21650).

> So you would need a Z2M coordinator, i suppose you have one?

I have three: [Electrolama zzh Multiprotocol RF Stick](https://electrolama.com/products/zzh-multiprotocol-rf-stick); [Home Assistant SkyConnect](https://www.home-assistant.io/skyconnect/); [nRF52840 Dongle](https://www.nordicsemi.com/Products/Development-hardware/nrf52840-dongle). The **zzh** is live, the **SkyConnect** is for testing and the **nRF52840 Dongle** is currently configured as a Zigbee sniffer. Four if you count my **Hue Bridge**, v2.1 (I haven't tried [rooting](https://blog.andreibanaru.ro/2018/03/27/philips-hue-2-1-enabling-wifi/) it).

> The Ally Gateway is used for the App and Danfoss AllyPro Web UI, but even it doesnt support everything. For example Humidity data is now shown on the Web.

I suppose it might be possible to learn something about the Danfoss manufacturer specific options by observing what is available via the **Ally™ Gateway**?

JohnConnett, 2024-03-25:
I have recommissioned my **Icon2™ RT 088U2121** to **Home Assistant** / **Zigbee2MQTT**. No external converter. It is now recognised as supported. However, the Model is **Icon** and the picture is of a framed Icon thermostat.

However, there now appear to be 15 of them, labelled **L1** through **L15**, as if it is an **Icon2™ Main Controller**!

Maxxis20, 2024-03-25:
> I have recommissioned my **Icon2™ RT 088U2121** to **Home Assistant** / **Zigbee2MQTT**. No external converter. It is now recognised as supported. However, the Model is **Icon** and the picture is [...]

So it is not recognized as: Zigbee Model 0x0210?

JohnConnett, 2024-03-25:
> So it is not recognized as: Zigbee Model 0x0210?

It is Zigbee Model **0x8040**. This is a line from the log:
```
info  2024-03-25 14:54:34: Icon2_Room_Sensor (0x8cf681fffe824a00): Icon - Danfoss Icon floor heating (regulator, Zigbee module & thermostats) (EndDevice)
```

JohnConnett, 2024-03-26:
I have generated a picture for the **Icon2™ Room Sensor** using the following:
```
$ # Grab the picture
$ curl https://www.danfoss.com/media/hy5fgdkz/danfoss-icon2-rt-fo-21-degrees-1120x747.jpg --output temp.jpg
$ # Use https://new.express.adobe.com/tools/remove-background
$ # Crop to square
$ convert temp.png -crop 2668x2668+666+0 +repage temp2.png
$ # Resize
$ convert temp2.png -resize 512x512 088U2121.png
$
```
It is rather flat! I was unable to find a high quality _slightly turned clockwise_ image similar to `public/images/devices/Icon.png`. Haven't submitted it as it is uncertain what filename would be used. Please feel free to use it or replace it with something better.
[image]

JohnConnett, 2024-09-24:
This is still an issue. Particularly, differenciating between a single **Icon2 Room Sensor** and several of them behind an **Icon2 Controller**. My single **Icon2 Room Sensor** is detected as 16(?) of them which rather clutters my Home Assistant overview.

JohnConnett, 2025-01-03:
Happy New Year 2025!

This is still an issue. Particularly, differentiating between a single Icon2 Room Sensor and several of them behind an Icon2 Controller. My single Icon2 Room Sensor is detected as 16(?) of them which rather clutters my Home Assistant overview.

JohnConnett, 2025-03-06:
This is still an issue.Particularly, differentiating between single **Icon2™ Room Sensors** and several of them behind an **Icon2™ Main Controller**.

My single **Icon2™ Room Sensor** is detected as 16(?) of them which rather clutters my Home Assistant overview. It might also be annoying for those with less than the full complement of **Icon2™ Room Sensors** behind their **Icon2™ Main Controller**.

Can anyone who has an **Icon2™ Main Controller** with attached **Icon2™ Room Sensors** provide details of how they appear on the Zigbee network? Perhaps an image of the part of the Zigbee2MQTT map showing the devices?

closed 2025-05-15 by github-actions[bot]