Koenkk/zigbee2mqtt #20223: Moes BHT-002 sets the Local temperature calibration to really high values randomly

KrOnAsK, 2023-12-15:
### What happened?

Hi, I just installed this thermostat and its not really working as expected because when I set the Local temperature calibration to -1 after a random amoun of time it sets its Value to 1043200 for no apparent reason. I have attached a logfile and the device in Question is ThermostatWoZi

### What did you expect to happen?

That the local temp calibration stay the same as I set it 

### Zigbee2MQTT version

1.34.0

### Adapter firmware version

 6.10.3.0 build 297

### Adapter

SONOFF Zigbee Gateway, ZBDongle-E

### Debug log

[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/13685810/log.txt)

lukeskymuh, 2024-01-13:
I have the same efect for the Moes BHT-006. I suspected that this was caused by "better thermostat". Do you use "better thermostat "?

KrOnAsK, 2024-01-13:
Yes I do

arnassavickas, 2024-03-07:
Have you found a solution? I've removed Better Thermostat but the value still gets increased randomly to thousands.

closed 2024-10-04 by github-actions[bot]