Koenkk/zigbee2mqtt #20230: [Wrong device]: MHCOZY TYWB 4ch-RF wrongly recognized as TuYa TS0004

tlc76, 2023-12-15:
### Link

https://www.aliexpress.com/item/1005005457296054.html

### Model

zg-005-rf

### Description

Relay 4CH Switch Module

### Vendor

Chinese no-name

### Picture (link)

https://avatars.dzeninfra.ru/get-zen_doc/3828869/pub_63ad2443b13e7b3bf9cf4327_63af10533869ed199ef9864b/scale_1200

### Database entry

{"id":19,"type":"Router","ieeeAddr":"0xa4c138811fc6804d","nwkAddr":30500,"manufId":4417,"manufName":"_TZ3000_imaccztn","powerSource":"Mains (single phase)","modelId":"TS0004","epList":[1,2,3,4,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":256,"inClusterList":[3,4,5,6,57344,57345,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65506":54,"65508":0,"65534":0,"modelId":"TS0004","manufacturerName":"_TZ3000_imaccztn","powerSource":1,"zclVersion":3,"appVersion":82,"stackVersion":0,"hwVersion":1,"dateCode":""}},"manuSpecificBosch":{"attributes":{"53251":"AAAAAgAABAAABgAA"}},"genOnOff":{"attributes":{"32768":0,"onOff":0,"onTime":0,"offWaitTime":0,"tuyaBacklightMode":1,"moesStartUpOnOff":2,"tuyaBacklightSwitch":1}},"manuSpecificTuya_3":{"attributes":{"powerOnBehavior":2,"switchType":0}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x804b50fffe9cd9d4","endpointID":1}],"configuredReportings":[],"meta":{}},"2":{"profId":260,"epId":2,"devId":256,"inClusterList":[4,5,6,57345],"outClusterList":[],"clusters":{"genOnOff":{"attributes":{"onOff":0,"onTime":0,"offWaitTime":0}},"manuSpecificTuya_3":{"attributes":{"powerOnBehavior":2}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x804b50fffe9cd9d4","endpointID":1}],"configuredReportings":[],"meta":{}},"3":{"profId":260,"epId":3,"devId":256,"inClusterList":[4,5,6,57345],"outClusterList":[],"clusters":{"genOnOff":{"attributes":{"onOff":0,"onTime":0,"offWaitTime":0}},"manuSpecificTuya_3":{"attributes":{"powerOnBehavior":2}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x804b50fffe9cd9d4","endpointID":1}],"configuredReportings":[],"meta":{}},"4":{"profId":260,"epId":4,"devId":256,"inClusterList":[4,5,6,57345],"outClusterList":[],"clusters":{"genOnOff":{"attributes":{"onOff":1,"onTime":0,"offWaitTime":0}},"manuSpecificTuya_3":{"attributes":{"powerOnBehavior":2}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x804b50fffe9cd9d4","endpointID":1}],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":82,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":889996980},"lastSeen":1702678421614,"defaultSendRequestWhen":"immediate"}

### Notes

Besides the "wrong device" issue, there's an additional one: the 4 relays cannot be triggered independently. There can only be 1 switch on and 3 off, at any time. When you try to swith on the 2nd switch, the one already on is turning off.
Thank you.

uli-rpi, 2024-01-21:
> Besides the "wrong device" issue, there's an additional one: the 4 relays cannot be triggered independently. There can only be 1 switch on and 3 off, at any time. When you try to swith on the 2nd switch, the one already on is turning off.
Thank you.

Use the onboard mode switch to change interlock/self-locking/inching modes. What you are looking for is the self-locking mode.

tlc76, 2024-01-24:
Thanks for the tip, I managed to change the mode by pressing the "switch mode" button.

While waiting for koenkk to add this device in the repository, I tested it myself by adding its signature in the tuya.js directly in the directory structure containing the z2m docker image (I actually found 3 instances of tuya.js in 2 different containers):

```
@@ -4218,6 +4218,7 @@
         whiteLabel: [
             tuya.whitelabel('TuYa', 'DS-111', 'Smart light switch - 4 gang with neutral wire', ['_TZ3000_mdj7kra9']),
             tuya.whitelabel('MHCOZY', 'TYWB 4ch-RF', '4 channel relay', ['_TZ3000_u3oupgdy']),
+            tuya.whitelabel('MHCOZY', 'TYWB 4ch-RF', '4 channel relay', ['_TZ3000_imaccztn']),
             tuya.whitelabel('Avatto', 'TS0004_1', 'Smart light switch - 4 gang with neutral wire', ['_TZ3000_nivavasg', '_TZ3000_gexniqbq']),
         ],
         meta: { multiEndpoint: true },
```

Some notes:
- backlight function doesn't seem to affect the device in any way
- "mode switch" is not supported in z2m, it can only be manually operated by pressing the 5th button (entitled "mode switch") on the device.

uli-rpi, 2024-01-24:
As I understand, the mode switch can only be used manually directly on the board, this function is basically not available via Zigbee. 

and, 

I haven't understood the backlight function until now what that is supposed to be

tlc76, 2024-01-24:
Thanks a lot. This means the device I've reported here works as expected and its signature can be safely added in z2m.

uli-rpi, 2024-01-24:
I guess so. I have been using the device in the two-channel and four-channel versions for several weeks without any problems.

The four channel version is:

Zigbee-Modell
TS0004
Zigbee-Hersteller
_TZ3000_u3oupgdy

Koenkk, 2024-01-25:
If it works, can you create a pull request for this? Click [here](https://github.com/Koenkk/zigbee-herdsman-converters/edit/master/src/devices/tuya.ts) to do so

closed 2024-01-31 by Koenkk

heikkis, 2024-07-29:
Any idea why Z2M opens or closes allways all together with four messages? @uli-rpi @tlc76 

Directly from the device on/off works one by one (mode is correct) but from Z2M I can see that 4 msgs is published:
1: on 2-4: off
1-2: on 3-4: off
1-3: on 4: off
1-4: on

Same for off. From Z2M UI and HA UI produce same result.

ZGM is latest 1.39.0.

```
Info 2024-07-29 22:33:47z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/reletesti', payload '{"backlight_mode":"ON","linkquality":92,"power_on_behavior_l1":"off","power_on_behavior_l2":"off","power_on_behavior_l3":"off","power_on_behavior_l4":"off","state_l1":"ON","state_l2":"OFF","state_l3":"OFF","state_l4":"OFF"}'
Info 2024-07-29 22:33:47z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/reletesti', payload '{"backlight_mode":"ON","linkquality":88,"power_on_behavior_l1":"off","power_on_behavior_l2":"off","power_on_behavior_l3":"off","power_on_behavior_l4":"off","state_l1":"ON","state_l2":"ON","state_l3":"OFF","state_l4":"OFF"}'
Info 2024-07-29 22:33:48z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/reletesti', payload '{"backlight_mode":"ON","linkquality":92,"power_on_behavior_l1":"off","power_on_behavior_l2":"off","power_on_behavior_l3":"off","power_on_behavior_l4":"off","state_l1":"ON","state_l2":"ON","state_l3":"ON","state_l4":"OFF"}'
Info 2024-07-29 22:33:48z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/reletesti', payload '{"backlight_mode":"ON","linkquality":92,"power_on_behavior_l1":"off","power_on_behavior_l2":"off","power_on_behavior_l3":"off","power_on_behavior_l4":"off","state_l1":"ON","state_l2":"ON","state_l3":"ON","state_l4":"ON"}'
```

uli-rpi, 2024-07-29:
@heikkis 
i've checked my device with 1.39.0
I can easily switch each individual channel on and off separately. I see 2 entries for each switching operation in the log.

Eg: switching channel 1 from off to on

`info 2024-07-29 22:30:15z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/test/plug/4ch', payload '{"backlight_mode":"OFF","last_seen":"2024-07-29T22:30:16+02:00","linkquality":193,"power_on_behavior_l1":"off","power_on_behavior_l2":"off","power_on_behavior_l3":"off","power_on_behavior_l4":"off","state_l1":"OFF","state_l2":"OFF","state_l3":"OFF","state_l4":"OFF"}'`

`info 2024-07-29 22:35:59z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/test/plug/4ch', payload '{"backlight_mode":"OFF","last_seen":"2024-07-29T22:36:00+02:00","linkquality":171,"power_on_behavior_l1":"off","power_on_behavior_l2":"off","power_on_behavior_l3":"off","power_on_behavior_l4":"off","state_l1":"ON","state_l2":"OFF","state_l3":"OFF","state_l4":"OFF"}'`

From time to time I get 3 entries, the switch-on command is then sent twice. I do not know the reason for this

heikkis, 2024-07-30:
Finally probably the restart of the HA Z2M addon helped. I already removed the device from Z2M, booted the relay device etc. but nothing help. I was adding debug log level to get more info and suddenly the restart helped also for this matter. Thanks @uli-rpi for your reply.