Koenkk/zigbee2mqtt #20234: [New device support]: Innr SP 244

brad-newman, 2023-12-16:
### Link

https://www.innr.com/en/product/innr-smart-plug-15a-us/

### Database entry

{"id":4,"type":"Router","ieeeAddr":"0xa4c138423cacae64","nwkAddr":47061,"manufId":4454,"manufName":"innr","powerSource":"Mains (single phase)","modelId":"SP 244","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":266,"inClusterList":[0,3,4,5,6,8,4096,1794,2820,57345],"outClusterList":[10,25],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":20,"stackVersion":54,"hwVersion":1,"dateCode":"20231019","swBuildId":"1.4.3","zclVersion":8,"interviewCompleted":false,"meta":{},"lastSeen":1702714807922,"defaultSendRequestWhen":"immediate"}

### Comments

zigbee2mqtt running via HA Add-on. 

Followed support new devices instructions but couldn't get past "trigger some actions on the device". Here is the debug log:
```
Debug 2023-12-16 00:33:45Device '0xa4c138423cacae64' announced itself
Info 2023-12-16 00:33:45MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xa4c138423cacae64","ieee_address":"0xa4c138423cacae64"},"type":"device_announce"}'
Info 2023-12-16 00:33:46Successfully interviewed '0xa4c138423cacae64', device has successfully been paired
Warning 2023-12-16 00:33:46Device '0xa4c138423cacae64' with Zigbee model 'SP 244' and manufacturer name 'innr' is NOT supported, please follow https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html
Info 2023-12-16 00:33:46MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"definition":null,"friendly_name":"0xa4c138423cacae64","ieee_address":"0xa4c138423cacae64","status":"successful","supported":false},"type":"device_interview"}'
```

### External converter

```shell
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const {onOff, electricityMeter} = require('zigbee-herdsman-converters/lib/modernExtend');
const ota = require ('zigbee-herdsman-converters/lib/ota');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
zigbeeModel: ['SP 244'], // The model ID from: Device with modelID 'lumi.sens' is not supported.
model: 'SP 244', // Vendor model number, look on the device for a model number
vendor: 'inrr', // Vendor of the device (only used for documentation and startup logging)
description: 'Smart plug', // Description of the device, copy from vendor site. (only used for documentation and startup
fromZigbee: [fz.on_off], // We will add this later
toZigbee: [tz.on_off], // Should be empty, unless device can be controlled (e.g. lights, switches).
exposes: [e.switch()], // Defines what this device exposes, used for e.g. Home Assistant discovery and in the frontend
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(1);
        await reporting.bind(endpoint, coordinatorEndpoint, ['genOnOff']);
        await reporting.onOff(endpoint);
    },
ota: ota.zigbeeOTA,
};

module.exports = definition;
```

Koenkk, 2023-12-17:
Added!

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

closed 2023-12-17 by Koenkk

bengeach, 2023-12-20:
Innr also have a new UK plug called Innr SP 242 which is not currently supported.
https://www.innr.com/en/product/innr-smart-plug-uk/

Is it possible to add this too? (sorry if this is a silly question, I'm relatively new using HomeAssistant)