Koenkk/zigbee2mqtt #20237: Xiaomi Aqara Double Switch change mode failure

vaderag, 2023-12-16:
### What happened?

Xiaomi Switch
Model
QBKG03LM
Trying to change mode to decoupled and I get the error:
Publish 'set' 'operation_mode' to 'Playroom Double Fixed Switch ' failed: 'Error: undefined is not a object, got string ("decoupled")'

### What did you expect to happen?

Expected to successfully change to Decoupled (or alternative relay)

### How to reproduce it (minimal and precise)

Select the option in Zigbee2MQTT to change mode - fails

### Zigbee2MQTT version

1.34

### Adapter firmware version

20220219

### Adapter

Sonoff 

### Debug log

[.168.1.125](http://192.168.1.125:8123/api/hassio_ingress/1ttI3c_nDW9v0dZaribVSkiFH3B6mCn52jtZUYylFsc/#/)

Show only
all
Filter by text
Enter search criteria
Log level
debug
Logging level
 
Debug 2023-12-16 13:42:00Received Zigbee message from 'Playroom Double Fixed Switch ', type 'read', cluster 'genTime', data '["time"]' from endpoint 1 with groupID 0
Debug 2023-12-16 13:42:19Received MQTT message on 'zigbee2mqtt/Playroom Double Fixed Switch /set' with data '{"operation_mode_left":"decoupled"}'
Debug 2023-12-16 13:42:19Publishing 'set' 'operation_mode' to 'Playroom Double Fixed Switch '
Error 2023-12-16 13:42:19Publish 'set' 'operation_mode' to 'Playroom Double Fixed Switch ' failed: 'Error: undefined is not a object, got string ("decoupled")'
Debug 2023-12-16 13:42:19Error: undefined is not a object, got string ("decoupled") at Object.assertObject (/app/node_modules/zigbee-herdsman-converters/src/lib/utils.ts:511:15) at Object.convertSet (/app/node_modules/zigbee-herdsman-converters/src/converters/toZigbee.ts:2439:19) at Publish.onMQTTMessage (/app/lib/extension/publish.ts:248:52) at EventEmitter.emit (node:events:525:35) at EventBus.emitMQTTMessage (/app/lib/eventBus.ts:109:22) at MQTT.onMessage (/app/lib/mqtt.ts:141:27) at WebSocket.<anonymous> (/app/lib/extension/frontend.ts:130:27) at WebSocket.emit (node:events:513:28) at Receiver.receiverOnMessage (/app/node_modules/ws/lib/websocket.js:1192:20) at Receiver.emit (node:events:513:28)
Debug 2023-12-16 13:42:52Received Zigbee message from 'Tuya ZigBee with Power Monitor X', type 'readResponse', cluster 'haElectricalMeasurement', data '{"activePower":41,"rmsCurrent":252,"rmsVoltage":247}' from endpoint 1 with groupID 0
Info 2023-12-16 13:42:52MQTT publish: topic 'zigbee2mqtt/Tuya ZigBee with Power Monitor X', payload '{"child_lock":"UNLOCK","current":0.25,"energy":485.08,"indicator_mode":null,"linkquality":120,"power":41,"power_outage_memory":null,"state":"ON","update":{"installed_version":-1,"latest_version":-1,"state":null},"update_available":null,"voltage":247}'
Debug 2023-12-16 13:42:52Received Zigbee message from 'Tuya ZigBee with Power Monitor X', type 'readResponse', cluster 'seMetering', data '{"currentSummDelivered":[0,48508]}' from endpoint 1 with groupID 0
Info 2023-12-16 13:42:52MQTT publish: topic 'zigbee2mqtt/Tuya ZigBee with Power Monitor X', payload '{"child_lock":"UNLOCK","current":0.25,"energy":485.08,"indicator_mode":null,"linkquality":112,"power":41,"power_outage_memory":null,"state":"ON","update":{"installed_version":-1,"latest_version":-1,"state":null},"update_available":null,"voltage":247}'
Debug 2023-12-16 13:42:52Received Zigbee message from 'Playroom Double Fixed Switch ', type 'attributeReport', cluster 'genBasic', data '{"65281":{"1":3300,"10":0,"100":1,"101":1,"110":1,"111":1,"153":0,"155":0,"3":23,"5":568,"8":9746}}' from endpoint 1 with groupID 0
Debug 2023-12-16 13:42:52QBKG03LM: unknown key 110 with value 1
Debug 2023-12-16 13:42:52QBKG03LM: unknown key 111 with value 1
Debug 2023-12-16 13:42:52QBKG03LM: unknown key 153 with value 0
Debug 2023-12-16 13:42:52QBKG03LM: unknown key 155 with value 0
Debug 2023-12-16 13:42:52QBKG03LM: Processed data into payload {"voltage":3300,"device_temperature":23,"power_outage_count":567,"state_left":"ON","state_right":"ON"}
Debug 2023-12-16 13:42:52QBKG03LM: Processed data into payload {"voltage":3300,"device_temperature":23,"power_outage_count":567,"state_left":"ON","state_right":"ON"}

pusheax0, 2023-12-16:
This has been fixed in dev branch. https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html

vaderag, 2023-12-16:
> This has been fixed in dev branch. https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html

Dumb question - if I install the edge addon do I have to re-set up everything or will it pull the config from the standard?
And can I change back without issue?

I may just wait for this to release with the full version... But interested to know?

pusheax0, 2023-12-16:
This issue should answer your questions about Edge https://github.com/Koenkk/zigbee2mqtt/issues/20036#issuecomment-1843700672

closed 2023-12-17 by Koenkk