Koenkk/zigbee2mqtt #20239: Unable to change Xiaomi Vibration sensor sensitivity

Rainz3, 2023-12-16:
### What happened?

I am trying to set the sensitivity on two vibration sensors. I've tried directly through the device page in zigbee2mqtt and get the error
Debug 2023-12-16 10:46:22Publishing 'set' 'sensitivity' to 'Vibration_sensor_1'
Error 2023-12-16 10:46:58Publish 'set' 'sensitivity' to 'Vibration_sensor_1' failed: 'Error: Write 0x00158d0009ee3e2b/1 genBasic({"65293":{"value":1,"type":32}},
Info 2023-12-16 10:46:58MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'sensitivity' to 'Vibration_sensor_1' failed: 'Error: Write 0x00158d0009ee3e2b/1 genBasic({\"65293\":{\"value\":1,\"type\":32}}, {\"sendWhen\":\"immediate\",\"timeout\":35000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":true,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":4447,\"transactionSequenceNumber\":null,\"writeUndiv\":false}) failed (no response received (10))'","meta":{"friendly_name":"Vibration_sensor_1"},"type":"zigbee_publish_error"}'

I've tried using developer tools in home assistant by publishing to the topic zigbee2mqtt/[Vibration_sensor_1]/set with the payload '{"sensitivity": "high"}' but I get the error 
Debug 2023-12-16 10:48:06Received MQTT message on 'zigbee2mqtt/[Vibration_sensor_1]/set' with data ''{"sensitivity": "low"}''
Info 2023-12-16 10:48:06MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"friendly_name":"[Vibration_sensor_1]"},"type":"entity_not_found"}'

I've changed the friendly name several times, restarting zigbee2mqtt after each just in case and still get that error. I've also tried continuously pressing the button to keep it awake. Any ideas? I've got 2 of these sensors, both give the same error.

### Zigbee2MQTT version

1.34.0-1

### Adapter firmware version

0x26780700

### Adapter

ConBee2

lux73, 2023-12-18:
You have to push the pairing Button short after changing Sensitivity - so the Device wake up and receive the change

Rainz3, 2023-12-18:
> You have to push the pairing Button short after changing Sensitivity - so the Device wake up and receive the change

I've tried that several times on both devices. I've made sure they are awake by pressing the button repeatedly before publishing and after. Still get the entity not found error. This goes for both devices. Same with changing it directly in zigbee2mqtt. I get the error posted above.

fama-lama, 2023-12-20:
i have the same issue

`Publish 'set' 'sensitivity' to 'Workshop seat' failed: 'Error: Write 0x00158d0003d47fa1/1 genBasic({"65293":{"value":1,"type":32}}, {"sendWhen":"immediate","timeout":35000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":4447,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'No network route' (205))'`

additional details.

Zigbee2MQTT version
[1.34.0](https://github.com/Koenkk/zigbee2mqtt/releases/tag/1.34.0) commit: [aae7312](https://github.com/Koenkk/zigbee2mqtt/commit/aae7312)
Coordinator type
zStack3x0
Coordinator revision
20221226
Coordinator IEEE Address
0x00124b0024c0be64
Frontend version
0.6.147
Zigbee-herdsman-converters version
15.130.1
Zigbee-herdsman version
0.25.0

Rainz3, 2024-01-07:
> i have the same issue
>
> `Publish 'set' 'sensitivity' to 'Workshop seat' failed: 'Error: Write 0x00158d0003d47fa1/1 genBasic({"65293":{"value":1,"type":32}}, [...]

Did you ever get this working?

fama-lama, 2024-01-07:
I just updated to z2m 1.35.1 to see if any fix but nope. Same error.

`error 2024-01-08 04:21:27Publish 'set' 'sensitivity' to 'Workshop seat' failed: 'Error: Write 0x00158d0003d47fa1/1 genBasic({"65293":{"value":1,"type":32}}, {"sendWhen":"immediate","timeout":35000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":4447,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'No network route' (205))'`

bobynator93, 2024-01-13:
any updates? Same issue here!

Rainz3, 2024-01-13:
> any updates? Same issue here!

Nope. Tried discord as well. Got it narrowed down to the topic I was using. It should be zigbee2mqtt/Vibration1/set not zigbee2mqtt/[Vibration1]/set. Also the payload was formatted wrong. Should be {"sensitivity": "high"}. Doing this gets around the entity unknown issue but gives me 

`Error 2024-01-13 12:47:10Publish 'set' 'sensitivity' to 'Vibration1' failed: 'Error: Write 0x00158d0009ee3e2b/1 genBasic({"65293":{"value":1,"type":32}}, {"sendWhen":"immediate","timeout":35000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":4447,"transactionSequenceNumber":null,"writeUndiv":false}) failed (no response received (235))'`

closed 2024-08-11 by github-actions[bot]