Koenkk/zigbee2mqtt #20241: zdoRequest error. EZSP ethernet

mhetzi, 2023-12-16:
### What happened?

hi,

Herdsman randomly crashes.
Since bringing the device count from 99 to 65 (by buying a second one) its less frequent.

But a OTA update, will trigger the crash in 1-5 minutes.

### What did you expect to happen?

Beeing stable for weeks, as it was with the TI Stick.

### How to reproduce it (minimal and precise)

Too much activity, OTA update. Joining 3 devices at the same time

### Zigbee2MQTT version

1.34.0 commit: aae7312

### Adapter firmware version

 7.3.1.0 build 176

### Adapter

SLZB-06M

### Debug log

[crash.log](https://github.com/Koenkk/zigbee2mqtt/files/13693390/crash.log)

RoGu777, 2023-12-25:
I am experiencing the same issue with a EFR32MG21 based ZBDongle-E
My Zigbee2MQTT container is randomly crashing and being restartet.

here the crash log:

2023/12/25 17:43:54	stdout	    at Queue.executeNext (/app/node_modules/zigbee-herdsman/src/utils/queue.ts:32:32)
2023/12/25 17:43:54	stdout	    at Object.func (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/adapter/ezspAdapter.ts:370:32)
2023/12/25 17:43:54	stdout	    at EZSPAdapter.nodeDescriptorInternal (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/adapter/ezspAdapter.ts:380:28)
2023/12/25 17:43:52	stdout	    at Driver.zdoRequest (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/driver.ts:572:19)
2023/12/25 17:43:51	stdout	Error: ZdoRequest error

startup sequence:

info  2023-12-25 17:44:40: Starting Zigbee2MQTT version 1.34.0 (commit #aae7312)
info  2023-12-25 17:44:40: Starting zigbee-herdsman (0.25.0)
info  2023-12-25 17:44:43: zigbee-herdsman started (resumed)
info  2023-12-25 17:44:43: Coordinator firmware version: '{"meta":{"maintrel":"3 ","majorrel":"6","minorrel":"10","product":8,"revision":"6.10.3.0 build 297"},"type":"EZSP v8"}'
info  2023-12-25 17:44:43: Currently 10 devices are joined

IgnacioHR, 2024-01-02:
I'm having the same issue but Zigbee2Mqtt version is 1.35.0 and ZBDongle-E version is 7.3.2.0 build 212 EZSP Version is 12

mhetzi, 2024-01-05:
After migration everything to another network coordinator with older firmware

Zigbee2MQTT-Version [1.35.0](https://github.com/Koenkk/zigbee2mqtt/releases/tag/1.35.0) commit: [b5a90e9 (https://github.com/Koenkk/zigbee2mqtt/commit/b5a90e9)
Coordinator-Typ: EZSP v11
Coordinator-Version: 7.2.3.0 build 215

The network has not gone down since 4 days.

so is it firmware version specific or hardware problem? or did i just have lucky days...

mhetzi, 2024-01-15:
Ok and the problems are back.
[z2m.log](https://github.com/Koenkk/zigbee2mqtt/files/13941190/z2m.log)

closed 2024-08-13 by github-actions[bot]