Koenkk/zigbee2mqtt #20242: RTCGQ14LM issues on Conbee 3

emandtf, 2023-12-16:
### What happened?

I'm running latest Z2M DEV using Conbee 2 and all works fine.
When I switch to Conbee 3 (latest firmware 0x264e0900 of 8 December) I'm unable to set any setting (occupancy_timeout, trigger_indicator, motion_sensitivity, etc...) and 99% of times I'm unable to pair new devices of same type.
The red error toast message report that "no response" from the device.

If I switch to Conbee 3 but use "./data/database.db" of Conbee 2 (so without doing a re-pairing all devices) most of those sensors works well as paired with Conbee 2.

### Zigbee2MQTT version

1.34-dev

### Adapter firmware version

0x264e0900

### Adapter

Conbee 3

emandtf, 2023-12-19:
Pairing via a specific Router (and not use "Permit Join (All)" and neither via Coordinator) seems to have fixed the pairing procedure.
However using Conbee and latest 1.34-dev version I'm unable to set any device settings as I mentioned above.

Switching to Conbee 2 fixes this issue.

emandtf, 2024-01-02:
Same "unable to write settings" in latest 1.35 version using Conbee 3....Neither do a "_re-configure_" do anything.
Using Conbee2 works well.

What can be the cause? Just to point me somewhere in the code.... (I'm a developer, I could check)

joujou333, 2024-01-23:
Got the same issue, conbee 2 is working fine, with 3 i can not pair them.
Using latest firmware: deCONZ_ConBeeIII_0x264f0900.bin
Zigbee2mqtt version 1.35.1-1

emandtf, 2024-01-23:
> deCONZ_ConBeeIII_0x264f0900.bin

Did you tried even the newest firmware of January 2024?

joujou333, 2024-01-23:
It is the latest: https://deconz.dresden-elektronik.de/deconz-firmware/?C=M;O=D
Also tried it on my local machine with Deconz, pairing devices works fine on there.

Koenkk, 2024-01-24:
Conbee 3 firmware does not support all Xiaomi devices fully yet (which Conbee 2 does). Dresden Elektronik is working to fix this (at least that's what I heard)

closed 2024-08-23 by github-actions[bot]