Koenkk/zigbee2mqtt #20245: [New device support]:

terierul, 2023-12-16:
### Link

https://www.aliexpress.com/item/1005005897062814.html?spm=a2g0o.order_list.order_list_main.38.21ef1802RfA7UM

### Database entry

{"id":85,"type":"EndDevice","ieeeAddr":"0xa4c13866cfb08c0c","nwkAddr":28414,"manufId":4417,"manufName":"_TZE200_vvmbj46n","powerSource":"Battery","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[4,5,61184,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"stackVersion":0,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":72,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1702758110334,"defaultSendRequestWhen":"immediate"}

### Comments

I did not used any external converter as I do not have a Tuya hub
I do not know which one from repository is equal to test

sweepy-77, 2023-12-17:
Can you test this code?

https://gist.github.com/sweepy-77/772f25eb2efca35689a5309377d0da93

terierul, 2023-12-17:
i will do it in Friday when I will go home
the test to be done following this guide? thanks
https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html

terierul, 2023-12-22:
Hi ,
test results (seems ok - except the battery which probably will need some time to refresh values):
1. database
{"id":85,"type":"EndDevice","ieeeAddr":"0xa4c13866cfb08c0c","nwkAddr":29536,"manufId":4417,"manufName":"_TZE200_vvmbj46n","powerSource":"Battery","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[4,5,61184,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65506":56,"65508":0,"65534":0,"appVersion":72,"manufacturerName":"_TZE200_vvmbj46n","zclVersion":3,"modelId":"TS0601","powerSource":3}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":72,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":1335756102},"lastSeen":1703228942567,"defaultSendRequestWhen":"immediate"}

2. device about page
[image]

3. device exposes page
[image]

terierul, 2023-12-22:
battery value updated to 109% :)
[image]

terierul, 2024-01-07:
Hi and Happy New Year
One more thing I tested and not working trying to change from Celsius to Fahrenheit--> nothing happens, remains in Celsius

sweepy-77, 2024-01-07:
Hi and Happy New Year

I modified the lines below to take into account the set command with the right parameters, it works for me. On the other hand, the min and max alarm temperature settings will remain in °C, no change possible in Fahrenheit on the Tuya app.

`    

Before           e.enum('temp_unit_convert', ea.STATE_SET, ['celsius', 'fahrenheit']).withDescription('Temperature unit'),
After            e.enum('temp_unit_convert', ea.STATE_SET, ['c', 'f']).withDescription('Temperature unit'),

`

terierul, 2024-01-07:
thanks sweepy-77
I will not be able to test the change in the next mouth as I will be out of the location where the sensor is installed
My suggestion is to do the code merging and to promote it
Thanks!

terierul, 2024-01-18:
well tested the new parameters and nothing change. It stucks in Celsius

[image]

> Hi and Happy New Year
>
> I modified the lines below to take into account the set command with the right parameters, it works for me. On the other hand, the min and max alarm temperature settings will [...]

closed 2024-08-17 by github-actions[bot]