Koenkk/zigbee2mqtt #20252: Entity category `config` is invalid for sensors, stops working from HA Core 2024.2.0

seblu, 2023-12-17:
### What happened?

When a device export a "[numeric](https://github.com/Koenkk/zigbee2mqtt/blob/03ba647dc6b5f299f8f3ab441712999fcb3a253e/lib/extension/homeassistant.ts#L597C48-L597C48)" type,  z2m may creates 2 entities, one with domain `sensor` and one with domain `number`. When `number` is allowed, `sensor` is disabled by default.

This numeric entity may represent a configuration property and  you may want to customize it like this:
```yaml
  '0x0123456789abcdef':
    homeassistant:
      temperature_threshold:
        entity_category: config
```

In this case, entity_ids are `sensor.0x0123456789abcdef_temperature_threshold` and `number.0x0123456789abcdef_temperature_threshold`. The both entities share a lot, like `object_id` or `unique_id`.

It causes the following warning message:
```
WARNING (MainThread) [homeassistant.components.mqtt.mixins] Entity category `config` is invalid for sensors, ignoring. This stops working from HA Core 2024.2.0
```
I guess than HA devs, do not recognize read-only entities like sensors as configuration entities and that makes sense.

### What did you expect to happen?

Be able to override each `entity_category` attribute for each entities created by z2m.

```yaml
  '0x0123456789abcdef':
    homeassistant:
      temperature_threshold:
        entity_category: config
      number_temperature_threshold:
        entity_category: diagnostic
```

### How to reproduce it (minimal and precise)

Find a device which export a modifiable numeric value and set it `entity_category` to "config".

### Zigbee2MQTT version

1.34.0

### Adapter firmware version

20230507

### Adapter

zStack3x0

Koenkk, 2023-12-17:
Fixed!

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

closed 2023-12-17 by Koenkk

mundschenk-at, 2023-12-17:
@Koenkk Rather than not having an entity category, I think these sensors should be classified as `diagnostic`.

Koenkk, 2023-12-17:
@mundschenk-at agree, done!