Koenkk/zigbee2mqtt #20253: "Failed to ping" on 2 devices (out of 25)

merlinpimpim, 2023-12-17:
### What happened?

Hello,

Among my 25 devices, I often get "failed to ping" messages on only 2 of them. I don't know why.
All 2 are devices connected to the power supply and are routers.

**This one:**
https://www.zigbee2mqtt.io/devices/4058075729308.html 

But the 1st problem is that it is not recognised as such by Z2M.
Once paired, Z2M detects it as such: https://www.zigbee2mqtt.io/devices/AB3257001NJ.html#osram-ab3257001nj

Then the recurring warning message is as follows:
`Warning 2023-12-16 09:51:49Failed to ping 'ALLÉE - Prise connectée ext. - Osram/Ledvance' (attempt 1/2, Read 0x7cb03eaa0a0c81ba/3 genBasic(["zclVersion"], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":true,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 51063 - 3 - 170 - 0 - 1 after 10000ms))`

**And this one:**
https://www.zigbee2mqtt.io/devices/ZBMINIL2.html#sonoff-zbminil2

`Warning 2023-12-16 10:01:00Failed to ping 'SÀM - Plafonnier - Module SonOff' (attempt 1/2, Read 0x84ba20fffee241d4/1 genBasic(["zclVersion"], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":true,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 5 - 1 - 175 - 0 - 1 after 10000ms))`

### What did you expect to happen?

- Recognition of the right device.

- Stop "failed to ping" warning messages.

### How to reproduce it (minimal and precise)

Nothing special, just pair the devices and read the logs.

### Zigbee2MQTT version

 1.34.0 commit: unknown (!?)

### Adapter firmware version

7.3.1.0 build 0

### Adapter

Skyconnect / EZSP v12

OUARZA, 2023-12-23:
Hello,
I have exactly the same problem with my connected socket since the last update
https://github.com/Koenkk/zigbee2mqtt/issues/20100#issuecomment-1866876752

coco673, 2024-01-07:
I have the same problem and even if i apply a backup nothing change. 
The issue seems to be only for device which can be set because all my aquara sensor are working well 

Any idea where the bug come from @Koenkk ? 
Connection with mosquitto is working, m2z is running

Koenkk, 2024-01-07:
This error means the device is not reachable, I guess setting the state is also not working?

Regarding the device detection, LEDVANCE uses a generic model ID for multiple plugs, therefore we cannot distinguish them  (it's only a cosmetic thing)

merlinpimpim, 2024-01-07:
> This error means the device is not reachable, I guess setting the state is also not working?

Yes.
And that's what I thought too.

But the problem is that it only happens on those 2 devices that are close to the coordinator. And these 2 devices are active devices (connected to the power supply).
What's more, I've got 25 devices and other battery-powered peripherals, known to be less reliable (like Xiaomi Aqara), don't have this problem!
How is this possible?

merlinpimpim, 2024-01-08:
With Z2M 1.35.1 it continues.
Only these 2 devices (active and routers) out of 25 (including several on battery and further away from the coordinator...)
I don't understand why.

> Warning 2024-01-08 03:32:52Failed to ping 'ALLÉE - Portail - Module Nodon' (attempt 1/2, Read 0x9035eafffed8d440/1 genBasic(["zclVersion"], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":true,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (sendZclFrameToEndpointInternal error))
Warning 2024-01-08 03:32:55Failed to ping 'ALLÉE - Portail - Module Nodon' (attempt 2/2, Read 0x9035eafffed8d440/1 genBasic(["zclVersion"], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (sendZclFrameToEndpointInternal error))
Warning 2024-01-08 03:33:33Failed to ping 'SÀM - Plafonnier - Module SonOff' (attempt 1/2, Read 0x84ba20fffee241d4/1 genBasic(["zclVersion"], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":true,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (sendZclFrameToEndpointInternal error))
Warning 2024-01-08 03:33:36Failed to ping 'SÀM - Plafonnier - Module SonOff' (attempt 2/2, Read 0x84ba20fffee241d4/1 genBasic(["zclVersion"], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (sendZclFrameToEndpointInternal error))

merlinpimpim, 2024-01-08:
Just in case: one of the 2 devices (Nodon SIN-4-1-20 module) is also the one mentioned in this bug: https://github.com/Koenkk/zigbee2mqtt/issues/20202

Koenkk, 2024-01-09:
@merlinpimpim I think it's due to an issue with the adapter, updating it's firmware may help (the adapter you are using is not fully supported yet)

merlinpimpim, 2024-01-09:
> @merlinpimpim I think it's due to an issue with the adapter, updating it's firmware may help (the adapter you are using is not fully supported yet)

You mean wait for the next firmware release?
Because right now I'm on the last available fw for this adapter (I think).

OUARZA, 2024-01-10:
Same problem here https://github.com/Koenkk/zigbee2mqtt/issues/20100

merlinpimpim, 2024-01-10:
@OUARZA Bonjour cher ami Français ! :)

In addition: does anyone know why the LQI column is always 255?

merlinpimpim, 2024-01-11:
> @merlinpimpim I think it's due to an issue with the adapter, updating it's firmware may help (the adapter you are using is not fully supported yet)

[image]

@Koenkk : I think it would be better to avoid updating the latest firmware after all! 😅

Koenkk, 2024-01-13:
@merlinpimpim the multiprotocol firmware is experimental from what I understood, It's recommended to use the non-multiprotocol version (just Zigbee)

merlinpimpim, 2024-01-13:
You're right.
(but it would be nice if one day it came out of experimental mode, because it's been around for a while now...)

closed 2024-08-11 by github-actions[bot]