Koenkk/zigbee2mqtt #20259: ZBDongle-P + Ikea bulbs - no pairing - 'zcl' data is from unknown device

nicolasbrailo, 2023-12-17:
### What happened?

I'm having issues pairing Ikea light bulbs with a Sonoff adapter. At least one Hue light bulb seems to work fine. I'm doing:

1. touchlink scan
2. touchlink identify
3. touchlink reset

[NB: I do this with only one light on, all devices except the one under test are off while I run this test]

For all bulbs, the scan and identify works fine (I can see the expected light flashing). For Hue bulbs, the reset works fine, and the bulb pairs after the reset is complete. For Ikea bulbs, the reset finishes successfully, and nothing happens afterwards. I tested this with two different Ikea bulbs.

Checking herdsman logs, I can see the Hue device announces itself after a reset, but the Ikea device doesn't. If I trigger a reset manually (by switching the Ikea light off and on a few times) it seems to be sending messages, but not pairing.

In the logs, I see a bunch of messages like `'zcl' data is from unknown device with address '26036', skipping..` for the failed devices. I don't know if this is relevant.

Any hints to debug the problem?

### Zigbee2MQTT version

1.34.0-dev (c26a6a2) and 1.34.0 (master)

### Adapter firmware version

CC1352P2_CC2652P_launchpad_coordinator_20230507

### Adapter

SONOFF Zigbee 3.0 USB Dongle Plus ZBDongle-P [P, not E]

### Debug log

[logs.tar.gz](https://github.com/Koenkk/zigbee2mqtt/files/13697083/logs.tar.gz)

Hoohm, 2024-05-23:
I have the same problem
```
Scan request failed or was not answered: 'Error: Timeout - null - 254 - null - 4096 - 1 after 500ms'
```

Using a sonoff, ZBdongle-E with a EFR32MG21

Any ideas?

nicolasbrailo, 2024-05-23:
> Using a sonoff, ZBdongle-E with a EFR32MG21
> Any ideas?

I ended up returning it, found the Sonoff products too unstable.

Hoohm, 2024-05-23:
I wouldn't mind trying to solve this but I would need a bit more understanding of the underlying functions.

I can't find where the error messages comes from, it's not in this repo. Might be on some other software running on the dongle?

Hoohm, 2024-05-23:
Found this that seems to try and address a similar problem but that's way beyond what I could solve :(

https://e2e.ti.com/support/wireless-connectivity/zigbee-thread-group/zigbee-and-thread/f/zigbee-thread-forum/1059908/simplelink-cc13x2-26x2-sdk-sdk_5_20_00_52-memory-overflow/3925007#3925007

Hoohm, 2024-05-23:
Tried to downgrade the firmware to [ncp-uart-sw_6.10.3_115200.gbl](https://github.com/xsp1989/zigbeeFirmware/blob/master/firmware/Zigbee3.0_Dongle-NoSigned/EZSP/ncp-uart-sw_6.10.3_115200.gbl) but to no avail

Hoohm, 2024-05-24:
Some updates.

1. Updated the firmware to 7.4.3 on the dongle
2. Changed the zigbeetomqtt config as follows
`adapter: ember`
`baudrate: 230400`

I now get a new error
```
{"target":null,"apsFrame":{"profileId":49246,"clusterId":4096,"sourceEndpoint":0,"destinationEndpoint":0,"options":0,"groupId":65535,"sequence":0},"zclSequence":0,"commandIdentifier":1} timed out after 500ms
```

closed 2025-01-01 by github-actions[bot]