Koenkk/zigbee2mqtt #20265: [New device support]:

heimchemiker, 2023-12-18:
### Link

https://de.aliexpress.com/item/1005004778923575.html

### Database entry

{"id":21,"type":"Router","ieeeAddr":"0xa4c13890ff75fcf8","nwkAddr":12041,"manufId":4417,"manufName":"_TZE204_2imwyigp","powerSource":"Mains (single phase)"

### Comments

The Device is a 3 gang wall switch from Aliexpress.
It has 3 switches, each with one blue LED indicating the state of the switch (dimmed when OFF, bright when ON) and one green LED that flashes when the device is in pairing mode. I don't know it the green LEDs can be turned on individually, I've only seen them on all together. (Pairing done by holding any button for longer than 10 seconds).
It has one beeper that can be turned on or off by pressing any button 5 times and then bolding the button again until it beeps once. This does not produce a zigbee message. (other than the 5 button presses)
It does not have a button hold function (or at least it does not send any Zigbee message when I hold any button, at least in the state it is in now, since I don't know if this could be configured)
It is possible to press any button **twice** then press and hold that button. The associated blue LED then starts to flash **five** times but nothing happens (no Zigbee message is sent and no relay is clicking)
It is possible to press any button **three** times then press and hold that button. The associated blue LED then flashes **twice** but again nothing happens.
Nothing happens pressing any button **4** times then press and hold that button.
I tried to create an external converter. It works in the sense of "when I press the switch, it reports and gets shown in the state of the switch in the dashboard. But when I click the switch in the dashboard, i get the error "Device '0xa4c13890ff75fcf8' has no endpoint 'l1'"

What is my mistake in the converter file?

Also every 30 seconds the device sends 3 messages:
`Debug 2023-12-18 02:06:19Received Zigbee message from '0xa4c13890ff75fcf8', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,0],"type":"Buffer"},"datatype":2,"dp":7}],"seq":19200}' from endpoint 1 with groupID 0
Debug 2023-12-18 02:06:19Received Zigbee message from '0xa4c13890ff75fcf8', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,0],"type":"Buffer"},"datatype":2,"dp":8}],"seq":19456}' from endpoint 1 with groupID 0
Debug 2023-12-18 02:06:19Received Zigbee message from '0xa4c13890ff75fcf8', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,0],"type":"Buffer"},"datatype":2,"dp":9}],"seq":19712}' from endpoint 1 with groupID 0`

The number after the "seq" changes with each message.
I have no idea, what these messages are.

### External converter

```shell
GNU nano 4.8                                                               3er.js                                                                         
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const definition = {
    // Since a lot of TuYa devices use the same modelID, but use different datapoints
    // it's necessary to provide a fingerprint instead of a zigbeeModel
    fingerprint: [
        {
            // The model ID from: Device with modelID 'TS0601' is not supported
            // You may need to add \u0000 at the end of the name in some cases
            modelID: 'TS0601',
            // The manufacturer name from: Device with modelID 'TS0601' is not supported.
            manufacturerName: '_TZE204_2imwyigp',
        },
    ],
    model: 'WS-ZP3F_W3',
    vendor: 'TuYa',
    description: '3 gang wall switch, live only version',
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [tuya.tz.datapoints],
    onEvent: tuya.onEventSetTime, // Add this if you are getting no converter for 'commandMcuSyncTime'
    configure: tuya.configureMagicPacket,
    exposes: [
        // Here you should put all functionality that your device exposes
        e.switch().withEndpoint('l1').setAccess('state', ea.STATE_SET),
        e.switch().withEndpoint('l2').setAccess('state', ea.STATE_SET),
        e.switch().withEndpoint('l3').setAccess('state', ea.STATE_SET)
    ],
    meta: {
        // All datapoints go in here
        tuyaDatapoints: [
                [1, 'state_l1', tuya.valueConverter.onOff],
                [2, 'state_l2', tuya.valueConverter.onOff],
                [3, 'state_l3', tuya.valueConverter.onOff]
        ],
    },
};
```

heimchemiker, 2024-03-06:
Hi again,
I have been able to get the witch to work:

`const {} = require('zigbee-herdsman-converters/lib/modernExtend');
// Add the lines below
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const {} = require('zigbee-herdsman-converters/lib/tuya');
const utils = require('zigbee-herdsman-converters/lib/utils');
const globalStore = require('zigbee-herdsman-converters/lib/store');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    fingerprint: [
        {
                modelID: 'TS0601',
                manufacturerName: '_TZE204_2imwyigp',
        },
    ],
    model: 'TS0601',
    vendor: 'TuYa',
    description: '3 Gang Wallswitch, live only',
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [tuya.tz.datapoints],
    onEvent: tuya.onEventSetTime,
    configure: tuya.configureMagicPacket,
    exposes: [
        e.switch().withEndpoint('l1'),
        e.switch().withEndpoint('l2'),
        e.switch().withEndpoint('l3'),
    ],
    endpoint: (device) => {
        return {'l1': 1, 'l2': 1, 'l3': 1};
    },
    meta: {
        multiEndpoint: true,
        tuyaDatapoints: [
            [1, 'state_l1', tuya.valueConverter.onOff],
            [2, 'state_l2', tuya.valueConverter.onOff],
            [3, 'state_l3', tuya.valueConverter.onOff],
        ],
    },
};

module.exports = definition;`

Now I only have part of my original problem:
Every 30 seconds I get 3 pairs of debug messages:

`Zigbee2MQTT:debug 2024-03-06 04:44:47: Received Zigbee message from '0xa4c13890ff75fcf8', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,0],"type":"Buffer"},"datatype":2,"dp":7}],"seq":40960}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2024-03-06 04:44:47: Datapoint 7 not defined for '_TZE204_2imwyigp' with value 0
Zigbee2MQTT:debug 2024-03-06 04:44:47: Received Zigbee message from '0xa4c13890ff75fcf8', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,0],"type":"Buffer"},"datatype":2,"dp":8}],"seq":41216}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2024-03-06 04:44:47: Datapoint 8 not defined for '_TZE204_2imwyigp' with value 0
Zigbee2MQTT:debug 2024-03-06 04:44:47: Received Zigbee message from '0xa4c13890ff75fcf8', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,0],"type":"Buffer"},"datatype":2,"dp":9}],"seq":41472}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2024-03-06 04:44:47: Datapoint 9 not defined for '_TZE204_2imwyigp' with value 0`

I have no Idea, what these Datapoints could be and how to map them. I'll continue to see if I can find any clues, but first I'll need to get to sleep.
Any help is welcome.

Koenkk, 2024-03-06:
see https://www.zigbee2mqtt.io/advanced/support-new-devices/03_find_tuya_data_points.html on how to find what these datapoints mean, if everything works, you can also leave it and just submit a pull request

VictorPavlushin, 2024-03-31:
{"1":"Switch 1","2":"Switch 2","3":"Switch 3","7":"Timer 1","8":"Timer 2","9":"Timer 3","14":"Restart Status","16":"Backlight Switch"}

Koenkk, 2024-04-01:
So these are the timers, I propose to ignore those. @heimchemiker do you want to make a pull request to [tuya.ts](https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/src/devices/tuya.ts) to add support for this device out-of-the-box?

VictorPavlushin, 2024-04-02:
1) Why do you want to ignore timers?
2) Will there be support for turning the backlight on and off?
3) Will support for setting the last state after a power outage be revised?

closed 2024-10-30 by github-actions[bot]

lyap179, 2024-12-09:
This converter is not working now. How can I make the switch work?

lyap179, 2024-12-09:
@Koenkk 
Can you add support for this device out of the box?

heimchemiker, 2024-12-14:
Sorry for not finishing this topic, but Life happenend and I didn't find the time.
@Koenkk Yes, I would like to have this switch supported out of the Box, as this is a generic swicht from Aliexpress, so I guess there might be others using the same or similar internals. I would also like to have the timers and Backlightcontrol supported, but would not know how and foremost I just don't have the time to complete this.
So @VictorPavlushin do you maybe have the time and knowledge to complete this?

lyap179, 2025-01-21:
> Sorry for not finishing this topic, but Life happenend and I didn't find the time. [@Koenkk](https://github.com/Koenkk) Yes, I would like to have this switch supported out of the Box, as this is a [...]

Hi! Can you reopen the issue? Maybe that will help us move forward?

Koenkk, 2025-01-22:
Added it!

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

heimchemiker, 2025-01-26:
Just tried it out, and it works. Thank you so much @Koenkk

gelmeone, 2025-04-08:
I didn't create a new query. I bought a similar 3-key wall switch: [Смотри, что есть на AliExpress! Интеллектуальный выключатель Eztofa ZigBee за 1 891 ₽ - уже со скидкой -45%
https://sl.aliexpress.ru/p?key=gAZuG2k](url)
Everything is similar in terms of functionality and management. The only thing that differs is the Manufacturer Zigbee_TZE200_2imwyigp. Support status
Not supported

Can it be added to z2m?