Koenkk/zigbee2mqtt #20268: JY-GZ-01AQ Going Offline every day

Ajhomepn, 2023-12-18:
### What happened?

The smoke alarm ( Xiaomi JY-GZ-01AQ) is getting offline every day, 
To reconnect I normally just have to "reconfigure" on the zigbee2mqtt interface. 

I have try to connect the device to different "coordinators" , but after a few hours it goes offline anyway 

### What did you expect to happen?

The device should be connected all the time 

### Zigbee2MQTT version

1.34.0-1

### Adapter firmware version

2019www. ( not specify) 

### Adapter

SkyConnect

Ra72xx, 2023-12-18:
Maybe #19747

MarvAmBass, 2024-01-13:
same issue here, my `JTYJ-GD-01LM/BW` work fine but the aqara `JY-GZ-01AQ` switch to offline after some time.

repairing using the three times button press works and it's back online

I use conbee II (FW Version: `0x26580700`) on a raspberry

lasota-piotr, 2024-01-24:
I have the same issue with my two JY-GZ-01AQ

ciB89, 2024-02-06:
Same issue here.

ccle610, 2024-03-02:
I unfortunately also have the same experience - Current version: 1.36.0-1

MarvAmBass, 2024-03-17:
new update from my side,

I've upgraded my conbee ii from firmware version: `deCONZ firmware version 26580700`

to version `deCONZ_ConBeeII_0x26720700.bin.GCF`
as sugested here: https://www.zigbee2mqtt.io/devices/JY-GZ-01AQ.html

I'll keep you updated if this helped

MarvAmBass, 2024-03-21:
so everyone, after a few days I can give some updates:

the recommended firmware update did indeed help with my conbee 2 usb stick on a raspberry pi.
I know that this information must have been added to the wiki after I wrote my enty in this issue.

maybe you also find a working firmware version, or maybe you find some changelogs between my two versions (`deCONZ_ConBeeII_0x26580700.bin.GCF` to `deCONZ_ConBeeII_0x26720700.bin.GCF`) and you are able to find out what the firmware for your zigbee adapter needs to support

anyways I'm out from this issue, good luck everyone - hope you also find a solution

Milenco, 2024-03-28:
I can confirm updating to firmware `deCONZ_ConBeeII_0x26720700.bin.GCF` resolved this issue for me. Now my smoke detectors report back about every 1-2 hours. Thanks @MarvAmBass.

Not that using a newer firmware might zigbee2mqtt to not be able to recognize the ConbeeII stick anymore. I was enthusiastic and updated to the newest firmware, only the downgrade again to `deCONZ_ConBeeII_0x26720700.bin.GCF`.

DAVIZINH0, 2024-04-02:
Hello.
I have the same problem with the same device and the same firmware in the device.
But im using conbee3 device in a haos instalation with zigbee2mqtt as addon

ccle610, 2024-04-02:
Hi there,
since I've updated my Sonoff  cc2652p to the latest Firmware => Hardware: zStack3x0 FW: 20230507 (coming from 2021), my Smoke dedector is always online and polls every zwo hours. 
Beside: I've re-paired it with only coordinator was accepting new devices to prevent, it will connect to my Smart Plug (where it was registered before!)
At all, I`m happy with this setup for several days.

DAVIZINH0, 2024-04-05:
Hello.
I have the same problem. aprox at 24 hours, the device apears offline in zigbee2mqtt.
I use a conbee3 in a home assistant OS instalation.

I tried to allow new devices only in the coordinator and not works, after 24 hours, disconected again.

my device is at 1 meter of the coordinator so the distance is not a problem :-(

DAVIZINH0, 2024-04-18:
hello again.
I make one test this days and it seams to work.
I launch a selftest action once per day. before 24 hours.
I must push the selftest buton 2 times, one push, wait 30 seconds and push again.

after the second push, the device sounds 3 times.

I make a automation that make this at 11:00 and at 20:00.
push button
wait 30sg
push button again

in the next few days I report to this topic if this works

DAVIZINH0, 2024-04-20:
After 3 days online.
Today the device go offline again.
The selfttest runa each 12 hours but, not works to solve this problem.

This device will make me crazy!!!!

DAVIZINH0, 2024-06-13:
nobody have progress with this device?

mihaipetrescutwwhp, 2024-06-13:
The only working workaround but not ideal by any means is to add an automation that every 4 hours triggers and does the following sequence in order:
1.Switch: Turn On for the Heartbeat Indicator attribute 
2.Delay for 10 seconds
3.Switch: Turn Off for the Heartbeat Indicator attribute

This works with 4 Smoke sensors for example - same model, firmware etc. Stupid need but...it works, until we get an official fix.

DAVIZINH0, 2024-06-24:
> The only working workaround but not ideal by any means is to add an automation that every 4 hours triggers and does the following sequence in order: 1.Switch: Turn On for the Heartbeat Indicator [...]

I dont test this metodo (of the heartbeat) and seams to work  during some days.
thanks a lot!

mihaicristianpetrescu, 2024-06-24:
The only working workaround but not ideal by any means is to add an automation that every 4 hours triggers and does the following sequence in order:
1.Switch: Turn On for the Heartbeat Indicator attribute
2.Delay for 10 seconds
3.Switch: Turn Off for the Heartbeat Indicator attribute

This works with 4 Smoke sensors for example - same model, firmware etc. Stupid need but...it works, until we get an official fix.

closed 2025-01-09 by github-actions[bot]

xelemorf, 2025-01-14:
I have 5 of these devices (4 of them are newly bought) and experiencing a similar issue only on one of them.
Right after pairing the device no longer responds. Pressing it's button should wake it and report back to Z2M but this one does not. Interestingly enough, the device works without any issue in Aqara ecosystem.
The firmware levels are all over the place. 

[image]

Shadesh, 2025-02-04:
Same problem here with the JY-GZ-01AQ device with IOBroker and latest Zigbee Packages installed.
I also already have installed the latest firmware 20240710 on my Sonoff ZBdongle-P.
The only way to bring it back is to repair each of my three devices.

xelemorf, 2025-02-04:
Not sure if this is a bigger Z2M problem or just a device compatibility problem, but started noticing a similar behaviour with other newly added (non-aqara) devices aswell.
The interesting point of not here is that I have at least one exactly the same type of device already paired and working reliably while addig a new one of the same just straight up not working, relevant thread: https://github.com/zigbee2mqtt/hassio-zigbee2mqtt/issues/724

PHB-fra, 2026-10-04:
The problem is still present for me too.